zephyrproject-rtos/zephyr #47477: qemu_leon3: tests/kernel/fpu_sharing/generic/ failed when migrating to new ztest API

LixinGuoX, 2022-07-07:
**Describe the bug**
I am migrating `tests/kernel/fpu_sharing/generic/` to new ztest API, see #47475
It seems like qemu_leon3 is incompatible with new ztest API on `tests/kernel/fpu_sharing/generic/`.

**To Reproduce**
Steps to reproduce the behavior:
1. See CI error of #47475 

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-1668-gbdcf1df255cf  ***
Running TESTSUITE fpu_sharing_generic
===================================================================
START - test_load_store
Load and store OK after 0 (high) + 78 (low) tests
Load and store OK after 100 (high) + 3618 (low) tests
Load and store OK after 200 (high) + 7158 (low) tests
Load and store OK after 300 (high) + 10698 (low) tests
Load and store OK after 400 (high) + 14238 (low) tests
Load and store OK after 500 (high) + 17778 (low) tests
 PASS - test_load_store in 10.24 seconds
===================================================================
START - test_pi
Pi calculation OK after 50 (high) + 19 (low) tests (computed 3.141611)
Pi calculation OK after 150 (high) + 58 (low) tests (computed 3.141611)
Pi calculation OK after 250 (high) + 97 (low) tests (computed 3.141611)
Pi calculation OK after 350 (high) + 136 (low) tests (computed 3.141611)
Pi calculation OK after 450 (high) + 175 (low) tests (computed 3.141611)
 PASS - test_pi in 25.59 seconds
===================================================================
TESTSUITE fpu_sharing_generic succeeded
===================================================================
RunID: c8a022d34c6fe37d9a834fb3193001cd
PROJECT EXECUTION SUCCESSFUL
E: 
E: tt = 0x02, illegal_instruction
E: 
E:       INS        LOCALS     OUTS       GLOBALS
E:   0:  40001f18   f3001fc0   40001e74   00000000
E:   1:  40001e74   40007f58   40001e94   000000e1
E:   2:  40001e94   40007f5c   33333332   00000002
E:   3:  33333332   00000f00   0000008e   00000001
E:   4:  00000000   40003c00   0000000f   00000004
E:   5:  000000e0   f3001fc0   00000000   00000010
E:   6:  4000ee48   40000890   4000ede8   00000000
E:   7:  40003864   00000000   40001f1c   00000000
E: 
E: psr: f30000c7   wim: 00000004   tbr: 40000020   y: 00000000
E:  pc: 4000b558   npc: 4000b7bc
E: 
E:       pc         sp
E:  #0   4000b558   4000ede8
E:  #1   40003864   4000ee48
E: 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x40009ca0 (idle)
E: Halting system
```

fabiobaltieri, 2022-07-12:
Possibly reported on stm32 as well in #47696?

LixinGuoX, 2022-07-14:
I think this is a platform specific issue.

fabiobaltieri, 2022-07-14:
@LixinGuoX can you try with

```
diff --git a/tests/kernel/fpu_sharing/generic/prj.conf b/tests/kernel/fpu_sharing/generic/prj.conf
index 3a00fddc8e..b48feb6d41 100644
--- a/tests/kernel/fpu_sharing/generic/prj.conf
+++ b/tests/kernel/fpu_sharing/generic/prj.conf
@@ -3,5 +3,5 @@ CONFIG_FPU=y
 CONFIG_FPU_SHARING=y
 CONFIG_STDOUT_CONSOLE=y
 CONFIG_CBPRINTF_FP_SUPPORT=y
-CONFIG_MAIN_STACK_SIZE=1024
+CONFIG_MAIN_STACK_SIZE=2048
 CONFIG_ZTEST_NEW_API=y
```

Worked for me on a local run.

julius-barendt, 2022-07-14:
SPARC is pretty stack heavy yes. But CONFIG_MAIN_STACK_SIZE=512 will also make the test run without problems.

It breaks somewhere in '_z_cbvprintf_impl_' and using _CONFIG_CBPRINTF_NANO=y_ will also fix the issue. But I haven't found a root cause yet.

LixinGuoX, 2022-07-18:
Yes, this could also be related to new ztest API, for now I will workaround it by increasing the main stack size in #47475.
Also, we should keep finding the root cause.

julius-barendt, 2022-07-20:
I can confirm that the root cause is a stack issue and that it worked with stack size 512 was just luck.

It stems from the commit _129ddfe5c62_ that adds the line `CONFIG_MAIN_STACK_SIZE=1024.`
This overrides and halves the default 2048 specified in `/arch/sparc/Kconfig`. 

So the fix you made is sufficient to solve this issue. But an argument could be made to remove the line to avoid overriding defaults in `prj.conf` and let platforms increase it's own stack when needed on a case-by-case basis. Such as with `tests/subsys/logging/log_api/boards/qemu_leon3.conf`

closed 2022-10-03 by github-actions[bot]