zephyrproject-rtos/zephyr #47508: tests: arch: the xtensa_asm2 test is broken

enjiamai, 2022-07-07:
**Describe the bug**
Found the tests/arch/xtensa_asm2 test is broken during trying to move it to new ztest API. (Maybe this test not has to migrate)  
Seems like two problem here:
(1) The tests/arch/xtensa_asm2 is broken.
(2) The qemu of xtensa cannot work well in zephry-sdk-0.14.x in this test. (It works with zephyr-sdk-0.13.2)

**More detail information and error logs**

(1) the test is broken due to not including the header files, shows following messages:
```
CMakeFiles/app.dir/src/asmhelp.S.obj.d -o CMakeFiles/app.dir/src/asmhelp.S.obj -c /home/mm/zephyrproject/zephyr/tests/arch/xtensa_asm2/src/asmhelp.S
/home/mm/zephyrproject/zephyr/tests/arch/xtensa_asm2/src/asmhelp.S:6:10: fatal error: xtensa-asm2-s.h: No such file or directory
    6 | #include <xtensa-asm2-s.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```
Then we can fix it by:
```
diff --git a/tests/arch/xtensa_asm2/CMakeLists.txt b/tests/arch/xtensa_asm2/CMakeLists.txt
index f46fa63acf..edcd80235c 100644
--- a/tests/arch/xtensa_asm2/CMakeLists.txt
+++ b/tests/arch/xtensa_asm2/CMakeLists.txt
@@ -6,4 +6,8 @@ project(xtensa_asm2)

 enable_language(ASM)

+target_include_directories(app PRIVATE
+  ${ZEPHYR_BASE}/arch/${ARCH}/include
+  )
+
target_sources(app PRIVATE src/main.c src/asmhelp.S)

```
But we'll see CONFIG_MULTITHEADING=n cannot be met.
```
/home/mm/zephyrproject/zephyr/tests/arch/xtensa_asm2/src/main.c:13:2: error: #error Disable multithreading for this unit test!
   13 | #error Disable multithreading for this unit test!
      |  ^~~~~
```

(2) If we skip this check, it can be built successfully by applying following changes. 

```
diff --git a/tests/arch/xtensa_asm2/src/asmhelp.S b/tests/arch/xtensa_asm2/src/asmhelp.S
index 651da7e02e..cb9d68547a 100644
--- a/tests/arch/xtensa_asm2/src/asmhelp.S
+++ b/tests/arch/xtensa_asm2/src/asmhelp.S
@@ -202,4 +202,4 @@ _handle_excint:
  * handler.  There is a timer on most cores (qemu and LX6/ESP-32 at
  * least) that can be used for unit testing.
  */
-DEF_EXCINT 5, _handle_excint, handle_int5_c
+//DEF_EXCINT 5, _handle_excint, handle_int5_c

diff --git a/tests/arch/xtensa_asm2/src/main.c b/tests/arch/xtensa_asm2/src/main.c
index ed7f6268c2..ec535f223d 100644
--- a/tests/arch/xtensa_asm2/src/main.c
+++ b/tests/arch/xtensa_asm2/src/main.c
@@ -9,9 +9,10 @@
 #include <zephyr/sys/printk.h>
 #include <xtensa-asm2.h>

-#ifdef CONFIG_MULTITHREADING
-#error Disable multithreading for this unit test!
-#endif
+
+//#ifdef CONFIG_MULTITHREADING
+//#error Disable multithreading for this unit test!
+//#endif

 /* Just random numbers intended to whiten the register contents during
  * the spill test and make every bit of every register in every call

```

But it cannot run well by "west build -t run", it shows following error messages: 

```
[150/151] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: sample_controller
/bin/sh: 1: QEMU-NOTFOUND: not found
FAILED: zephyr/CMakeFiles/run_qemu /home/mm/zephyrproject/zephyr/build/zephyr/CMakeFiles/run_qemu
cd /home/mm/zephyrproject/zephyr/build && QEMU-NOTFOUND -machine sim -semihosting -nographic -cpu sample_controller -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -icount shift=6,align=off,sleep=off -rtc clock=vm -kernel /home/mm/zephyrproject/zephyr/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/mm/zephyrproject/zephyr/build --target run
```
you will see error message of QEMU-NOTFOUND: not found

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. west build -p always tests/arch/xtensa_asm2 -b qemu_xtensa -t run
3. See error

**Expected behavior**
Test should pass.

**Impact**
The test is broken.  Not very clear how this impacts others.

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain Zephyr SDK-0.14.2
 - Commit: 9588ecdba8ba660a7616293db4cf8bf3f9aa6a22 (not dedicate to this commits, broken for some time)

edersondisouza, 2022-07-12:
@enjiamai Created a PR simply removing this test (#47710), but one thing got me curious. With your suggested changes, I was able to run the test (it hangs, so not really useful), but no "QEMU NOTFOUND" issues. I wonder if there's something wrong in your environment. Can you run anything using qemu_xtensa?

closed 2022-07-12 by nashif

enjiamai, 2022-07-12:
Hi @edersondisouza , the "QEMU NOTFOUND" is my SDK problem, it vanished after I re-install the zephyr-SDK. Sorry for not updating this information. Thanks for fixing this test!

PR #47710 (2022-07-12): test/arch: Remove xtensa_asm2 test
This test has been bitrotting for a while - years, in fact. As it didn't
have a `testcase.yaml`, it wasn't being run on twister. The code expects
to be run on single thread environment, but xtensa arch currently
doesn't support that.

This patch simply removes this test - if it's deemed important, it can
be reworked and readded in the future.

Fixes #47508.
commit 63a49728 (2022-07-12): test/arch: Remove xtensa_asm2 test

This test has been bitrotting for a while - years, in fact. As it didn't
have a `testcase.yaml`, it wasn't being run on twister. The code expects
to be run on single thread environment, but xtensa arch currently
doesn't support that.

This patch simply removes this test - if it's deemed important, it can
be reworked and readded in the future.

Fixes #47508.