zephyrproject-rtos/zephyr #47547: move parts of zephyr to a library

M1cha, 2022-07-08:
I've been working with zephyr for a few years now and the amount of C-goodies you invented and implemented is just insane(in a very good way).
The thing is that most of that doesn't have to be tied to zephyr and it would be extremely beneficial for all kinds of C-projects. And to be quite honest, If put all your non-zephyr specific code into a library, that library would have the potential to become an industry-standard libc-replacement.

In incomplete list of code I'd like to see moved out of zephyr:
- cbprintf
- pretty much all the macro utils(especially the loops)
- lists
- compiler-abstractions(e.g for static assertions)
- ringbuffer
- json
- logging(would quote hard though in it's current state)

henrikbrixandersen, 2022-11-02:
Having to maintain all of this in a library instead of in-tree would complicate the maintenance tasks considerably. I'm sorry, but I don't see this as beneficial to the Zephyr Project.

nashif, 2022-11-17:
I do not think the "move" part will work, however abstracting some of the code in a way to allow use in different environments is not a bad idea and AFAIK some of the code is already at the level of abstraction you can pull it out and use somewhere else.  If you have any specific proposal how to make this work for some of the code, in the zephyr project itself and not by moving things out, feel free to file a PR.

nashif, 2022-11-17:
> Those have things like useful macros that every project re-implements because libc doesn't provide them.

maybe start with those.

nashif, 2024-02-08:
closing as not-planned.

closed 2024-02-08 by nashif