zephyrproject-rtos/zephyr #47551: Enabling CONFIG_OPENTHREAD_SRP_CLIENT on NRF52840 dongle board leads to MBED compilation errors.

fcgdam, 2022-07-08:
**Describe the bug**
When enabling the option CONFIG_OPENTHREAD_SRP_CLIENT the compilation fails inside the MBED library with not all prerequisites;
```

/opt/zephyrproject/modules/crypto/mbedtls/include/mbedtls/check_config.h:107:2: error: #error "MBEDTLS_ECDSA_C defined, but not all prerequisites"
  107 | #error "MBEDTLS_ECDSA_C defined, but not all prerequisites"
      |  ^~~~~
/opt/zephyrproject/modules/crypto/mbedtls/include/mbedtls/check_config.h:128:2: error: #error "MBEDTLS_ECDSA_DETERMINISTIC defined, but not all prerequisites"
  128 | #error "MBEDTLS_ECDSA_DETERMINISTIC defined, but not all prerequisites"
      |  ^~~~~
/opt/zephyrproject/modules/crypto/mbedtls/include/mbedtls/check_config.h:145:2: error: #error "MBEDTLS_ECP_C defined, but not all prerequisites"
  145 | #error "MBEDTLS_ECP_C defined, but not all prerequisites"
      |  ^~~~~
ninja: build stopped: subcommand failed.

```
**Software versions**
Platform: Linux (Arch)
SDK: 0.14.2
Zephyr release: 3.1.99 (commit: b2b7665b9c90f91aa1dd21c82c4c0ed0865d3bf8)
Compiled with: **west** version v0.13.1
Board: nrf52840dongle_nrf52840 

**To Reproduce**
Steps to reproduce the behavior:

Not the real example, but shows the error when compiling to the NRF52840 dongle board:

1. cd (...)/zephyr/samples/net/sockets/echo_client
2. Compile with: west build -b nrf52840dongle_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf"
3. With the above command it compiles fine.
4. Add the following line to **overlay-ot.conf** file: CONFIG_OPENTHREAD_SRP_CLIENT=y
5. Compile again with: west build -b nrf52840dongle_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf"
6. See that the compilation fails

**Expected behavior**
Compilation should succeed and SRP commands and API should be available.

**Impact**
No SRP enabled nodes possible.

rachelh1998, 2022-07-13:
I get a similar error when enabling these configs
# TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA_ENABLED=y

The error:
#error "MBEDTLS_ECP_C defined, but not all prerequisites"

edmont, 2022-07-13:
Hi @fcgdam, could you try these changes? https://github.com/zephyrproject-rtos/zephyr/pull/47756

fcgdam, 2022-07-14:
Hi @edmont, yes it compiles now and the SRP cli commands are available. Nevertheless it seems that SRP is not working, but that is another issue.

closed 2022-07-14 by fcgdam