zephyrproject-rtos/zephyr #47568: uart_mcux_lpuart driver activates the noise error interrupt but does not clear the noise error flag

kt-peters, 2022-07-08:
**Describe the bug**

When activating error interrupts (function mcux_lpuart_irq_err_enable()), the uart_mcux_lpuart driver also activates the noise error interrupt.  However, the error check function (mcux_lpuart_err_check()) does not clear the corresponding error flag. Hence, if a noise error occurs, the device will be stuck in infinite noise error interrupts (until a watchdog triggers).

**To Reproduce**

Build any application using a UART for a deivce using the uart_mcux_lpuart driver. Generate a noise error on the UART.

**Expected behavior**

Either the error should be ignored (i.e. the interrupt should not be activated) or the noise error flag should be cleared in mcux_lpuart_err_check() to prevent further interrupts.

**Impact**

Severe but rare. Device freezes in an infinite interrupt.

**Additional context**

mcux_lpuart_err_check() does clear the Rx overrun flag, even though the corresponding interrupt is not enabeld.

dleach02, 2022-07-22:
@kt-peters, can you try #48191 in your setup to ensure it addresses your problem?

kt-peters, 2022-07-25:
> @kt-peters, can you try #48191 in your setup to ensure it addresses your problem?

It does. Thanks for the fix.

PR #48191 (2022-07-25): drivers: lpuart: clear noise status flag on errors
The LPUART driver activates the "noise" error interrupt but
neglects to clear the error bit when the interrupt occurs. This
causes the device to freeze. Clear the "noise" status to avoid
the lockup and track instances of this as a UART_ERROR_PARITY.

fixes #47568
commit 88ba2fe3 (2022-07-25): drivers: lpuart: clear noise status flag on errors

The LPUART driver activates the "noise" error interrupt but
neglects to clear the error bit when the interrupt occurs. This
causes the device to freeze. Clear the "noise" status to avoid
the lockup and track instances of this as a UART_ERROR_PARITY.

fixes #47568
closed 2022-07-25 by carlescufi