zephyrproject-rtos/zephyr #47576: undefined reference to `__device_dts_ord_20' When building with board hifive_unmatched on flash_shell samples

ngboonkhai, 2022-07-09:
**Describe the bug**
I was trying to build the flash_shell samples using the board hifive_unmatched and the compilation failed with undefined reference to `__device_dts_ord_20'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

**Board**
hifive_unmatched 

**What have you tried to diagnose or workaround this issue?**
- I verified the DT_DRV_COMPAT has named correctly
- I verified that the dt/bindings is in placed

**To Reproduce**
Steps to reproduce the behavior:
1. west init zephyrproject
2. cd zephyrproject
3. west update
4. cd zephyr
5. west build -b hifive_unmatched samples/drivers/flash_shell --pristine
6. see error

**Expected behavior**
Build success without error

**Impact**
(Showstoppers) This is blocking the PR https://github.com/zephyrproject-rtos/zephyr/pull/46443

**Logs and console output**
```
[09/07/2022 - 07:19.01] [boonkhai.boonkhai-container-55d96b56d8-fgzrj]
/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr
$ west build -b hifive_unmatched samples/drivers/flash_shell --pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr repository).
-- Application: /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/samples/drivers/flash_shell
-- Found Python3: /build/zephyrproject/.venv/bin/python3.6 (found suitable exact version "3.6.15") found components: Interpreter 
-- Cache files will be written to: /nfs/site/home/boonkhai/.cache/zephyr
-- Zephyr version: 3.1.99 (/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: hifive_unmatched
-- Found host-tools: zephyr 0.13.2 (/build/zephyr-sdk-0.13.2)
-- Found dtc: /build/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.2 (/build/zephyr-sdk-0.13.2)
-- Found BOARD.dts: /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/boards/riscv/hifive_unmatched/hifive_unmatched.dts
-- Generated zephyr.dts: /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/samples/drivers/flash_shell/Kconfig
Loaded configuration '/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/boards/riscv/hifive_unmatched/hifive_unmatched_defconfig'
Merged configuration '/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/samples/drivers/flash_shell/prj.conf'
Configuration saved to '/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /build/zephyr-sdk-0.13.2/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build
-- west build: building application
[1/163] Preparing syscall dependency handling

[3/163] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr), build: zephyr-v3.1.0-1882-gb161554ee862
[49/163] Building C object CMakeFiles/app.dir/src/main.c.obj
/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/samples/drivers/flash_shell/src/main.c: In function 'do_write_unaligned':
/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/samples/drivers/flash_shell/src/main.c:304:23: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  304 |   off_t data_offset = (off_t)buf;
      |                       ^
/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/samples/drivers/flash_shell/src/main.c:317:7: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  317 |       (const void *)data_offset,
      |       ^
[153/163] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /build/zephyr-sdk-0.13.2/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -L"/build/zephyr-sdk-0.13.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.3.0/rv64imac/lp64/medany"  -L/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcmodel=medany  -mabi=lp64  -march=rv64imac  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build/zephyr && /usr/local/bin/cmake -E echo
/build/zephyr-sdk-0.13.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.3.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(spi_nor.c.obj):/p/psg/swip/w/boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/drivers/flash/spi_nor.c:1230: undefined reference to `__device_dts_ord_20'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /nfs/png/disks/swuser_work_boonkhai/git/boonkhai_github/zephyr-from-mainline/zephyrproject/zephyr/build

```

**Environment (please complete the following information):**
- Linux 
- Zephyr SDK
- zephyr-sdk-0.13.2

ngboonkhai, 2022-07-09:
@katsuster @cfriedt Please take a look at this

carlocaione, 2022-07-09:
@galak 
```
92ad49e17384cee75f73646800470a4d4bee92e7 is the first bad commit
commit 92ad49e17384cee75f73646800470a4d4bee92e7
Author: Kumar Gala <galak@kernel.org>
Date:   Tue Jul 5 22:20:10 2022 -0500

    drivers: flash: spi-nor: Set driver Kconfig default based on dts
    
    Change Kconfig default to be based on if the devicetree has the
    jedec,spi-nor driver enabled.
    
    Signed-off-by: Kumar Gala <galak@kernel.org>

 drivers/flash/Kconfig.nor | 3 +++
 1 file changed, 3 insertions(+)

```

gmarull, 2022-07-11:
FYI https://github.com/zephyrproject-rtos/zephyr/pull/47602

closed 2022-07-11 by carlescufi

madscientist159, 2022-12-25:
I ran into this bug recently, and as the error messages are very opaque and it took a while to track down I figured I'd post the root cause here.

The trigger seems to be having multiple SPI Flash devices active (possibly related to bug #32100)  The patches referenced above effectively disable all but one SPI Flash device, which is far from ideal but as there seems to be no desire to implement multi-device SPI Flash support this seems to be the best that Zephyr can do right now.