zephyrproject-rtos/zephyr #47588: twister : frdm_k64f : zero-length gnu_printf format string

shaoanxli, 2022-07-11:
**Describe the bug**
The print function is called and the function is not intended to be used to print characters, in this case shows build failure.

For example :
tests/lib/sprintf/src/main.c:778:24    ret = fprintf(stdout, "");
usb/device/usb_dc_kinetis.c:840:10     LOG_DBG("");
tests/lib/sprintf/src/main.c:760:15    ret = printf("");

It seems to be an error to print noting with twister command , but it is fine to use west build command.

Take tests/lib/sprintf/ as an example .

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p frdm_k64f --device-testing --hardware-map map.yml -T tests/lib/sprintf/ -vv
3. See error

**Expected behavior**
The test passes.

**Impact**
The test can not run by twister command.

**Logs and console output**
```
[25/136] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj
ccache /home/ztest/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE="CONFIG_FLASH_SIZE*1024" -DCPU_MK64FN1M0VLL12 -DKERNEL -DTC_RUNID=ec44835519e9be9f92aef986773c3ea9 -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/ztest/zephyrproject/zephyr/include/zephyr -I/home/ztest/zephyrproject/zephyr/include -I/home/ztest/zephyrproject/zephyr/twister-out/frdm_k64f/tests/lib/sprintf/libraries.libc.sprintf/zephyr/include/generated -I/home/ztest/zephyrproject/zephyr/soc/arm/nxp_kinetis/k6x -I/home/ztest/zephyrproject/zephyr/soc/arm/nxp_kinetis/common -I/home/ztest/zephyrproject/zephyr/subsys/testsuite/include -I/home/ztest/zephyrproject/zephyr/subsys/testsuite/ztest/include -I/home/ztest/zephyrproject/zephyr/subsys/testsuite/include/zephyr -I/home/ztest/zephyrproject/zephyr/subsys/testsuite/ztest/include/zephyr -I/home/ztest/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/ztest/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12 -I/home/ztest/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12/drivers -I/home/ztest/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/home/ztest/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/uart -I/home/ztest/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/port -I/home/ztest/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/sysmpu -isystem /home/ztest/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/ztest/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /home/ztest/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -Os -imacros /home/ztest/zephyrproject/zephyr/twister-out/frdm_k64f/tests/lib/sprintf/libraries.libc.sprintf/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mfp16-format=ieee --sysroot=/home/ztest/zephyr-sdk-0.14.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/ztest/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr/tests/lib/sprintf=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ztest/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/ztest/zephyrproject/zephyr/tests/lib/sprintf/src/main.c
/home/ztest/zephyrproject/zephyr/tests/lib/sprintf/src/main.c: In function 'test_print':
/home/ztest/zephyrproject/zephyr/tests/lib/sprintf/src/main.c:760:15: error: zero-length gnu_printf format string [-Werror=format-zero-length]
  760 |  ret = printf("");
      |               ^~
/home/ztest/zephyrproject/zephyr/tests/lib/sprintf/src/main.c: In function 'test_fprintf':
/home/ztest/zephyrproject/zephyr/tests/lib/sprintf/src/main.c:778:24: error: zero-length gnu_printf format string [-Werror=format-zero-length]
  778 |  ret = fprintf(stdout, "");
      |                        ^~
cc1: all warnings being treated as errors
[26/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj
[27/136] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/abort.c.obj
[28/136] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/strtoul.c.obj
[29/136] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/strtoll.c.obj
[30/136] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/strtoull.c.obj
[31/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/nanosleep.c.obj
[32/136] Building C object zephyr/arch/arch/arm/core/aarch32/mpu/CMakeFiles/arch__arm__core__aarch32__mpu.dir/nxp_mpu.c.obj
[33/136] Building C object zephyr/subsys/testsuite/ztest/CMakeFiles/subsys__testsuite__ztest.dir/src/ztest.c.obj
[34/136] Building C object zephyr/arch/arch/arm/core/aarch32/cortex_m/CMakeFiles/arch__arm__core__aarch32__cortex_m.dir/fault.c.obj
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
OS: Linux,
Toolchain Zephyr SDK 0.14.1
Commit SHA or Version: b161554ee862

**Additional context**
The same error here: 
tests/lib/sprintf/src/main.c:760:15                  ret = printf("");
tests/lib/sprintf/src/main.c:778:24                  ret = fprintf(stdout, "");
drivers/usb/device/usb_dc_kinetis.c:188:10    LOG_DBG("");
drivers/usb/device/usb_dc_kinetis.c:188:2      LOG_DBG("");
drivers/usb/device/usb_dc_kinetis.c:221:10    LOG_DBG("");
drivers/usb/device/usb_dc_kinetis.c:251:10    LOG_DBG("");
drivers/usb/device/usb_dc_kinetis.c:261:10    LOG_DBG("");
drivers/usb/device/usb_dc_kinetis.c:810:10    LOG_DBG("");
drivers/usb/device/usb_dc_kinetis.c:840:10    LOG_DBG("");
subsys/usb/device/class/hid/core.c:617:10     LOG_DBG("");
subsys/usb/device/class/hid/core.c:617:2       LOG_DBG("");

stephanosio, 2022-07-14:
https://github.com/zephyrproject-rtos/zephyr/blob/0e00cccb7ae2a037b7f5ce21f6c61da5d75733bb/cmake/compiler/gcc/compiler_flags.cmake#L28-L35

Indeed I can see that `-Wformat-security` and `-Wno-format-zero-length` are being omitted even though they are supported.

cc @tejlmand

stephanosio, 2022-07-14:
> `-Wformat-security` and `-Wno-format-zero-length` are being omitted even though they are supported.

This is because `check_set_compiler_property` tests the specified flags individually, and specifying `-Wformat-security` alone without `-Wformat` generates the following warning:

```
/opt/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -c test.c -Wformat-security
cc1: warning: '-Wformat-security' ignored without '-Wformat' [-Wformat-security]
```

In case of `-Wno-format-zero-length`, specifying it without `-Wformat` does not generate any warnings, but the `zephyr_check_compiler_flag` actually uses `-Wformat-zero-length` for testing the availability of this flag:

https://github.com/zephyrproject-rtos/zephyr/blob/772143248661a4c6f01a9ac063da1f703b748410/cmake/modules/extensions.cmake#L1066-L1074

And that results in the following warning:

```
$ /opt/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -c test.c -Wformat-zero-length
cc1: warning: '-Wformat-zero-length' ignored without '-Wformat' [-Wformat-zero-length]
```

mengxianglinx, 2022-07-15:
If there is no cache of -Wformat-zero-length, it will be checked at
https://github.com/zephyrproject-rtos/zephyr/blob/772143248661a4c6f01a9ac063da1f703b748410/cmake/modules/extensions.cmake#L1789

With cmake verison  3.21.0 get value 1, but cmake version version 3.23.0-rc1 get nothing. the check result depends on the version of camke.
So, the better way is to set "check_set_compiler_property" back to "set_compiler_property".

stephanosio, 2022-07-26:
https://github.com/zephyrproject-rtos/zephyr/pull/47828#issuecomment-1194951749

PR #50097 (2022-09-14): cmake: support grouping of compile options for compiler testing
Fixes: #47588

Kitware decided to fail compiler tests of single options if the
compiler reported only a warning and not an error:
https://github.com/Kitware/CMake/commit/f745e0497ee71d35fd1b3524b1636a72da76c266
which affects CMake 3.23.0 and 3.23.1

This change was later reverted in CMake >=3.23.2 and >=3.24.0
https://github.com/Kitware/CMake/commit/4941887d7defecb3016d2bd94d3a45754251ca56

Although the immediate issue is related to CMake, and has been fixed
reverted in later releases, then it still makes sense to be able to
group [... cut]
commit c2974e22 (2022-09-14): cmake: test format options together

Fixes: #47588

Use `check_set_compiler_property(... "SHELL:-Wformat -W<format-option>")`
to ensure that compiler options requiring `-Wformat` to work properly
are tested with said option.
commit c32e4a71 (2022-09-14): cmake: support grouping of compile options for compiler testing

Fixes: #47588

Kitware decided to fail compiler tests of single options if the
compiler reported only a warning and not an error:
github.com/Kitware/CMake/commit/f745e0497ee71d35fd1b3524b1636a72da76c266
which affects CMake 3.23.0 and 3.23.1

This change was later reverted in CMake >=3.23.2 and >=3.24.0
github.com/Kitware/CMake/commit/4941887d7defecb3016d2bd94d3a45754251ca56

Although the immediate issue is related to CMake, and has been fixed
reverted in later releases, then it still makes sense to be able to
group compiler optio [... cut]
closed 2022-09-14 by fabiobaltieri