zephyrproject-rtos/zephyr #47592: test: tests/drivers/gpio/gpio_basic_api failed after commit 2a8e3fe82daf43fe9e52ae25507e8e0ef72f4d59

hunterhu3000, 2022-07-11:
**Describe the bug**
tests/drivers/gpio/gpio_basic_api failed after commit 2a8e3fe82daf43fe9e52ae25507e8e0ef72f4d59

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? it8xxx2_evb
 - What have you tried to diagnose or workaround this issue? :do git bisect and found out commit 2a8e3fe82daf43fe9e52ae25507e8e0ef72f4d59 is the first bad patch
 
**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b it8xxx2_evb tests/drivers/gpio/gpio_basic_api/
2. sudo -S ~/itetool/ite -f build/zephyr/zephyr.bin

**Expected behavior**
The test can pass. 
Do git reset --hard 1f82e32518a66c19e202387f9193ae05a5333a69 and test again. You will get the expected log.

**Impact**
Got 5 test cases error in daily test

**Logs and console output**
```
***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.1.0-1849-g2a8e3fe82daf  (delayed boot 1ms) ***
Running TESTSUITE gpio_basic_test
===================================================================
START - test_gpio_port
Validate device GPIO_A
Check GPIO_A output 0 connected to input 1
E:
E:  mcause: 2, Illegal instruction
E:   mtval: 0
E:      a0: 00000000    t0: aaaaaaaa
E:      a1: 80101afc    t1: 10000000
E:      a2: 80008a14    t2: aaaaaaaa
E:      a3: 0000005a    t3: 00000020
E:      a4: 80007c24    t4: 00000001
E:      a5: 00000000    t5: aaaaaaaa
E:      a6: a0000000    t6: aaaaaaaa
E:      a7: 0000a000
E:      ra: 80002054
E:    mepc: 00000000
E: mstatus: 00001880
E:
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80101050 (test_gpio_port)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...) 0.14.1
 - Commit SHA or Version used 2a8e3fe82daf43fe9e52ae25507e8e0ef72f4d59

hunterhu3000, 2022-07-11:
@galak As you are the owner of the patch, could you please help to check first? Thanks.

ABOSTM, 2022-07-13:
@hunterhu3000 , it looks similar the issue I faced: #47671
so you can check the patch I pushed which has just been merged

hunterhu3000, 2022-07-18:
> @hunterhu3000 , it looks similar the issue I faced: #47671 so you can check the patch I pushed which has just been merged

The issue is fixed by the patch.

closed 2022-07-18 by hunterhu3000