zephyrproject-rtos/zephyr #47598: net:lwm2m: Send data longer than max coap block size (block-wise transfer)

RomainPelletant, 2022-07-11:
**Describe the bug**
lwm2m client is not able to send data if greather than max coap block size. According to [COAP](https://datatracker.ietf.org/doc/html/rfc7959#page-12), block-wise transfer shall be initiated (block2 option) for READ requests and SEND requests.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 ESP32
 - What have you tried to diagnose or workaround this issue?
Event log object read (opaque) 1000 bytes is OK. Sending 1100 bytes failed (lwm2m client returns read failed)

**Expected behavior**
Block-wise transfer with option 2.

sylvioalves, 2022-07-11:
@RomainPelletant Would be good yo understand whether this is ESP32 issue or CoAP issue. Would you provide some additional log?

rlubos, 2022-07-13:
This is not a bug but a missing feature (more or less covered by https://github.com/zephyrproject-rtos/zephyr/issues/28120). Yes, for large transfers block-wise CoAP transfer should be used, but support for this is not mandatory per LwM2M spec. The only part where block transfer is mandatory is FW update, and this we do support.

RomainPelletant, 2022-07-13:
I would implement that feature (at least for send method). I am not sure if the message (payload) shall be serialized once and then sent by blocks or if each block shall be serialized.
In the first case, it means reserving memory for the whole object serialization, and the second one only for payload size block.
I suggest for READ requests to be block-serialized and for SEND method, the object to be fully serialized once and then sent by blocks.
It seems ok for you?

RomainPelletant, 2022-07-16:
Serialization on the fly is the way we should implement it:

- Data change during transfer : can we implement it with CRC or checksum on data instead of storing timestamp?
- Support for write seems to be compromise but the SEND method shall be the prior method to do a long data write on that case : do you agree ?

SeppoTakalo, 2022-07-18:
Currently our content-formatters try to use standard libraries for encoding and decoding the content. This places a requirement for content formatting libraries. I believe currently our standard JSON or CBOR libraries don't allow partial data to be decoded or encoded.

So even when serialization on the fly sounds logical choice, it might require so much tweaks or hand writing the decoder that it is not feasible. I would much rather vote for simpler implementation and serialize the data into a memory buffer. Then it is just up to an application designed to design the system so that this limitation is worked around.

RomainPelletant, 2022-07-20:
@SeppoTakalo : according to your point of view, the whole serialization method seems to be the most reasonnable choice.
As the partial encoding/decoding feature is currently not integrated in JSON or CBOR librairies, could we open an issue/enhancement to see it implemented in a next futur and then have the tools to serialize on the fly in lwm2m client?

broglep-work, 2022-08-30:
Block 2 transfers should be supported generically thought all responses as part of the lwm2m engine (and as transparent as possible to the lwm2m engine caller)

We did already run into this missing functionality on multiple occasions (e.g. registration message getting to big because of many object instances, or big observation notify messages as reported in #28120)

I would also vote for `Serialize the whole message`. 

serialize on the fly while supporting block 2 transfer everywhere will be quite difficult to achieve and come with complexity the lwm2m engine caller will have to deal with, which imho, will not be needed in most cases. 

The usual occasion where we need block 2 transfer is when something does barely not fit into one block. So we are maybe talking about needing to support 2-3 blocks, or in extrem situation maybe up to 10 blocks. I think that order of magnitude of blocks can easily be dealt with using an intermediate buffer (that can be configurable). This will allow the lwm2m engine to serialize the whole message, see how many blocks are needed, and then start block 2 process if necessary (or abort and return proper error code if can't do it because the buffer is too small)

The case where there is a need for a big number of blocks, and where serialize on the fly would be necessary, I would rather consider as special cases and as such, this could be supported seperately with special api (in a similar way as the firmware update apis)

huelsenfruchtzwerg, 2022-09-29:
I'm also currently looking into implementing blockwise transfers, primarily considering registration(-updates) and (regular, still need to think a bit about composite-) reads/notifications for now; and I'd like some quick feedback on some thoughts I'm having:
I'm currently considering  ~3 approaches:

cache ops: serialize the operation payload at once into a single buffer when triggered (i.e. in case of read when the first read op is performed), built further responses/requests from there, discard buffer when op is done or after EXCHANGE_LIFETIME
+ ~somewhat simple
- limits max operation size to intermediate/cache buffer size
- requires n intermediate buffers, where n = max cocurrent block-wise operations. somewhere in the ballpark of NSTART (SHOULD be 1 according to coap spec) or what's currently the closest equivalent of that in the lwm2m engine I think, COAP_MAX_PENDING. Though this might actually not be enough, since that only limits the number of concurrent exchanges, but the server could e.g. perform block-wise gets of multiple resources alternating. Though arguably that'd be a violation of "Clients that want to retrieve all blocks of a resource SHOULD strive to do so without undue delay" (rfc7959 2.4).

cache blocks: "chunk" the operations into separate msgs at the time the operation is triggered, trigger sending of next block-msg when previous is acked/next block is requested
+ reuses existing msg buffers
- max op size is limited by (LWM2M_MAX_MESSAGES * LWM2M_COAP_MAX_MSG_SIZE - whatever is already in use)
- sort of breaks apart when the server requests a smaller block size then we were expecting to handle, at the very least it would need to be re-serialized/chunked and then limit max op size to MAX_MESSAGES * requested blocksize

build on the fly: modify writers to be able to only write parts of the payload, only serialize requested/required block
+ no additional buffers/caching required
+ no limit on max size of block-wise operations
+ conceptually the simplest (no special case for blockwise ops)/"most in line" with how the coap spec expects block-wise operations to work (imho)
- re-serialize request/response payload for each block (maybe cache writer context? but probably not)
- resources might change during an operation (less bad than it sounds imho, as long as we implement etag. the onous is then on the server to handle this)
- not sure how feasible it is to modify all writers to supports this, especially cbor/senml since the libraries expect to write the entire msg. Maybe have a single intermediate buffer for all operations to write into and then just copy out the block we care about? would reintroduce a max operation size again though

I started out preferring the "cache-blocks", but now I'm leaning towards either building on the fly or caching the entire op. I see the arguments for the latter, but I kind of really want the former to work. :sweat_smile: 

And some other thoughts:
block-wise writes: coap allows to handle these either atomically or on-the-fly, but on the fly seems almost impossible for lwm2m considering that the payload could contain multiple resources, and a single resource (like e.g. any single integer) could span across multiple blocks. also we'd need to be able to decode all supported formats in a "streaming" fashion.
So it's probably simplest to cache these until the whole write is received. There currently "sort of" is block-wise write support using the on-the-fly approach, but this is afaict limited to writes to single opaque resources.

etag: block-wise gets (reads, notifications) should implement etag (especially useful in the "on-the-fly") scenario. how to implement this? first thought is observer->event_timestamp, but we need/want it for all supported get payloads (so also non-observed resources/objects). next thought is some sort of hash/checksum over the payload (easier in the "cache-ops" scenario) or all resource(-instances), but that has some collision potential.

rlubos, 2022-09-30:
@hakehuang Thank you for your input and summary. I think that given the discussion above, we were leaning towards the first method, to serialize the whole message into a single, large buffer, despite of the method drawbacks. The other, on the fly serialization method seemed quite complex to implement for serveral reasons (problematic write operaion, as you've also noticed; also a need to modify writers which may not be straightoforward due to use of other underlying libraries for json or cbor).

As for the "cache blocks" approach, I don't really find it that different from on the fly serialization, with the exception to when the message is encoded/decoded. It may solve the write operation uncertainity, but still would require changes in the underlying content writers (what if a resource value is scattered between blocks?).

> etag: block-wise gets (reads, notifications) SHOULD implement etag (especially useful in the "on-the-fly") scenario. how to implement this?

We've agreed in the discussion above that hash/checksum approach might be a reasonable approach.

RomainPelletant, 2022-12-20:
@huelsenfruchtzwerg By cache ops, do you mean don't serialize directly into msg->out.out_cpkt (related data buffer is uint8_t msg_data[MAX_PACKET_SIZE]; as associated in _lwm2m_init_message_ function) and creating a new (exclusive for all?) dedicated to serialization and pull N coap packets directly into pending messages to be sent further?

LukasWoodtli, 2022-12-22:
FIY: I'm in the process of implementing block1 transfer. Currently only for registration messages. But it will need also some adjustments for other messages as well.

@RomainPelletant In my implementation I serialize the complete message first into a special large buffer. Then the buffer is split into several CoAP blocks before sending them over the wire.

RomainPelletant, 2022-12-28:
@LukasWoodtli great news and thanks for the implementation tips.

LukasWoodtli, 2023-01-10:
See #53660 for my implementation attempt. Please give me some feedback... Thanks!

GrantRolls, 2023-05-12:
@LukasWoodtli 
I am pretty green with LWM2M in general, but have got projects with blockwise transfer requirements.

The requirement is to get backlogged data in excess of ~5-10K bytes to the server in a timely manner.
Due to restrictions on the responsivity of the server to rapid resource writes, and not guaranteeing end-to-end delivery in this case, we need to use blockwise transfer (as opposed to breaking it up and writing the observed resource multiple times).

Does your implementation support on-the-fly serialization? From application side, my expected call would be something like a lwm2m_send_blockwise fn that has a next_block callback, so the caller can load the next chunk of data.

LukasWoodtli, 2023-05-15:
@GrantRolls 
My implementation does not support on-the-fly serialization. Currently the whole message body needs to be encoded (serialized) into a buffer before sending.
Your requirements of serializing on-the-fly is more complex than the current implementation.

If you want to implement such a thing I can try to give you some hints. But I'm also pretty green with LWM2M. It just happend that we need blockwise transfer so I read the spec implemented it in Zephyr.

My implementation is far from perfect. But it has two (IMHO) imprtant properties:
1. It's transparent to existing writers. So no code that produces LWM2M messages needed to be changed.
2. The whole code can be excluded by configuration. So no big encoding buffers are reserved, if blockwise transfer is not needed. And also the related code is eliminated in that case.

RomainPelletant, 2023-07-02:
@LukasWoodtli solved this, thank you!

closed 2023-07-02 by RomainPelletant