zephyrproject-rtos/zephyr #47604: New ztest fx cannot specify different setup/teardown functions for each test function within the same test suite.

smrtos, 2022-07-11:
**Is your feature request related to a problem? Please describe.**
In code like below, the `ztest_1cpu_user_unit_test()` and `ztest_user_unit_test()` macros register test functions with different setup/teardown functions. This is because the old ztest fx can specify setup and teardown functions at the "test function" level.

https://github.com/zephyrproject-rtos/zephyr/blob/ff54de3e1697e50f05cb8354f5a4ddb548107b31/tests/kernel/mutex/mutex_api/src/test_mutex_apis.c#L397-L403

https://github.com/zephyrproject-rtos/zephyr/blob/f2ae4b67b2ab6efeda116bd362cd078f2b1e5793/subsys/testsuite/ztest/include/zephyr/ztest_test_deprecated.h#L242-L243

https://github.com/zephyrproject-rtos/zephyr/blob/f2ae4b67b2ab6efeda116bd362cd078f2b1e5793/subsys/testsuite/ztest/include/zephyr/ztest_test_deprecated.h#L215-L216

But with the new ztest fx, setup/teardown can only be done at the "test suite" level as the before/after functions. Like below:

https://github.com/zephyrproject-rtos/zephyr/blob/f2ae4b67b2ab6efeda116bd362cd078f2b1e5793/subsys/testsuite/ztest/include/zephyr/ztest_test_new.h#L114

[image]

This will cause inflexibility for test suite arrangement.

**Describe the solution you'd like**
Maybe we should be able to specify tear/down functions for each individual test function.

smrtos, 2022-07-11:
I am not sure if this should be labeled as a bug or feature request. Please help remove the inappropriate one. Thx.

smrtos, 2022-07-12:
May be fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/47660

yperess, 2022-07-12:
@aaronemassey @nashif 

This was a very poor design by the original ztest and I would very much not like to see us going back in that direction. Prior to the big refactor of Ztest test suites actually didn't mean anything. All the before/after logic was bound to the test and the suite basically just provided some kind of name scoping.

Looking at other test frameworks as a model, the point of suites is to abstract away common code. If you need custom logic before/after your individual test, then that's not common code.

Basically, if you find yourself doing the same thing in multiple tests, use a suite. If you need to do things differently for just one test, just include that in the start of the test.

I'm leaving this issue open for visibility, but would like to see this marked as "working as intended".

smrtos, 2022-07-12:
I think test suite should just be a logic container of **all** the test functions targeting the same feature. That doesn't necessarily mean test functions mustn't do anything specific to themselves. 

It is possible to create yet another test suite with 1cpu specific before/after functions. But in doing that we are merely grouping test suites for the sake of the implementation of the framework, rather than by the feature we are testing.

nashif, 2022-07-12:
If we take this as a reference: [Test fixture](https://en.wikipedia.org/wiki/Test_fixture#Software)

Test fixtures can be set up three different ways: in-line, delegate, and implicit.

- In-line setup creates the test fixture in the same method as the rest of the test. While in-line setup is the simplest test fixture to create, it leads to duplication when multiple tests require the same initial data.
- Delegate setup places the test fixture in a separate standalone helper method that is accessed by multiple test methods.
- Implicit setup places the test fixture in a setup method which is used to set up multiple test methods. This differs from delegate setup in that the overall setup of multiple tests is in a single setup method where the test fixture gets created rather than each test method having its own setup procedures and linking to an external test fixture.

This issue is asking for  the delegate solution.

The 1cpu thing could be seen as a fixture that only some tests need, right now we are using this per testsuite, which is in my opinion not that great and forces the creation of additional suite just for 1cpu. Obviously we could use the in-line solution and in my early tests that is exactly what I did, but now we have many tests using the before/after of  suites to accomplish this. 

@yperess  which suites did you look at for reference?

nashif, 2022-07-12:
pytest has the above per testcase, [Robot Framework ](https://robotframework.org/robotframework/latest/RobotFrameworkUserGuide.html#test-setup-and-teardown )as well

yperess, 2022-07-12:
I assume the pytest flow you mean is:

```
import pytest

@pytest.fixture
def my_cleanup_fixture():
    # Startup code
    ...
    yield
    # Cleanup code
    ...

@pytest.mark.usefixtures('my_cleanup_fixture')
def test_with_special_cleanup():
    pass
```

Where the fixture is added to the `test_with_special_cleanup()` using the annotation `pytest.mark.usefixtures`? If so, I'm still not seeing where this is a missing feature. For example, while teardown is a bit harder, if you really didn't want to split things off into a separate suite, you could do the same via:

```
struct my_suite_fixture {
  bool is_1cpu;
};

static void *my_suite_setup(void) {
  static struct my_suite_fixture;

  my_suite_fixture.is_1cpu = false;
  return &my_suite_fixture;
}

static void my_suite_before(void *fixture) {
  ((struct my_suite_fixture *)fixture)->is_1cpu = false;
}

static void my_suite_after(void *f) {
  struct my_suite_fixture *fixture = f;

  if (fixture->is_1cpu) {
    z_test_1cpu_stop();
  }
}

ZTEST_SUITE(my_suite, NULL, my_suite_setup, my_suite_before, my_suite_after, NULL);

ZTEST_F(my_suite, test_x) {
  z_test_1cpu_start(); //< This is the same as adding function specific before
  fixture->is_1cpu = true;
}
```

As per testing frameworks:
JUnit, TestNG, NUnit, Catch2, gTest, boost, C# tests (which I forget the name of the framework), there are more but I think it's enough.

Ones that do support it:
- XCTest - Apple's testing framework for swift, to achieve this they use a DSL to extend the test function such that blocks can be created inside the test function:
```
func testMethod() throws {
  addTeardownBlock {}
}
```
- I'm sure there are others, but see below, they are all for higher level functions that support annotation and/or DSLs

Overall, the languages that do support these features are highly flexible (considerably more so than C or even C++). Adding yet another entry point to the `ZTEST*` macros makes the API messy and hard to read, Unlike the examples above that do support these features, we do not have the benefit of annotations or DSLs. Basically, the cost tradeoff from what I'm looking at is:

- It makes the API more complex
- It doesn't add a feature that can't already be done

yperess, 2022-07-12:
I've been toying with an idea that I think might solve the 1cpu issue, but it isn't as flexible as per test before/after.

We could simply provide `bool is_1cpu()` and check it in the ztest logic. If it returns true, we can call `z_test_1cpu_stop()` for the test.

nashif, 2022-07-12:
@yperess  I also do not like the idea of `ZTEST_1CPU` although I suggested it on discord the other day, this does not scale if we have something else similar to 1CPU and adding those to the API just does not make sense, I agree. But we need to have a way to deal with per testcase before/after or something else that is not 1cpu specific and more of a general solution.
We should not rush into solutions, let think about it a bit more and wait for some more results from the migration to the new API, we can then address all issues at once in a more orgnized fashion rather than hot fixes and worarounds.

yperess, 2022-07-12:
@nashif I had another thought just now, we could actually do something like XCtest, ztest could support a single linked list of cleanup functions that we can register when the test is running, in a very rough example:

```
ZTEST (s, test_x) {
  ztest_add_cleanup_function(test_x_after);
}
```
I would want to pass the fixture when available, but I haven't quite thought of how.

yperess, 2022-07-13:
> @yperess 
> 
> Registering a linked list of cleanup functions with some new API like `ztest_add_cleanup_function()` or using the `static void my_suite_before(void *fixture)` in the example above seem to make things unnecessarily complicated/twisted.
> 
> Adding 3 more `ZTEST_*_WITH_BEFORE_AFTER` macros won't make the API more messy IMO. Rather it offers a flexible and intuitive way for test writers to fine tune each test function. And we shouldn't limit ourselves to 1cpu too much. I think https://github.com/zephyrproject-rtos/zephyr/pull/47660 is generic enough for other similar scenarios.
> 
> BTW, I briefly checked the frameworks you mentioned. TestNG seems to support [before/after functions at different levels](https://automationhacks.io/2020/02/22/using-testng-setup-teardowns-with-data-provider-combinations/#the-case-for-setup-and-teardowns-via-before-and-after).
> 
> So maybe it's not that a bad idea. :)
> 
> 

Afaik, TestNG's before/after will run on every function in the class/suite. https://howtodoinjava.com/testng/testng-before-and-after-annotations/#1-testng-before-and-after-annotations explains it pretty well.

smrtos, 2022-07-13:
Yeah... it's 4 more... miscounted... 😵 

```
ZTEST_WITH_BEFORE_AFTER
ZTEST_USER_WITH_BEFORE_AFTER
ZTEST_F_WITH_BEFORE_AFTER
ZTEST_USER_F_WITH_BEFORE_AFTER
```

I agree that we should be cautious about API design.
Let's hear more from the community and see how to balance between intuitiveness and purity.

smrtos, 2022-07-22:
Current `before/after` functions may run on different threads, which may cause potential issues. See: https://github.com/zephyrproject-rtos/zephyr/issues/48147