zephyrproject-rtos/zephyr #47606: nvs_read return value is not correct

renweiwu, 2022-07-11:
**Describe the bug**
A clear and concise description of what the bug is.
when use nvs_read to read data from flash, the size returned is not correct.

**To Reproduce**
use the code in below to test nvs read/write
static void nvs_flash_init(void)
{
    const struct device *flash_dev = NULL;
    int ret;

    struct flash_pages_info info;

	flash_dev = FLASH_AREA_DEVICE(STORAGE_NODE_LABEL);
	if (!device_is_ready(flash_dev)) {
		LOG_ERR("Flash device %s is not ready", flash_dev->name);
		return;
	}
    LOG_ERR("Flash device %s", flash_dev->name);
	fs.offset = FLASH_AREA_OFFSET(STORAGE_NODE_LABEL);
	ret = flash_get_page_info_by_offs(flash_dev, fs.offset, &info);
	if (ret) {
		LOG_ERR("Unable to get page info");
		return;
	}
	fs.sector_size = 32 * 1024;
	fs.sector_count = FLASH_AREA_SIZE(STORAGE_NODE_LABEL) / fs.sector_size;

    LOG_INF("nvs_fs info:");
    LOG_INF("flash_device: %s", flash_dev->name);
    LOG_INF("sector_size: %d", fs.sector_size);
    LOG_INF("sector_count : %d", fs.sector_count);
    LOG_INF("offset : 0x%.8x", fs.offset);

	ret = nvs_init(&fs, flash_dev->name);
	if (ret) {
		LOG_ERR("Flash Init failed");
		return;
	}

    ret =  nvs_read(&fs, NVS_ID_FACE_ID_HEADER, buf, 1024);
    LOG_ERR("nvs read ret = %d", ret);

    ret =  nvs_write(&fs, NVS_ID_FACE_ID_HEADER, buf, 8192);
    LOG_ERR("nvs write ret = %d", ret);

    ret =  nvs_read(&fs, NVS_ID_FACE_ID_HEADER, buf, 2048);
    LOG_ERR("nvs read ret = %d", ret);
}

**Expected behavior**
Expected to get correct len from nvs_read.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
[00:00:06.690,000] <inf> fs_nvs: 32 Sectors of 32768 bytes
[00:00:06.690,000] <inf> fs_nvs: alloc wra: 3, 7fc8
[00:00:06.700,000] <inf> fs_nvs: data wra: 3, 6000
[00:00:06.720,000] <err> security_service: nvs read ret = **8192**
[00:00:06.840,000] <dbg> fs_nvs: nvs_add_gc_done_ate: Adding gc done ate at 7ff0
[00:00:06.840,000] <dbg> fs_nvs: nvs_flash_erase_sector: Erasing flash at f28000, len 32768
[00:00:10.690,000] <err> security_service: nvs write ret = 8192
[00:00:10.730,000] <err> security_service: nvs read ret = **8192**

**Environment (please complete the following information):**
 - OS: Linux
 - zephyr sdk 0.13.1

**Additional context**
ssize_t nvs_read_hist(struct nvs_fs *fs, uint16_t id, void *data, size_t len,
		      uint16_t cnt)
{
	int rc;
	uint32_t wlk_addr, rd_addr;
	uint16_t cnt_his;
	struct nvs_ate wlk_ate;
	size_t ate_size;

	if (!fs->ready) {
		LOG_ERR("NVS not initialized");
		return -EACCES;
	}

	ate_size = nvs_al_size(fs, sizeof(struct nvs_ate));

	if (len > (fs->sector_size - 2 * ate_size)) {
		return -EINVAL;
	}

	cnt_his = 0U;

	wlk_addr = fs->ate_wra;
	rd_addr = wlk_addr;

	while (cnt_his <= cnt) {
		rd_addr = wlk_addr;
		rc = nvs_prev_ate(fs, &wlk_addr, &wlk_ate);
		if (rc) {
			goto err;
		}
		if ((wlk_ate.id == id) &&  (nvs_ate_valid(fs, &wlk_ate))) {
			cnt_his++;
		}
		if (wlk_addr == fs->ate_wra) {
			break;
		}
	}

	if (((wlk_addr == fs->ate_wra) && (wlk_ate.id != id)) ||
	    (wlk_ate.len == 0U) || (cnt_his < cnt)) {
		return -ENOENT;
	}

	rd_addr &= ADDR_SECT_MASK;
	rd_addr += wlk_ate.offset;
	rc = nvs_flash_rd(fs, rd_addr, data, **_MIN(len, wlk_ate.len)_**);
	if (rc) {
		goto err;
	}

	return _**wlk_ate.len**_;

err:
	return rc;
}

the return value should be MIN(len, wlk_ate.len)?

Laczen, 2022-07-11:
@renweiwu, thank you for reporting this.

This is not really a bug, when `nvs_read` is called it always returns the size of the entry that can be used to determine if all bytes were read. From the documentation of `nvs_read`:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/fs/nvs.h#L123

If all bytes are read the return value will be equal to the requested read bytes. If not all bytes are read it returns how many bytes are available. If the return value would be equal to the read length there would be no way to know that there is more data inside the record.

renweiwu, 2022-07-11:
thanks for the clarification. looks not an issue.

closed 2022-07-11 by renweiwu