zephyrproject-rtos/zephyr #47607: Settings with FCB backend does not pass test on stm32h743

nvincent-vossloh, 2022-07-11:
**Describe the bug**
I tried to run the `zephyr/tests/subsys/settings/fcb/raw/` on a nucleo h743zi2 but I have a test errors, bus faults and stack overflows:

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b nucleo_h743zi -d build_test/fcb zephyr/tests/subsys/settings/fcb/raw/`
2. `west flash -d build_test/fcb`
3. See error on serial output.

**Expected behavior**
All tests should pass

**Impact**
Settings with FCB backend cannot be used on stm32h7 (NVS is not an option either: #45031)

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.1.0-1919-g8a2eed6cbdbc  ***
Running TESTSUITE test_config_fcb
===================================================================
START - test_config_empty_lookups
 PASS - test_config_empty_lookups in 0.1 seconds
===================================================================
START - test_config_insert
 PASS - test_config_insert in 0.1 seconds
===================================================================
START - test_config_getset_unknown
 PASS - test_config_getset_unknown in 0.1 seconds
===================================================================
START - test_config_getset_int
 PASS - test_config_getset_int in 0.1 seconds
===================================================================
START - test_config_getset_int64
 PASS - test_config_getset_int64 in 0.1 seconds
===================================================================
START - test_config_commit
 PASS - test_config_commit in 0.1 seconds
===================================================================
START - test_settings_check_target

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/settings/fcb/src/settings_test_fcb.c:354: test_settings_check_target: (wbs <= 16 is false)
Flash driver is not compatible with the settings fcb-backend
 FAIL - test_settings_check_target in 0.19 seconds
===================================================================
START - test_config_save_fcb_unaligned
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x934b2921
E: r0/a1:  0x00000022  r1/a2:  0x00000020  r2/a3:  0x24008870
E: r3/a4:  0x00000002 r12/ip:  0x00000732 r14/lr:  0x08007a1d
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x08007fb4
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x240002c8 (test_config_save_fcb_unaligned)
E: Halting system

```
**Environment (please complete the following information):**
 - Linux
 - zephyr-sdk-0.14.2
 - 8a2eed6cbdbc82550d3b39e8ba808102e82c3fb8

**Additional context**
I fixed the first bus fault by increasing the 
buffer used to read/write to flash from 16 bytes to 32 bytes [settings_fcb_h7.diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/9082889/settings_fcb_h7.diff.txt)
Then the first crash appears on `test_config_save_2_fcb`: 
```
START - test_config_save_2_fcb
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x240086fc
E: r0/a1:  0x5d9e865c  r1/a2:  0xc96aa6ea  r2/a3:  0xf624b53f
E: r3/a4:  0x9baa9bd7 r12/ip:  0xcf8fb769 r14/lr:  0xcc5f6b2c
E:  xpsr:  0x24008600
E: Faulting instruction address (r15/pc): 0xe1af64c5
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x240002c8 (test_config_save_2_fcb)
E: Halting system
```

Laczen, 2022-07-11:
@nvincent-vossloh, shouldn't the read/write be aligned to 64 bytes (256 bit)?

Laczen, 2022-07-11:
@nvincent-vossloh Have you tried increasing the stack size (CONFIG_ZTEST_STACK_SIZE) ?

nvincent-vossloh, 2022-07-11:
> @nvincent-vossloh Have you tried increasing the stack size (CONFIG_ZTEST_STACK_SIZE) ?

Yes, that fixed the stackoverflow. 

Now I'm down to two test errors:

```
*** Booting Zephyr OS build zephyr-v3.1.0-1919-g8a2eed6cbdbc  ***
Running TESTSUITE test_config_fcb
===================================================================
START - test_config_empty_lookups
 PASS - test_config_empty_lookups in 0.1 seconds
===================================================================
START - test_config_insert
 PASS - test_config_insert in 0.1 seconds
===================================================================
START - test_config_getset_unknown
 PASS - test_config_getset_unknown in 0.1 seconds
===================================================================
START - test_config_getset_int
 PASS - test_config_getset_int in 0.1 seconds
===================================================================
START - test_config_getset_int64
 PASS - test_config_getset_int64 in 0.1 seconds
===================================================================
START - test_config_commit
 PASS - test_config_commit in 0.1 seconds
===================================================================
START - test_settings_check_target

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/settings/fcb/src/settings_test_fcb.c:354: test_settings_check_target: (wbs <= 16 is false)
Flash driver is not compatible with the settings fcb-backend
 FAIL - test_settings_check_target in 0.19 seconds
===================================================================
START - test_config_save_fcb_unaligned
 PASS - test_config_save_fcb_unaligned in 2.365 seconds
===================================================================
START - test_config_empty_fcb
 PASS - test_config_empty_fcb in 2.889 seconds
===================================================================
START - test_config_save_1_fcb
 PASS - test_config_save_1_fcb in 0.3 seconds
===================================================================
START - test_config_delete_fcb
 PASS - test_config_delete_fcb in 0.4 seconds
===================================================================
START - test_config_insert2
 PASS - test_config_insert2 in 0.1 seconds
===================================================================
START - test_config_save_2_fcb
 PASS - test_config_save_2_fcb in 4.136 seconds
===================================================================
START - test_config_insert3
 PASS - test_config_insert3 in 0.1 seconds
===================================================================
START - test_config_save_3_fcb
 PASS - test_config_save_3_fcb in 3.126 seconds
===================================================================
START - test_config_compress_reset

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/settings/fcb/src/settings_test_compress_reset.c:82: test_config_compress_reset: (fcb_free_sector_cnt(&cf.cf_fcb) == 1 is false)
expected one free sector
 FAIL - test_config_compress_reset in 8.650 seconds
===================================================================
START - test_config_save_one_fcb
 PASS - test_config_save_one_fcb in 3.75 seconds
===================================================================
START - test_config_compress_deleted
E: Write offset not aligned on flashword length. Offset: 0x21050, flashword length: 32
E: Write range invalid. Offset: 135248, len: 32
 PASS - test_config_compress_deleted in 1.682 seconds
===================================================================
TESTSUITE test_config_fcb failed.
===================================================================
PROJECT EXECUTION FAILED
```

The first one should just be a matter of increasing the value to 32.

For the second one (test_config_compress_reset) I added a dts overlay in the test folder for nucleo h7 board to make sure that I have 2 sectors for the storage partition but I'm still having the issue.

```dts
&flash0 {
        partitions {
                storage_partition: partition@20000 {
                        label = "storage";
                        reg = < 0x20000 DT_SIZE_K(256) >;
                };  
        };  
};
```

Laczen, 2022-07-11:
@nvincent-vossloh the definition of `fcb_sectors` in settings_test_fcb.c and `fcb_small_sectors` in settings_test_compress_deleted.c do not match your erase block size.

nvincent-vossloh, 2022-07-11:
Yes, that was it, the test is very long though.
@Laczen Thanks for your help !

Laczen, 2022-07-11:
@nvincent-vossloh, no problem. I can imagine it takes a long time (it already takes a long time with 16kB sectors). It is very difficult to make this test more general (especially given the complexity of the several backends for settings). As this test also takes a very long time it doesn't seem appropriate trying to generalize the test. I would propose to keep it like it is.

If you agree could you close the issue ?

Laczen, 2022-07-11:
I guess that with this setting the test will fail in all cases because the stack size is to small. I will leave it up to @nvlsianpu to decide how to get `settings_fcb` working nicely with larger write block sizes. For me fcb (and nvs) are no longer the way to go to provide persistent storage. I am working on something else that will no longer be zephyr-only (and will be maintained outside of zephyr).

erwango, 2022-08-22:
> I guess that with this setting the test will fail in all cases because the stack size is to small. I will leave it up to @nvlsianpu to decide how to get `settings_fcb` working nicely with larger [...]

@nvlsianpu can you have a check on this ?

erwango, 2022-08-31:
@nvlsianpu ping

FRASTM, 2022-11-03:
With https://github.com/zephyrproject-rtos/zephyr/pull/47632  
don't you get another error on test_config_compress_reset ?
```
START - test_config_compress_reset                                                                                                                              
                                                                                                                                                                
    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/settings/fcb/src/settings_test_compress_reset.c:85: settings_config_fcb_test_config_compress_reset: (fc)
expected one free sector                                                                                                                                        
 FAIL - test_config_compress_reset in 6.612 seconds  
```

Laczen, 2022-11-03:
@FRASTM, did you increase the storage partition ? For settings to be fail-safe it is required to have at least 2 erase-blocks (also for fcb), it will first copy/move entries to one block before erasing the second.

closed 2023-01-18 by github-actions[bot]