zephyrproject-rtos/zephyr #47613: Samples / Tests without a testcase.yaml or sample.yaml

nashif, 2022-07-11:
**Describe the bug**

The following samples and tests do no have an meta data to allow them to be run via twister:

A few bsim tests were also found, but those are being run differently.

- tests/cmake/config_dir/foo does not have a sample.yaml or testcase.yaml
- tests/arch/xtensa_asm2 does not have a sample.yaml or testcase.yaml
- samples/boards/nrf/nrf53_sync_rtc/net does not have a sample.yaml or testcase.yaml
- samples/drivers/mbox/remote does not have a sample.yaml or testcase.yaml
- samples/drivers/fpga/fpga_controller_shell does not have a sample.yaml or testcase.yaml
- samples/drivers/fpga/fpga_controller does not have a sample.yaml or testcase.yaml
- samples/drivers/ipm/ipm_mcux/remote does not have a sample.yaml or testcase.yaml
- samples/subsys/ipc/openamp/remote does not have a sample.yaml or testcase.yaml
- samples/subsys/ipc/rpmsg_service/remote does not have a sample.yaml or testcase.yaml
- samples/subsys/ipc/ipc_service/icmsg/remote does not have a sample.yaml or testcase.yaml
- samples/subsys/ipc/ipc_service/static_vrings/remote does not have a sample.yaml or testcase.yaml
- samples/subsys/ipc/openamp_performance/remote does not have a sample.yaml or testcase.yaml
- samples/subsys/shell/devmem_load does not have a sample.yaml or testcase.yaml

**To Reproduce**

Looks a the directories above and find that there are no yaml files.

**Expected behavior**
All code in the tree needs to be tested, or at least built, to avoid bitrot. 

**Impact**
Bitrot.

fabiobaltieri, 2022-07-12:
Assigning to @nashif for coordination, guess the owners of the individual tests should take care of them.

MaureenHelm, 2022-07-18:
> @MaureenHelm did you mean to close this? Still several tests to be addressed.

It auto-closed when I merged #47810

aasthagr, 2022-07-27:
@nashif I think a yaml file is not needed for the samples listed below as they are the sub-applications being built as part of the main application using the main sample.yaml.

- samples/boards/nrf/nrf53_sync_rtc/net
- samples/subsys/ipc/openamp/remote
- samples/drivers/ipm/ipm_mcux/remote
- samples/subsys/ipc/rpmsg_service/remote

The "tests/cmake/config_dir/foo" just contains a config file (referred to from the main sample tests/cmake/config_dir) and is not an application to be built separately. 

The test case tests/arch/xtensa_asm2 doesn't exist as it was removed [#47710](https://github.com/zephyrproject-rtos/zephyr/pull/47710). 

Working to submit the fix for samples/subsys/shell/devmem_load.  Let me know if you have feedback on any of the above.

nashif, 2022-07-27:
@aasthagr  something like `samples/subsys/ipc/openamp/remote` can be built and that is the point with the sample.yml, it can be marked as build_only for now so that it does not bitrot if it is not a standalone application. The least I expect is a README explaining how this is used and how it relates to other samples so there is no confusion.

If it get complicated, please open sepearate issues with  the reamining items and assign them to respective  owners

commit 66786ed6 (2022-07-28): samples: Add missing sample.yaml

Adding the yaml files to sub-applications to avoid bitrot and
specifying the main applications with remote harness instead of
build_only.

Fixes #47613
closed 2022-07-28 by carlescufi