zephyrproject-rtos/zephyr #47640: Zephyr and caches: a difficult love story.

carlocaione, 2022-07-11:
The cache management in Zephyr is a bit problematic.

We have:
- `CONFIG_DCACHE`: honestly no idea what this is about and why this is here. I'm probably the one to blame because I probably ACK-ed this in the past but it is only used by stm32f7
- `CPU_CORTEX_M_HAS_CACHE`: this is specific for Cortex-M
- `CACHE_MANAGEMENT` (+ others): this was introduced by me and it is used to enable the cache management API, but this is not depending on anything really.

My proposal:
- Remove the redundancy between `CONFIG_DCACHE` and `CPU_CORTEX_M_HAS_CACHE`
- Introduce an arch symbol, like `ARCH_HAS_DCACHE` / `ARCH_HAS_ICACHE`
- Make `CACHE_MANAGEMENT` depending on these new symbols.

tagging @microbuilder @andyross @dcpleung @stephanosio @npitre @quic-egmc @gmarull

microbuilder, 2022-07-11:
Sounds reasonable to me.

aurel32, 2022-07-11:
> * Remove the redundancy between `CONFIG_DCACHE` and `CPU_CORTEX_M_HAS_CACHE`

Those two symbols are not redundant. `CONFIG_DCACHE` is not about having a data-cache or not but rather about enabling it. There are cases where you do not want to enable it, because DMA plays a role in the difficult love story...

carlocaione, 2022-07-12:
> Those two symbols are not redundant. CONFIG_DCACHE is not about having a data-cache or not but rather about enabling it.

Ok, that makes sense. In this case we should also have `CONFIG_ICACHE` (depending on `ARCH_HAS_{D,I}CACHE`)

> There are cases where you do not want to enable it, because DMA plays a role in the difficult love story...

Uhm, I'm not fully convinced about this. DMA is usually done on selected buffers or memory regions (that can be set in HW or MPU/MMU) disabling the whole DCACHE because you need DMA seems an overkill.

aurel32, 2022-07-12:
> > Those two symbols are not redundant. CONFIG_DCACHE is not about having a data-cache or not but rather about enabling it.
> 
> Ok, that makes sense. In this case we should also have `CONFIG_ICACHE` (depending on `ARCH_HAS_{D,I}CACHE`)

Not sure there is a real need to disable the instruction cache. It only require some management for flash driver and code in RAM, and that is already handled.

> > There are cases where you do not want to enable it, because DMA plays a role in the difficult love story...
> 
> Uhm, I'm not fully convinced about this. DMA is usually done on selected buffers or memory regions (that can be set in HW or MPU/MMU)

That was indeed the case before the async UART and SPI API have been introduced. They can be passed any RAM buffer (possibly even on the stack like in the tests), and DMA is used on them. As no alignment is required on the buffers, it means cache management operations can't be used either. At this stage, there is no other way than disable D-cache if you want to use those API.

>  disabling the whole DCACHE because you need DMA seems an overkill.

I agree with you, any idea to solve the situation are welcome.

carlocaione, 2022-07-12:
> Not sure there is a real need to disable the instruction cache. It only require some management for flash driver and code in RAM, and that is already handled.

Agree, it was just to keep it consistent with DCACHE really.

> That was indeed the case before the async UART and SPI API have been introduced. They can be passed any RAM buffer (possibly even on the stack like in the tests), and DMA is used on them. As no [...]

Ok I was not aware of that. But why you cannot make the driver pointing to a proper buffer to use maybe leveraging DT and just disable caching on that buffer (if possible) like it was done here https://github.com/zephyrproject-rtos/zephyr/commit/f6a880a2f9c for example

aurel32, 2022-07-12:
> Ok I was not aware of that. But why you cannot make the driver pointing to a proper buffer to use maybe leveraging DT and just disable caching on that buffer (if possible) like it was done here [f6a880a2f9c](https://github.com/zephyrproject-rtos/zephyr/commit/f6a880a2f9c) for example

That is when using the driver internal buffer for DMA when using the standard API. I am talking about the async API like in this example: https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/drivers/uart/uart_async_api/src/test_uart_async.c#L76 . Here the buffer used for DMA is `rx_buf[10]`, and is directly allocated on the stack.