zephyrproject-rtos/zephyr #47652: The client-server based cavstool.py might be stuck when the ROM is not start

enjiamai, 2022-07-11:
**Describe the bug**
The client-server based cavstool.py might be stuck when the ROM is not start. Most of the time this won't happens. But we need to improve this, because once this happens, the following tests results could be incorrect.  
 (Actually it could happens at any while loop in the coroutine of the loop)

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?  intel_adsp_cavs15/18/25
 - What have you tried to diagnose or workaround this issue?
 

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister  --hardware-map cavs.map  --device-testing-T  tests/ -v
 You can see test failed after tests/boards/intel_adsp/hda.

**Expected behavior**
The testing results should be correct. 

**Impact**
When it happens the following tests could not be executed or have invalid results.

**Logs and console output**
```
INFO:cavs-fw:192.168.xx.xx wrote: b'download'
INFO:cavs-fw:Receiving...
INFO:cavs-fw:size:102416, filename:b'zephyr.ri.b001f51436f8c316ab05772d20f951bc', MD5:b'85a8e15b8c0046b8bfa7c03f5e454bba'
INFO:cavs-fw:Done Receiving 102416.
INFO:cavs-fw:Firmware well received. Ready to download.
INFO:cavs-fw:192.168.xx.xx wrote: b'download'
INFO:cavs-fw:Receiving...
INFO:cavs-fw:size:106512, filename:b'zephyr.ri.4d37f973841180879d08c983f64d5733', MD5:b'0798cf7036058355035ed3c072f32a9e'
INFO:cavs-fw:Done Receiving 106512.
INFO:cavs-fw:Firmware well received. Ready to download.
INFO:cavs-fw:192.168.xx.xx wrote: b'download'
INFO:cavs-fw:Receiving...
INFO:cavs-fw:size:94224, filename:b'zephyr.ri.08e7717d76957edcb8ef7d1f4eb3b0b9', MD5:b'c8db226f936ad64c61759e18589938ca'
INFO:cavs-fw:Done Receiving 94224.
INFO:cavs-fw:Firmware well received. Ready to download.
INFO:cavs-fw:192.168.xx.xx wrote: b'download'
INFO:cavs-fw:Receiving...
INFO:cavs-fw:size:98320, filename:b'zephyr.ri.cc88632b8ebfd9a87cbf87d6d9ca7460', MD5:b'8a33cf9e13b6c614f01912d20133660f'
INFO:cavs-fw:Done Receiving 98320.
INFO:cavs-fw:Firmware well received. Ready to download.
INFO:cavs-fw:192.168.xx.xx wrote: b'download'
INFO:cavs-fw:Receiving...
INFO:cavs-fw:size:98320, filename:b'zephyr.ri.7534603d73adfc842e96be576147f07c', MD5:b'5b2307f528f616f03dce863a0e90cda9'
INFO:cavs-fw:Done Receiving 98320.
INFO:cavs-fw:Firmware well received. Ready to download.
INFO:cavs-fw:192.168.xx.xx wrote: b'download'
INFO:cavs-fw:Receiving...
INFO:cavs-fw:size:98320, filename:b'zephyr.ri.541a662c2a4beeafd300eb3c5483266f', MD5:b'20f39482b98d5735934c828e7017b9c9'
INFO:cavs-fw:Done Receiving 98320.
INFO:cavs-fw:Firmware well received. Ready to download.
INFO:cavs-fw:Ignore the interface enp2s0 which is not activated.

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version: 17251bc249cec4ff96252821a3d427132fc447e3

enjiamai, 2022-08-04:
Waiting for https://github.com/zephyrproject-rtos/zephyr/pull/46972 got merged.

marc-hb, 2022-08-29:
> The client-server based cavstool.py might be stuck when the ROM is not start.

Similar to this?
- https://github.com/thesofproject/linux/pull/3238

enjiamai, 2022-08-30:
> > The client-server based cavstool.py might be stuck when the ROM is not start.
> 
> Similar to this?
> 
> * [ASoC: SOF: log warning on not fatal booting retry thesofproject/linux#3238](https://github.com/thesofproject/linux/pull/3238)
> 
> Or rather this?
> 
> * [Intel CAVS: cavstool_client.py sporadically fails #46864 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/46864#issuecomment-1167545791)

Thanks for the comments,  @marc-hb , let me take a look first. Besides, I am working on addressing your comment on the #49366 .

enjiamai, 2022-08-30:
Hi @marc-hb , it seems like this issue is similar to https://github.com/thesofproject/linux/pull/3238, but I'm not very sure.  But I can make sure it is not the later one, thanks.

enjiamai, 2022-09-19:
Verify issue fixed. After split them and run cavstool.py in another process, it can prevent the cavs' server tool won't stuck when the ROM start failed. (design problem).
We can checkout back to a commit which a little early than this commit: b161554ee862e9508f699e93403f7b2e667c9377 , for example, bc8419e20a0a0efc396b504b99b10c3c2869f823. Then run the all tests or tests under tests/boards/intel_adsp, it can complete all tests without blocking.

PR #49366 (2022-09-16): soc: xtensa: split the cavs server and the FW loader
The PR is to spilt the cavs server and FW loader, to make the 
cavstool.py a pure FW loader and runner as it used to be. 
The reasons to do this is:

- Keep the FW loader won't be affected by the changes of
   the client-server-based HW service as possible. It also
   make the debugging more easier.
- Spawn another process for the FW loader. If the FW loader
   got stuck during executing, the service can stop it.

Fixes #47652. 
 
And change the server program's name to remote-fw-service.py
because we also try make the client-server-based FW [... cut]
closed 2022-09-19 by enjiamai