zephyrproject-rtos/zephyr #47670: STM32 QSPO/OSPI XiP support

msasdk, 2022-07-12:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, Excute in Place/memory mapped mode on QSPI/OSPI flash does not seem to be implemented on STM32.
STM32H750VB comes with 128K of internal flash which makes its use with Zephyr limited. 

**Describe the solution you'd like**
Implement XIP for STM32 on external QSPI/OSPI NOR flash.
Would be handy if it's selected per default where implemented by CONFIG_SOC_STM32x just like for instance with CONFIG_SOC_SERIES_NRF52X.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

FRASTM, 2022-08-17:
See helpful Appicaiton Notes 
 [AN5050: Octo-SPI interface on STM32 microcontrollers](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/91/dd/af/52/e1/d3/48/8e/DM00407776/files/DM00407776.pdf/jcr:content/translations/en.DM00407776.pdf)  and https://www.st.com/content/ccc/resource/technical/document/application_note/group0/b0/7e/46/a8/5e/c1/48/01/DM00227538/files/DM00227538.pdf/jcr:content/translations/en.DM00227538.pdf

joelguittet, 2023-03-18:
@FRASTM you recently modified the milestone here. Is it a feature that you are working on ?
I'm also interested by this for the STM32F746 Discovery, with the proper openocd config to flash the QSPI memory.
Thanks for the update!

MaheshaHManchegowda-eaton, 2023-05-22:
Is STM32 QSPI/OSPI XiP support available now??

erwango, 2023-05-23:
> Is STM32 QSPI/OSPI XiP support available now??

This is not available for now. This is in the plans but not for current release and next one at risk.

MaheshaHManchegowda-eaton, 2023-07-17:
@erwango product designed with STM32H730 and it needs XIP. When can we expect XIP support for STM32 in Zephyr??
Are there any alternatives at present?? If possible share work in progress code.

erwango, 2023-07-18:
@MaheshaHManchegowda-eaton Sorry, we don't have any workeable solution available for sharing right now. Please contact ST official support channel for more information.

msasdk, 2024-08-08:
I guess this issue can be closed? 
https://docs.zephyrproject.org/3.7.0/kconfig.html#!CONFIG_STM32_MEMMAP

joelguittet, 2024-08-08:
So the XIP support has been merged ? An available exemple to point in the boards maybe ?

msasdk, 2024-08-08:
samples/application_development/code_relocation_nocopy has stm32 boards examples.

closed 2024-08-08 by erwango