zephyrproject-rtos/zephyr #47682: bt_gatt_unsubscribe creates write request to CCC and then cancels it

sem1-nordic, 2022-07-12:
**Describe the bug**
bt_gatt_unsubscribe calls bt_gatt_cancel straight after gatt_write_ccc, this means that the write callback function returns BT_ATT_ERR_UNLIKELY.

This bug was introduced in commit [8d5be19](https://github.com/zephyrproject-rtos/zephyr/commit/8d5be19e0ba26c3cadc26fdf454e64ad684af96c).

Please find link to Nordic Devzone ticket highlighting the issue - https://devzone.nordicsemi.com/f/nordic-q-a/89813/bug-bt_gatt_unsubscribe-creates-write-request-to-ccc-and-then-cancels-it

jori-nordic, 2022-07-15:
There is another behavior that's mentioned in the DevZone ticket:

> Additionally, there is an edge case of bt_gatt_unsubscribe where a notification received between the write request and the write response (CCC) won't get sent to the application because it removes the subscription from sub->list before getting the write response. Since there are server implementations that send out pending notifications prior to sending the write response,

My opinion is that it is okay to miss those notifications since there was an intent from the app to unsubscribe anyway. What do you guys think @Vudentz @alwa-nordic @jhedberg ?

closed 2022-07-15 by carlescufi

alwa-nordic, 2022-07-16:
> Additionally, there is an edge case of bt_gatt_unsubscribe where a notification received between the write request and the write response (CCC) won't get sent to the application because it removes [...]

> My opinion is that it is okay to miss those notifications since there was an intent from the app to unsubscribe anyway.

It's fine, I think. This choice is a property of Host API, so we can choose. It's a future-problem for when a profile that actually cares about this pops up.