zephyrproject-rtos/zephyr #47696: tests: arch: arm: arm_thread_swap: regression since use of new ztest API

ABOSTM, 2022-07-12:
**Describe the bug**
Since use of new ztest API (PR #47525) there is a regression on some board for tests/arch/arm/arm_thread_swap:
| test | Board | Status |
| -------- | --------|--------|
|tests/arch/arm/arm_thread_swap/arch.arm.swap.common	                                                |stm32f3_disco	|failed|
|tests/arch/arm/arm_thread_swap/arch.arm.swap.common.fpu_sharing	                        |disco_l475_iot1	|failed|
|tests/arch/arm/arm_thread_swap/arch.arm.swap.common.fpu_sharing	                        |stm32f3_disco	|failed|
|tests/arch/arm/arm_thread_swap/arch.arm.swap.common.fpu_sharing.no_optimizations	|disco_l475_iot1	|error|
|tests/arch/arm/arm_thread_swap/arch.arm.swap.common.fpu_sharing.no_optimizations	|stm32f3_disco	|error|
|tests/arch/arm/arm_thread_swap/arch.arm.swap.common.no_optimizations	                |stm32f3_disco	|error|

Difference between `error `and `failed ` is that some time I got the test status after a lot of `assertion failed` sometime I only have `assertion failed` endless

**To Reproduce**
Steps to reproduce the behavior:
1. twister -N --device-testing --hardware-map ../map.yaml -c -p stm32f3_disco -T tests/arch/arm/arm_thread_swap
2.  See error

**Expected behavior**
Test passed

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-1960-g3d2b04c9a8a7  ***
Running TESTSUITE arm_thread_swap
===================================================================
START - test_arm_syscalls
Available IRQ line: 81
USR Thread: IRQ Line: 81
PASS - test_arm_syscalls in 0.5 seconds
===================================================================
START - test_arm_thread_swap
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_thread_swap/src/arm_syscalls.c:84: arm_isr_handler: ((_current->arch.mode & CONTROL_nPRIV_Msk) != 0 is false)
mode variable not set to nPRIV mode for user thread
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_thread_swap/src/arm_syscalls.c:84: arm_isr_handler: ((_current->arch.mode & CONTROL_nPRIV_Msk) != 0 is false)
mode variable not set to nPRIV mode for user thread
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_thread_swap/src/arm_syscalls.c:84: arm_isr_handler: ((_current->arch.mode & CONTROL_nPRIV_Msk) != 0 is false)
mode variable not set to nPRIV mode for user thread
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_thread_swap/src/arm_syscalls.c:84: arm_isr_handler: ((_current->arch.mode & CONTROL_nPRIV_Msk) != 0 is false)
mode variable not set to nPRIV mode for user thread
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_thread_swap/src/arm_syscalls.c:84: arm_isr_handler: ((_current->arch.mode & CONTROL_nPRIV_Msk) != 0 is false)
mode variable not set to nPRIV mode for user thread
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_thread_swap/src/arm_syscalls.c:84: arm_isr_handler: ((_current->arch.mode & CONTROL_nPRIV_Msk) != 0 is false)
....
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain Zephyr SDK
 - Commit SHA: bfd59b17a1913ee00cbe447b59cdfd514eeeb693

ABOSTM, 2022-07-12:
^^ @shaoanxli as your are the submitter of the PR causing the regression, could you help on this issue ?

fabiobaltieri, 2022-07-12:
@shaoanxli could you look into this? Think this is from #47525

shaoanxli, 2022-07-13:
> @shaoanxli could you look into this? Think this is from #47525

Of course , I think add the "NVIC_DisableIRQ(i);" command at the end of the test_arm_syscalls test or make the arm_isr_handler interrupt execute only once can fix it. I do not have the hardware platform here, can you help test the case?

And this may not be the best way, does anyone have a better idea?

PR #47717 (2022-07-13): tests: arch: fix arm_thread_swap on stm32 platform
The test fails on several stm32 platforms, due to the
correlation between test_arm_syscalls and
test_arm_thread_swap, so disable the corresponding IRQ
interrupt may fix it.

Fixes: #47696
closed 2022-07-13 by carlescufi