zephyrproject-rtos/zephyr #47700: Proposal: add runtime cacheability control interface

quic-egmc, 2022-07-12:
## Introduction

Define a dynamic cacheablity control interface in Zephyr to change cacheability attributes for memory ranges at runtime.

### Problem description

Even though core Zephyr memory management is largely static there are scenarios where some memory spaces / buffers become visible only at runtime.

One example of this is IPC with shared memory across cores.  The address, size and even lifetime of these shared memory regions is not necessarily known at compile time.  Rather when a buffer is passed at runtime, say containing ring descriptors, we need a way to dynamically map it as non-cacheable via a cacheability control interface.

Other execution environments like Linux and UEFI edk2 provide this capability.

Since control of cacheability and other memory attributes like protections (e.g. WP or XN) we could broaden the solution to control memory protections as well as cacheability.

### Proposed change

Define a new zephyr include include/zephyr/memattrs.h defining function prototypes for getting and setting memory attributes.  Memory attribute setting consists of:

1. Memory Region Base Address
2. Memory Region Length
3. Bitmask of attributes

Attributes are defined in an architecture-independent manner:

- non-cacheable for mmio register access (arm: device, x86: non-cached)
- non-cacheable but supporting write-combining and prefetching (arm: normal uncacheable, x86: write-combining)
- write through cacheable
- write back cacheable
- read protect
- write protect
- execute protect

Note that each architecture will have significantly more variants of attributes - but since the goal is to create a common cross-architecture API for use by drivers and services we must use a reasonable common subset of attributes to enable portability.  Additional architecture-specific Kconfig flags could be used to map the API attributes to specific values if needed.

The API will be available on all architecture but those architecture or implementations that do not support dynamic control will map the functions to dummy functions to ensure a consistent interface across Zephyr.

carlocaione, 2022-07-19:
tagging myself

carlocaione, 2022-07-19:
At least for MMU enabled platforms you can already do that by leveraging `arch_mem_map()` https://github.com/zephyrproject-rtos/zephyr/blob/4a64af0e9dc16cc1fb48f44a5705a5f8f00ada4e/kernel/include/kernel_arch_interface.h#L248-L284 

That is used for example by `z_phys_map()`.