zephyrproject-rtos/zephyr #47702: twister: regression : Failures are counted as errors

tpennors, 2022-07-12:
**Describe the bug**
Failures are counted as errors. Must be a regression because it was working well before

**To Reproduce**
Steps to reproduce the behavior:
1. Create a build error in `zephyr/samples/hello_world/main.c`
2. run ./scripts/twister -T samples/hello_world/ -p b_u585i_iot02a
3. See error:
[image]

**Impact**
The big problem is that as failure and error are inverted, twister return code is not 1 when failure occurs so CI does not pass in failure. As it can be seen in twister script:     
```
if results.failed or (tplan.warnings and options.warnings_as_errors):
        sys.exit(1)
```

**Environment (please complete the following information):**
 - OS: OS: ubuntu 20.04
 - Toolchain zephyr-sdk 0.13.1
 - Commit SHA or Version : v3.1.0 TAG

nashif, 2022-07-12:
backport fix available, this is 3.1 specific.

nashif, 2022-07-26:
https://github.com/zephyrproject-rtos/zephyr/pull/47703 merged

PR #47703 (2022-07-26): [Backport v3.1-branch] twister: Exit with a bad status if there were build errors
Backport 3f8d5c49b3fb54878504f50e87902eab113fc5e2 from #46550

Fixes #47702
closed 2022-07-26 by nashif