zephyrproject-rtos/zephyr #47731: JESD216 fails to read SFDP on STM32 targets

msasdk, 2022-07-13:
**Describe the bug**
when used on STM32 target with QSPI flash, /samples/drivers)/jesd216 sample outputs
```Read SFDP not supported: device not JESD216-compliant (err -134)```

 - What target platform are you using?
  STM32H750VB
 
- What have you tried to diagnose or workaround this issue?
  Modified flash_stm32_qspi.c to log the read SFDP magic word. This is the output from the log:
```
[00:00:00.000,000] <inf> flash_stm32_qspi: W25Q64JVSIQ: SFDP v 1.5 AP ff with 1 PH
[00:00:00.000,000] <inf> flash_stm32_qspi: PH0: ff00 rev 1.5: 16 DW @ 80
[00:00:00.000,000] <inf> flash_stm32_qspi: W25Q64JVSIQ: 8 MiBy flash
[00:00:00.000,000] <inf> flash_stm32_qspi: SFDP magic 50444653
```
It seems that the SFDP is read correctly.
 - Config
 ```
CONFIG_FLASH=y
CONFIG_FLASH_JESD216_API=y
CONFIG_FLASH_STM32_QSPI=y
CONFIG_SPI_NOR_SFDP_RUNTIME=y
```
- DT node
```
&quadspi {
    pinctrl-0 = <&quadspi_clk_pb2 &quadspi_bk1_ncs_pb6
                  &quadspi_bk1_io0_pd11 &quadspi_bk1_io1_pd12
                  &quadspi_bk1_io2_pe2 &quadspi_bk1_io3_pd13>;
    pinctrl-names = "default";
    status = "okay";
    w25q64jv: qspi-nor-flash@0 {
        compatible = "st,stm32-qspi-nor";
        label = "W25Q64JVSIQ";
        reg = <0>;
        qspi-max-frequency = <60000000>;
        size = <DT_SIZE_M(64)>;
        jedec-id = [ef 40 17];
        status = "okay";
    };
};`
```

**To Reproduce**
Other than the normal build procedure, I had to modify main.c with:
```
#elif DT_HAS_COMPAT_STATUS_OKAY(st_stm32_qspi_nor)
#define FLASH_NODE DT_COMPAT_GET_ANY_STATUS_OKAY(st_stm32_qspi_nor)
```

FRASTM, 2022-07-13:
The stm32 flash driver has no  API for read sfdp. (So this issue is more an enhancement than a _bug_)
That should be added to the flash_stm32_qspi_driver_api and also to flash_stm32_ospi_driver_api with : 
```
#if defined(CONFIG_FLASH_JESD216_API)
	.sfdp_read = qspi_sfdp_read,
	.read_jedec_id = qspi_read_jedec_id,
#endif /* CONFIG_FLASH_JESD216_API */
```

FRASTM, 2022-07-18:
to be completed for the octoflash once the https://github.com/zephyrproject-rtos/zephyr/issues/46740 is fixed

erwango, 2022-07-27:
This API is not supported yet. Not a bug, converting this as an enhancement.

closed 2022-08-17 by FRASTM