zephyrproject-rtos/zephyr #47732: Flash map does not fare well with MCU who do bank swaps

nvincent-vossloh, 2022-07-13:
**Describe the bug**
On devices with two flash banks, a partition allocated in the first bank before a bank swap would become located in the second bank after a bank swap. Currently there is no way to fetch the proper address of a partition depending on the bank swap state.

I've encountered this issue when using `settings_fcb` on nucleo_h743zi, when I load my application firmware in each bank, do some `settings write <id> <val>` then perform a bank swap, the firmware running on the _new_ bank do not see the settings written before the bank swap.

I have submitted a workaround for `settings_fcb` https://github.com/zephyrproject-rtos/zephyr/pull/47692, but as @de-nordic suggested it is just a workaround and the underlying issue could affect any subsystem using flash map and bank swap operations.

**To Reproduce**
Steps to reproduce the behavior:
On nucleo_h743zi, we need a patch in `settings_fcb` available on this PR: https://github.com/zephyrproject-rtos/zephyr/pull/47632 to be able to be able to support flash with larger than 16 write-size.
Here is a sample application with `settings` and `settings_shell` [settings_fcb.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/9102075/settings_fcb.tar.gz)

1. west build -b nucleo_h743zi samples/settings_fcb
2. STM32_Programmer_CLI --connect port=SWD reset=HWrst -ob SWAP_BANK=1 -e all
3. west flash
4. STM32_Programmer_CLI --connect port=SWD reset=HWrst -ob SWAP_BANK=0 -rst
5. west flash # at this point same firmware is flashed to each bank
6. from zephyr shell: `settings write foo 123`
7. STM32_Programmer_CLI --connect port=SWD reset=HWrst -ob SWAP_BANK=1 -rst
8. from zephyr shell: `settings list` yields nothing.

**Expected behavior**
The value returned by `FLASH_AREA_ID` should take into account the bank from which the firmware is running on and point to the same physical area.

**Impact**
The flash area returned does not contain the expected values due to a bank swap operation. 

**Logs and console output**
```
/* running from bank 0 */
uart:~$ settings list
uart:~$ settings write foo 123
uart:~$ settings list
foo

/* Swap to bank 1 */

uart:~$ *** Booting Zephyr OS build zephyr-v3.1.0-2094-gc0d56b070770  ***
Hello World! nucleo_h743zi
<inf> h7_fcb: pwet
uart:~$ settings list
uart:~$ 

```

**Environment (please complete the following information):**
 - OS: Linux 
 - 0.13.2, 0.14.2
 - 9ed194a00b18af81a037df9d74d63a26cb34c775 + patch mentionned above

Laczen, 2022-07-14:
@nvincent-vossloh, shouldn't this be integrated in the flash driver ? IMO the flash driver should use absolute addresses and the partition definition in the dts is a definition of absolute addresses. When executing from a "swapped" bank the driver should take this into account and correct the addresses (if required). Since it is possible to detect the bank that is executing this should not be a big change.

Automatic change of slot0 and slot1 could also use this detection to determine the correct location of the upgrade slot (although it might be challenging to support encrypted images). In general the support of bank swapping in combination with mcuboot might also require quite some rethinking (to provide all functionality) as it is not the standard setup.

@erwango, fyi.

de-nordic, 2022-07-14:
@Laczen As far as I understand bank swaps means that the MCU can swap address space mapping to ROM banks. For example when you have address space A and B ,  and banks 1 and 2, you can either have mapping A->1 and B->2 or A->1.
This means that you, at run-time, still have access to both banks but depending how MCU started, access to banks 1 and 2 will happen from different address spaces; this means that if your code is executed from bank 1 that is mapped to bank A, you still have access to bank 2, but through address space B, and opposite when code is executed from address space A, where bank 2 is mapped, it can still access contents of bank 1 but through address space B.
The DTS does not know that banks may be swapped it only defines address spaces, and the same thing happens to flash map.
This means that you man have some data stored relatively to address space A, mapped to bank 1, but when banks are swapped the data is actually visible through address space B.
I do not think that this is flash driver problem because drivers accesses flash by offset within device, but flash map "attaches meaning" to certain address spaces, physical/real one, and when banks get swapped the "meaning" is mapped to incorrect flash range.

Laczen, 2022-07-14:
@de-nordic, your understanding is correct. If the drivers would access flash by offset within a device this would not be a problem. The device definition states the flash starts from a certain (constant start address) and has a certain size. The device does not change when running from a different bank. However it seems that the driver does change the device definition based upon the bank it is running from, and this should not be done.

Laczen, 2022-07-18:
@de-nordic, could you please revisit and confirm/or reject that device definition should not be dependent on the bank the program is executing from.

If device definition should be independent from the bank the program is executing from IMO no changes should be made to flashmap (but instead to the flash driver), but I might be wrong.

nvincent-vossloh, 2022-07-18:
> In general the support of bank swapping in combination with mcuboot might also require quite some rethinking (to provide all functionality) as it is not the standard setup.

We have a patched version of mcuboot to provide firmware update through bank swapping capability on stm32h743:
https://github.com/SiemaApplications/mcuboot/commit/fffedbb57238e3bfa5d2529d727b6cc4732d9f04
Since this patch is really device specific I obviously could not upstream it, I'm sharing it here in case someone is interested.

de-nordic, 2022-07-18:
> @de-nordic, could you please revisit and confirm/or reject that device definition should not be dependent on the bank the program is executing from.
> 

Yes, you are right but I do not think that device definition changes here.
The problem is what software "believes"; software compiled for bank 0 assumes that it will be executed from bank 0, when you swap banks and run it from bank 1, it still believes it is in bank 0, even though this is different instance of a software.
If we decide in driver to correct address then doing  `flash_write(dev, 0, )` would  access different flash address depending on bank we are executing from, I do not think this is correct.
Maybe we should define banks as different flash devices, that is what you mean by changing driver? 

> If device definition should be independent from the bank the program is executing from IMO no changes should be made to flashmap (but instead to the flash driver), but I might be wrong.

If we would define banks as different flash devices then flash map change would not be needed. Otherwise the problem is, as I see it, that we have "soft" definition of address ranges that are based, during compilation, on something we consider that will be bank 0. When you swap banks, then that "soft" definition still points, by name, to the same address but in "wrong" bank.

Laczen, 2022-07-18:
> If we decide in driver to correct address then doing `flash_write(dev, 0, )` would access different flash address depending on bank we are executing from, I do not think this is correct.
> 

This is exactly the problem. It seems the current driver is accessing different flash address depending on the bank it is executing from. If it would respect this there wouldn't be a problem. That is why I think a driver correction is required.

nvincent-vossloh, 2022-07-20:
@Laczen I'm not sure what you mean by _absolute address_. 

From my understanding, the `get_sector()` routine takes an address between 0 and `FLASH_SIZE` (therefore I do not consider it an absolute address, in my mind an absolute address would start at 0x08000000, at least on stm32). And depending on the offset requested and the swap state it provides the register addresses for CR and SR.
(DUAL_BANK is defined).

If I were to reword the issue I would say that:
- I cannot specify the storage partition for the `settings_fcb` subsys in term of bank and offset in that bank, because I want my storage partition to always be at the same physical area no matter the bank swap state.
- Furthermore, I want to be able to define the fw partitions (slot0/1) as _current execution bank@offset X_ for slot 0 and _other bank@offset X_ for slot 1, whatever the bank swap state is.

Laczen, 2022-07-20:
> @Laczen I'm not sure what you mean by _absolute address_.

By absolute address I'm talking in the context of bank switching and the device addressing. No matter what bank the firmware is executing from, the flash device is always using the same addressing (as if the firmware was executing from bank 0). But yes for the stm32 an offset of 0x08000000 needs to be added to get fully absolute addresses.

> If I were to reword the issue I would say that:
> 
> * I cannot specify the storage partition for the `settings_fcb` subsys in term of bank and offset in that bank, because I want my storage partition to always be at the same physical area no matter the bank swap state.

The answer to this is: you shouldn't have any problems, the storage partition should always be at the same address. If it isn't there is something wrong and the fix shouldn't be in flash_area.

> * Furthermore, I want to be able to define the fw partitions (slot0/1) as _current execution bank@offset X_ for slot 0 and _other bank@offset X_ for slot 1, whatever the bank swap state is.

This is something completely different. As this functionality is probably limited to the images and mcuboot this should not be made a part of flash_area but should be a part of mcuboot/mcumgr (and some config option to enable bank swap). You probably don't even want to use the names slot0/slot1 but instead executing/other.

nvincent-vossloh, 2022-09-19:
As far as I am concerned I think there is something to do regarding flash addressing and bank swapping.
@erwango, could you give us your opinion on this matter please ?

erwango, 2022-09-19:
> As far as I am concerned I think there is something to do regarding flash addressing and bank swapping.
@erwango, could you give us your opinion on this matter please ?

According to this comment:
> @nvincent-vossloh, @de-nordic, looking at the flash driver get_sector routine: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/flash_stm32h7x.c#L186 I would conclude that it is the intention of the driver to use the (flash driver input) offset as an absolute address and I don't understand why the problem appears (unless DUAL_BANK is undefined).

I'd say that since the driver is able to identify the sector in use, it would be interesting to understand at which step this information is lost. If this is known, it will be easier to take the right decision.

Laczen, 2022-09-19:
I have a feeling that it might be lost in the flash_page_support API.

nvincent-vossloh, 2022-09-19:
@Laczen did you made a typo in the command I should run ?
```
uart:~$ List flash areas
List: command not found
```

I tried some variants of your command without success.

nvincent-vossloh, 2022-09-19:
[settings_fcb.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/9599243/settings_fcb.tar.gz) updated sample application to demonstrate the issue, tested on commit e61aec3e4c012e015008918a8a368701e35d2baa with this patch https://github.com/nvincent-vossloh/zephyr/commit/9d33ebdc629c0d182ef72d9670de90c33d93b713 cherry-picked

nvincent-vossloh, 2022-09-19:
**From partition 0**
```
uart:~$ flash_map list
ID | Device | Device Name       |   Offset   |   Size
-------------------------------------------------------
0    0        flash-controller@52002000  0x0          0x80000       
1    0        flash-controller@52002000  0xc0000      0x40000       
2    0        flash-controller@52002000  0x1c0000     0x40000
```

**From partition 1**
```
uart:~$ flash_map list
ID | Device | Device Name       |   Offset   |   Size
-------------------------------------------------------
0    0        flash-controller@52002000  0x0          0x80000       
1    0        flash-controller@52002000  0xc0000      0x40000       
2    0        flash-controller@52002000  0x1c0000     0x40000
```

Here is partitions node in my dts:
```dts
                                partitions {
                                        compatible = "fixed-partitions";
                                        #address-cells = < 0x1 >;
                                        #size-cells = < 0x1 >;
                                        slot0_partition: partition@0 {
                                                label = "image-0";
                                                reg = < 0x0 0x80000 >;
                                        };   
                                        storage_partition: partition@c0000 {
                                                label = "storage";
                                                reg = < 0xc0000 0x40000 >;
                                        };   
                                        storage_mirror: partition@1c0000 {
                                                label = "storage-mirror";
                                                reg = < 0x1c0000 0x40000 >;
                                        };   
                                };   
```

The `storage_mirror` partitions is here to be able to use the workaround from #47692 (see code under `if 0` in `main.c` in the attached tar.gz above).

Laczen, 2022-09-19:
@nvincent-vossloh, could you remove the mirror partition during the test, I think this hides the problem.

nvincent-vossloh, 2022-09-19:
**running from partition 0**
```
uart:~$ flash_map list
ID | Device | Device Name       |   Offset   |   Size
-------------------------------------------------------
0    0        flash-controller@52002000  0x0          0x80000       
1    0        flash-controller@52002000  0xc0000      0x40000
```

**running from partition 1**
```
uart:~$ flash_map list
ID | Device | Device Name       |   Offset   |   Size
-------------------------------------------------------
0    0        flash-controller@52002000  0x0          0x80000       
1    0        flash-controller@52002000  0xc0000      0x40000
```

Laczen, 2022-09-19:
Ok, so flash_map is always returning the "absolute" address for the partition. But when this is used in the flash driver the driver maps this to a different address because it is running from a different partition. Maybe the code in the flash driver should be changes not to take into account the running partition.

Laczen, 2022-09-19:
@erwango, the driver contains:
```
bank_swap = (READ_BIT(FLASH->OPTCR, FLASH_OPTCR_SWAP_BANK)
			== FLASH_OPTCR_SWAP_BANK);`
```
is this correct ? Or should it be:
```
bank_swap = (READ_BIT(FLASH->OPTCR, FLASH_OPTCR_SWAP_BANK)
			== 1);
```

nvincent-vossloh, 2022-09-26:
As far as I am concerned, the dual bank support works correctly: 
Whatever the bank swap state is, the same firmware works (no bank specific firmware). 
When I request an address from the lower range, the physical areas targeted is the one from the _current_ bank, and when requesting an address in the higher range it is _the other_ bank which is targeted.
Since I'm trying to use some flash sectors for a `settings` module, I want those sectors to be addressed whatever the bank I'm running from. Currently, in order to achieve this I have the workaround from #47692

Laczen, 2022-09-26:
As far as I'm concerned there are two distinct use cases of bank swap:

A. A real bank swap support in which case everything is separated, the images and the storage, and both banks are independent of each other.
B. A bank swap of only the images and shared storage.

For case A the dts should only describe half of the flash area (I think this is the supported case at the moment). For case B this would require describing the whole flash area and it would be needed to specify what the meaning of the addresses in the dts is. IMO the addresses should mean the addresses as if the image is running from bank 0 (as is the intention of flash_map).

Laczen, 2022-09-26:
@nvincent-vossloh, BTW have you tried disabling DUAL_BANK, I am not sure if this is really needed for something other than flash.

nvincent-vossloh, 2022-09-26:
I want to be sure of the differences in your cases.
In your case A., do you mean that each bank have a version of the image and the storage ? an image only use the storage which resides in the same bank (I do not fall in that use case).

In case B, each bank have an image. The storage is located in **one of** the banks and used by  images from both banks (I am in that use case).

If the above is true, for case A., we could describe only half of the flash area if the image is not responsible for the update (in which case the other half needs to be described in order to write to the second half of the flash from the first half).

For case B, what do you mean by _"it would be needed to specify what the meaning of the addresses in the dts is"_, like a node properties of a partition to tell if the address is relative to current bank or bank 0 ?

> BTW have you tried disabling DUAL_BANK

I have not tried to disable `DUAL_BANK` in the sample application mentioned above. However, I need the `DUAL_BANK` flag in my final application since it is responsible for the update and therefore needs to write to the other bank.
I'll give it a try on the sample application.

Laczen, 2022-09-26:
> For case B, what do you mean by _"it would be needed to specify what the meaning of the addresses in the dts is"_, like a node properties of a partition to tell if the address is relative to current bank or bank 0 ?

I mean it is needed to agree on the meaning, and if different functionality is required it is needed to agree on the way to describe this and to make sure that also case A remains supported.

nvincent-vossloh, 2022-10-20:
Just a quick update to let you know that I have been assigned to other task and could not investigate this issue further.
I intend to try disabling `DUAL_BANK` when I'll be available for this project.

Sorry for the lack of information.

Laczen, 2022-10-20:
> Just a quick update to let you know that I have been assigned to other task and could not investigate this issue further. I intend to try disabling `DUAL_BANK` when I'll be available for this project.
> 
> Sorry for the lack of information.

No problem, I was wondering what happened. I don't have such a board, so I can't test it.

Laczen, 2022-10-27:
@nvincent-vossloh, I can confirm the issue (a nice wind has blown a st board in my direction). The problem is that in dual-bank mode the flash is divided into 2 blocks, the current execution block and the other block. The addressing for read and write is also divided into 2 blocks: as long as the address is below FLASH_SIZE/2 it is in the current (execution) block, if the address is above FLASH_SIZE/2 it is in the other block. The addressing in the flash interface is not absolute in dual bank operation.
Disabling DUAL_BANK in the flash interface will not solve the problem as the reads are not using the DUAL_BANK information.
I don't see an elegant solution in devicetree to define partitions in a setup with dual-bank support, the definitions could be treated as absolute, but this would require changing the flash reading/writing of all devices with dual-bank support. A different way would be to add two partition descriptions (similar to your solution), but that would be needed for all partitions. A addition of a routine to flash_map (or maybe better the flash_api) to get the current execution slot would allow the selection of the correct partition array.

Laczen, 2022-10-28:
Maybe there is a simple solution: What if we add a Kconfig option `ENABLE_DUALBANK_ADDRESS_CORRECTION` to the flash driver configuration. If this is selected all addresses in the dts are considered as absolute (and a correction is applied by the driver). @erwango, @de-nordic what is your opinion ?

nvincent-vossloh, 2022-10-28:
Thanks for your feedback and your work on this issue.
I like this idea as it does not require to modify right away all flash drivers with dual bank support. The dts is kept simple (single partition definition).
If I'm reassigned to the our internal zephyr based project I think I may be able to implement this solution (for stm32h7 devices). I'll let you know when I may be able to start working on this.

Thanks again for your support.

de-nordic, 2022-10-28:
> Maybe there is a simple solution: What if we add a Kconfig option `ENABLE_DUALBANK_ADDRESS_CORRECTION` to the flash driver configuration. If this is selected all addresses in the dts are considered as absolute (and a correction is applied by the driver). @erwango, @de-nordic what is your opinion ?

We can probably try that.

Laczen, 2022-10-28:
@FRASTM, fyi

nvincent-vossloh, 2023-01-02:
I'm still working on non-zephyr projects and do not have currently the time to implement this feature.
The stale label can be removed.

closed 2023-03-20 by github-actions[bot]