zephyrproject-rtos/zephyr #47751: soc/arm/common/cortex_m doesn't work for out-of-tree socs

quic-egmc, 2022-07-13:
When an out of tree SOC is defined the Zephyr SOC_DIR points to a different location:

https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/soc.cmake#L59

This means the line in the top-level CMakeLists.txt that looks for a ${SOC_DIR}/${ARCH}/CMakeLists.txt fails to find it:

https://github.com/zephyrproject-rtos/zephyr/blob/main/CMakeLists.txt#L566

and as a result the processing of a "common" cortex-M location is never reached:

https://github.com/zephyrproject-rtos/zephyr/blame/main/soc/arm/CMakeLists.txt#L3

It looks like the ${SOC_DIR}/${ARCH}/CMakeLists.txt construct is at odds with out-of-tree SOC definitions as any in-tree common definitions are skipped.

carlescufi, 2022-07-13:
@quic-egmc please use backticks for monospace

tejlmand, 2022-08-10:
> It looks like the SOCDIR/{ARCH}/CMakeLists.txt construct is at odds with out-of-tree SOC definitions as any in-tree common definitions are skipped.

This is actually by design.
If you are providing your own SoC, we cannot know if you want to source the common SoC code or not, even if your SoC is arm based.

If your SoC wants to reuse any common code with other arm based SoC, you should add something similar to the following line to your `${SOC_DIR}/${ARCH}/CMakeLists.txt` file:
```
add_subdirectory(${ZEPHYR_BASE}/soc/arm/common/cortex_m soc/common)
```

closed 2022-08-10 by tejlmand

quic-egmc, 2022-08-10:
But this common code is architectural - not just something that some SOCs have sometimes - if you have an ARM M-architecture core you absolutely need these definitions.

So perhaps the real problem is that we are identifying M-architecture support code in the wrong way - by creating an SOC subdirectory when this should have been in arch all along.

tejlmand, 2022-08-12:
> So perhaps the real problem is that we are identifying M-architecture support code in the wrong way - by creating an SOC subdirectory when this should have been in arch all along.

In that case, feel free to open an issue that this code belongs in arch, not SoC.
Or even better, a PR relocating the code 😉 

I was simply replying to the raised issue of not loading common code from SoC folder, cause such common code is for the SoC to load, regardless of whether the SoC is in-tree or out-of-tree.

quic-egmc, 2022-08-12:
Thanks, let's get @microbuilder's thoughts on relocating this!

microbuilder, 2022-08-18:
@quic-egmc Happy to review a PR or proposal on relocating.