zephyrproject-rtos/zephyr #47817: qemu_xtensa samples/modules/nanopb/sample.modules.nanopb fails with protobuf > 3.19.0

cjgraydon, 2022-07-14:
[logfiles.zip](https://github.com/zephyrproject-rtos/zephyr/files/9116569/logfiles.zip)
[logfiles.zip](https://github.com/zephyrproject-rtos/zephyr/files/9116583/logfiles.zip)

qemu_xtensa samples/modules/nanopb/sample.modules.nanopb fails with protobuf > 3.19.0 in twister run

I've reproduced this on both Debian 10 and Ubuntu 20.04 with SDK 0.14.2, with rev 6b33c03892da7609814d20d039bf60793c6cc149

To reproduce....

This works:

Name: protobuf
Version: 3.19.0

Run:
scripts/twister -M -W -x=USE_CCACHE=0 -N --inline-logs -v --integration --force-color --retry-failed 3 --retry-interval 60 -p qemu_xtensa

Results:
INFO    - 118 of 2081 test configurations passed (100.00%), 0 failed, 1963 skipped with 0 warnings in 128.07 seconds
INFO    - In total 42233 test cases were executed, 673 skipped on 69 out of total 473 platforms (14.59%)
INFO    - 111 test configurations executed on platforms, 7 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /srv/build/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /srv/build/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /srv/build/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

This doesn't:

Name: protobuf
Version: 4.21.2

Run:
scripts/twister -M -W -x=USE_CCACHE=0 -N --inline-logs -v --integration --force-color --retry-failed 3 --retry-interval 60 -p qemu_xtensa

Results: 
INFO    - 117 of 2081 test configurations passed (99.15%), 2 failed, 1963 skipped with 0 warnings in 189.80 seconds
INFO    - In total 42233 test cases were executed, 673 skipped on 69 out of total 473 platforms (14.59%)
INFO    - 110 test configurations executed on platforms, 8 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /srv/build/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /srv/build/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /srv/build/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

Note that the two failures is really just one. Possibly related to https://github.com/zephyrproject-rtos/zephyr/issues/47702. The single failure is shown as a "failure" in twister.xml and as an "error" in twister_report.xml. 

From build.log:
FAILED: src/simple.pb.c src/simple.pb.h /srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/src/simple.pb.c /srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/src/simple.pb.h 
cd /srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb && /usr/bin/protoc -I/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/nanopb/generator -I/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/nanopb/generator/proto -I/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb -I/srv/build/zephyrproject/zephyr/samples/modules/nanopb -I. --plugin=protoc-gen-nanopb=/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/nanopb/generator/protoc-gen-nanopb "--nanopb_opt= -I/srv/build/zephyrproject/zephyr/samples/modules/nanopb/src -I/srv/build/zephyrproject/zephyr/samples/modules/nanopb" --nanopb_out=/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb /srv/build/zephyrproject/zephyr/samples/modules/nanopb/src/simple.proto
Traceback (most recent call last):
  File "/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/nanopb/generator/nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ModuleNotFoundError: No module named '__main__.proto'; '__main__' is not a package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/nanopb/generator/nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/nanopb/generator/proto/__init__.py", line 23, in <module>
    from . import nanopb_pb2
  File "/srv/build/zephyrproject/zephyr/twister-out/qemu_xtensa/samples/modules/nanopb/sample.modules.nanopb/nanopb/generator/proto/nanopb_pb2.py", line 37, in <module>
    type=None),
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/descriptor.py", line 755, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

**Expected behavior and impact**
Protobuf greater than 0.13.9 (i.e. 4.21.2) is installed by default by pip3 when installing requirements*.txt, which obviously breaks this sample and forces user to downgrade the version.  

This also has implications to automated twister runs in CI and docker environments. i.e. It breaks the build and forces a downgrade or pinned version in build environment.

Build.log and twister.xml attached. 

[logfiles.zip](https://github.com/zephyrproject-rtos/zephyr/files/9116586/logfiles.zip)

pdgendt, 2022-07-20:
I doubt this has anything to do with `xtensa`. I've had the same discussion in PR #46825, where eventually the installed python package is `grpcio-tools` which depends on `protobuf v3.20.1`.

Can you try with a reinstall of the requirements?
```sh
pip install --upgrade --force-reinstall -r scripts/requirements.txt
```

This limitation is because of a breaking upgrade from google as described [here](https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates).

cjgraydon, 2022-07-22:
> I doubt this has anything to do with `xtensa`. I've had the same discussion in PR #46825, where eventually the installed python package is `grpcio-tools` which depends on `protobuf v3.20.1`.
> 
> Can you try with a reinstall of the requirements? 
> 
> ```shell
> pip install --upgrade --force-reinstall -r scripts/requirements.txt
> ```
> 
> This limitation is because of a breaking upgrade from google as described [here](https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates).

I'll give that a go with the force reinstall and check on the version of grpcio-tools and see what I get and report back.

PR #55908 (2023-03-20): west.yml: Update nanopb to 0.4.7
Update `nanopb` to the latest released version 0.4.7.

Fixes #47817
closed 2023-03-20 by carlescufi