zephyrproject-rtos/zephyr #47822: Stack Overflow when calling spi at an interrupt on STM32l4

zdtyuiop4444, 2022-07-15:
**Describe the bug**
I'm testing the dw3000 driver I wrote, it works all fine when I use esp32 but it cause a MPU FAULT when I switch to my stm32l431cct6 board.

I have tried to fix this problem by setting
CONFIG_MAIN_STACK_SIZE=1024
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=8192
But it still crash when my irq_work_handler try to read the statue of dw3000 chip through spi, it stuck at zephyrproject/zephyr/drivers/spi/spi_ll_stm32.c:453

**Logs and console output**
..12:08:22:545 [00:00:02.993,000] .[1;31m<err> os: ***** MPU FAULT *****.[0m
..12:08:22:553 [00:00:02.993,000] .[1;31m<err> os:   Data Access Violation.[0m
..12:08:22:554 [00:00:02.993,000] .[1;31m<err> os:   MMFAR Address: 0x20002b18.[0m
..12:08:22:555 [00:00:02.993,000] .[1;31m<err> os: r0/a1:  0x00000000  r1/a2:  0x00000008  r2/a3:  0x00000010.[0m
..12:08:22:555 [00:00:02.993,000] .[1;31m<err> os: r3/a4:  0x00000018 r12/ip:  0x080111c4 r14/lr:  0x0800b487.[0m
..12:08:22:555 [00:00:02.993,000] .[1;31m<err> os:  xpsr:  0x61000000.[0m
..12:08:22:556 [00:00:02.993,000] .[1;31m<err> os: Faulting instruction address (r15/pc): 0x0800b4a2.[0m
..12:08:22:556 [00:00:02.993,000] .[1;31m<err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0.[0m
..12:08:22:556 [00:00:02.993,000] .[1;31m<err> os: Current thread: 0x20001150 (unknown).[0m
..12:08:22:556 [00:00:03.001,000] .[1;31m<err> os: Halting system.[0m

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - 3.0.99

**Additional context**
[zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/9117782/zephyr.zip)

henrikbrixandersen, 2022-07-19:
Looks like you are attempting to use the SPI API from an interrupt service routine? You cannot do that, as a SPI transaction will sleep - and you cannot sleep from within an ISR.

erwango, 2022-07-25:
@zdtyuiop4444 Can you check comment from @henrikbrixandersen and get back to us ?

zdtyuiop4444, 2022-07-26:
> Looks like you are attempting to use the SPI API from an interrupt service routine? You cannot do that, as a SPI transaction will sleep - and you cannot sleep from within an ISR.

Thanks for your reply. 
So that's a hardware limitation, I cannot use interrupt to check the statue of dw3000.

closed 2022-07-26 by zdtyuiop4444