zephyrproject-rtos/zephyr #47830: Intel CAVS: Build failure due to 47713 commit.

shuai1x, 2022-07-15:
**Describe the bug**
On intel_adsp_cavs15/18/25, /tests/misc/print_format, /tests/lib/newlib, /tests/subsys/cpp/cxx, /tests/subsys/cpp/libcxx, /tests/subsys/logging and /tests/posix build failure.

Please also mention any information which could help others to understand
the problem you're facing:
 - intel_adsp_cavs15, intel_adsp_cavs18, intel_adsp_cavs25
 - build failure due to 47713 commit. 

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/misc/print_format/ --no-skipped-report -vv
2. See error

**Logs and console output**
```
[67/143] Linking C static library zephyr/lib/posix/liblib__posix.a
[68/143] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/rimage_modules.c.obj
[69/143] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/sram.c.obj
FAILED: zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/sram.c.obj 
ccache /home/ztest/zephyr-sdk-0.14.1/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/include -I/home/ztest/zephyrproject/zephyr/drivers -I/home/ztest/zephyrproject/zephyr/include/zephyr -I/home/ztest/zephyrproject/zephyr/include -I/home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/include/generated -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/cavs_v25 -I/home/ztest/zephyrproject/zephyr/lib/libc/newlib/include -I/home/ztest/zephyrproject/modules/hal/xtensa/include -I/home/ztest/zephyrproject/modules/hal/xtensa/zephyr/soc/intel_cavs_25 -Os -imacros /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always --sysroot=/home/ztest/zephyr-sdk-0.14.1/xtensa-intel_s1000_zephyr-elf/xtensa-intel_s1000_zephyr-elf -imacros /home/ztest/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr/tests/misc/print_format=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ztest/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/sram.c.obj -MF zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/sram.c.obj.d -o zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/sram.c.obj -c /home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/sram.c
/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/sram.c:39:13: error: expected ';' before 'void'
   39 | static __imr void hp_sram_pm_banks(uint32_t banks)
      |             ^~~~~
      |             ;
/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/sram.c:109:6: error: expected ';' before 'void'
  109 | __imr void hp_sram_init(uint32_t memory_size)
      |      ^~~~~
      |      ;
/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/sram.c:124:6: error: expected ';' before 'void'
  124 | __imr void lp_sram_init(void)
      |      ^~~~~
      |      ;
[70/143] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/boot.c.obj
FAILED: zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/boot.c.obj 
ccache /home/ztest/zephyr-sdk-0.14.1/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/include -I/home/ztest/zephyrproject/zephyr/drivers -I/home/ztest/zephyrproject/zephyr/include/zephyr -I/home/ztest/zephyrproject/zephyr/include -I/home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/include/generated -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/cavs_v25 -I/home/ztest/zephyrproject/zephyr/lib/libc/newlib/include -I/home/ztest/zephyrproject/modules/hal/xtensa/include -I/home/ztest/zephyrproject/modules/hal/xtensa/zephyr/soc/intel_cavs_25 -Os -imacros /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always --sysroot=/home/ztest/zephyr-sdk-0.14.1/xtensa-intel_s1000_zephyr-elf/xtensa-intel_s1000_zephyr-elf -imacros /home/ztest/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr/tests/misc/print_format=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ztest/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/boot.c.obj -MF zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/boot.c.obj.d -o zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/boot.c.obj -c /home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/boot.c
/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/boot.c:76:13: error: expected ';' before 'void'
   76 | static __imr void parse_module(struct sof_man_fw_header *hdr,
      |             ^~~~~
      |             ;
/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/boot.c:110:6: error: expected ';' before 'void'
  110 | __imr void parse_manifest(void)
      |      ^~~~~
      |      ;
/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/boot.c:130:6: error: expected ';' before 'void'
  130 | __imr void win_setup(void)
      |      ^~~~~
      |      ;
/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/boot.c:151:6: error: expected ';' before 'void'
  151 | __imr void boot_core0(void)
      |      ^~~~~
      |      ;
[71/143] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/ipc.c.obj
[72/143] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/multiprocessing.c.obj
[73/143] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/multiprocessing_cavs.c.obj
[74/143] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/xtensa-asm2.c.obj
ninja: build stopped: subcommand failed.
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/ztest/zephyrproject/zephyr/tests/misc/print_format
-- Cache files will be written to: /home/ztest/.cache/zephyr
-- Zephyr version: 3.1.99 (/home/ztest/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: intel_adsp_cavs25
-- Found host-tools: zephyr 0.14.1 (/home/ztest/zephyr-sdk-0.14.1)
-- Found dtc: /home/ztest/zephyr-sdk-0.14.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.1 (/home/ztest/zephyr-sdk-0.14.1)
-- Found BOARD.dts: /home/ztest/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs25/intel_adsp_cavs25.dts
-- Generated zephyr.dts: /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/include/generated/device_extern.h
[... cut ...]
No change to Kconfig header in '/home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/misc/print_format/printk.format_newlib
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 14.1
 - Commit SHA: 76f990af8a56638d12a1479493cb66dbb89341ad

enjiamai, 2022-07-16:
Seems like due to this commit:
https://github.com/zephyrproject-rtos/zephyr/pull/47713/commits/826c7205f184bf2883182b17e0f52eb3ba80607e

Follow tests have build error on CAVS boards:
tests/lib/mem_alloc
tests/lib/newlib
tests/subsys/logging
tests/posix/common

Try to submit https://github.com/zephyrproject-rtos/zephyr/pull/47897 to fix it, but I'm not sure is this a correct fix.
Hi @nashif , could you please help take look at this issue? Thanks.

closed 2022-07-16 by nashif

smrtos, 2022-07-17:
It's strange that west build didn't see this issue. But twister saw this.

PR #47897 (2022-07-16): soc: xtensa: include linker/section.h to fix the build error
Some tests was built failed due to cannot find the __imr macro:
Try to fix it by including the linker/section.h in soc.h.

Fixes #47830.
commit 9b8726c1 (2022-07-16): soc: xtensa: include linker/section.h to fix the build error

Some tests was built failed due to cannot find the __imr macro:
Try to fix it by including the linker/section.h in soc.h.

Fixes #47830.