zephyrproject-rtos/zephyr #47852: samples: boards: nrf: s2ram No valid output on console

nordic-piks, 2022-07-15:
**Describe the bug**
The test/sample  samples/boards/nrf/s2ram/sample.boards.nrf.s2ram fails on-target tests - no valid console output.

Observed/executed only for nrf5340dk_nrf5340_cpuapp.

When running sample on device there is no console output other than:
ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-2138-g700de003e849  ***

In this configuration (harness not specified), there is ztest output expected.
Looking at sample documentation seems like only some starting string "S2RAM demo" can be expected, but this is not visible.
Maybe sample should expect this text? (harness console)

Documentation describes need of human interaction with the board to check proper behavior,
maybe sample should be marked as build only?

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/boards/nrf/s2ram -p nrf5340dk_nrf5340_cpuapp --device-testing --device-serial /dev/ttyACM2 -v --inline-logs
4. See only zephyr starting

**Expected behavior**
Valid output on console or build only

**Impact**
Minimal

**Logs and console output**
ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-2138-g700de003e849  ***

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: zephyr-v3.1.0-2138-g700de003e849

carlocaione, 2022-07-15:
> maybe sample should be marked as build only?

that or `harness: remote`. I'll take care of this.

closed 2022-07-22 by fabiobaltieri

nordic-piks, 2022-07-26:
Issue no longer visible.

PR #48019 (2022-07-22): sample: s2ram: Use harness remote
The sample requires a manual user interaction. Make CI skipping that by
requiring a remote harness (the user).

Fixes: #47852