zephyrproject-rtos/zephyr #47856: lib: posix: allow eventfd on ARCH_POSIX platform

TomzBench, 2022-07-15:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to be able to use `eventfd` on `native_posix_64` platform for testing purposes.

**Describe the solution you'd like**
Currently - eventfd is treated as a general library feature and maintained under `lib/os` ... And this feature depends on `!ARCH_POSIX`.  However - other socket API do not have this same limitation.  ...In the context of zephyr, I think eventfd is more of a network feature as this API is only useful  when using `zsock_poll` (as opposed to the more universal `k_poll`).  

So my proposal is to name space this feature using the same infrastructure as other socket API methods. `zsock_eventfd`.  The feature could be an addon similar to `socketpair`.  

**Describe alternatives you've considered**
I could use zsock_socketpair - however, this is a more heavy weight solution for my use case, and adds addition `prj.conf` with extra requirements that should not be necessary for my use case (aka: HEAP_MEM_POOL_SIZE, SOCKETPAIR_BUFFER_SIZE)

**Additional context**
For history - here is the PR where it was decided to require `!ARCH_POSIX`, https://github.com/zephyrproject-rtos/zephyr/pull/22863
I hope this can be re-evaluated.

closed 2023-07-20 by aescolar

TomzBench, 2023-07-20:
great !

aescolar, 2025-09-01:
> does `native_sim/native/64` support eventfd with `zsock_poll`? 

@TomzBench It should , but maybe the default C library is not defaulting to PICO if you only enable it, so you have host headers breaking things. Can you create a separate bug report?