zephyrproject-rtos/zephyr #47872: Differentiating Samples, Tests & Demos

aasthagr, 2022-07-15:
## Introduction

This RFC is intended to be discussed by TSC members, maintainers, and collaborators.
 
Comments are welcomed.

### Problem description

There is confusion between what goes under "tests" versus what goes under "samples". Part of this is due to no proper distinction between a Test and a Sample.

### Proposed change

We currently have two categories of projects (Samples & Tests). Not only do these categories need to be properly defined so we can clearly distinguish between each of them, but we should have a third category (demos) as well.

## Detailed RFC

### **Samples**

A Sample is an application in Zephyr that briefly showcases how the functionality of domain/subsystem/feature is used.

**Minimum Requirements/ Criteria for Samples:**

- Test frameworks must not be used for samples.
- Leveraged in CI for quality confidence, its primary purpose is to exemplify the usage of domain/subsystem/feature.
   > Note: Must not serve the purpose of a Testcase.
- Must have ReadMe documentation clearly indicating the purpose of the sample, HW requirements, and sample output if any.
- Should ideally run on multiple platforms & architectures.
- Every sample should be enabled to run with Twister i.e should have a YAML file.
- Should provide sufficient coverage of domain/subsystem/feature.

**Samples (that do not meet minimum requirements):**

1. [samples/net/syslog_net ](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/syslog_net): No TestCases exist for this functionality and this sample is used to validate the Syslog protocol.

    _Action Plan:_ Move it to the tests directory and rework it to better serve the purpose as a TestCase or else write the tests for 
    the Syslog protocol to validate the transmission over UDP.

2. [samples/subsys/logging/dictionary](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/logging/dictionary): No TestCases exist for dictionary logging format and no other logging
sample supports validation of dictionary format.

    _Action Plan:_ Tests to be written for validation of dictionary logging format or enhance other tests in logging subsystem to
     support dictionary log format.

3. [samples/posix/eventfd](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/posix/eventfd): Missing ReadMe documentation.

     _Action Plan:_ Add ReadME documentation explaining the usage of the sample.

### Tests

A test is a complete test suite, the purpose of which is to flag twister if any functionality breaks in Zephyr. All the edge and negative cases should be part of the TestSuite.

**Minimum Requirements/ Criteria for Tests:**

- Clear PASS/FAIL indication.
- Checks whether the software matches the expected requirements.
- Should ideally run on multiple platforms & architectures.
- Every test should be enabled to run with twister i.e should have a YAML file.
- Doxygen should exist for every new test case added.
- All new APIs MUST be tested, and tests MUST cover at least 80% of the added functionality using the code coverage tool and reporting provided by the project. Reference here.

**Tests (that do not meet minimum requirements):**

1. [tests/drivers/dma/chan_link_transfer](https://github.com/aasthagr/zephyr/tree/condvar-kernel-sample/tests/drivers/dma/chan_link_transfer) : This testcase can run on other platforms & architectures which has dma support.

    _Action Plan:_ Enable this test for more platforms & architectures.

2. [tests/arch/xtensa_asm2](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/arch/xtensa_asm2):  This testcase doesn’t have a YAML File.

    _Action Plan:_ Add a testcase.yaml file for this test.

### Demos

It could be a set of applications demonstrating multiple subsystems that show how zephyr can be used in real-world applications involving multiple subsystems. It would then be a more extensive category than "Sample".

**Minimum Requirements/ Criteria for Demos:**

- Integrates multiple subsystems/domains/features.
- Must have ReadMe documentation clearly indicating the purpose of the demo, HW requirements, and demo output if any.
- Should ideally run on multiple platforms & architectures.
- Every demo should be enabled to run with twister i.e should have a YAML file.

**Sample that should be a Demo:**

1. [samples/boards/reel_board/mesh_badge](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/boards/reel_board/mesh_badge): This sample makes a more appropriate demo.

    _Action Plan:_ Move it under the Demo category.

### Proposed change (Detailed)

- Update the Samples & Tests so they meet the necessary requirements.
-    - Some samples may be changed to tests and vice-versa.
- Have a dedicated directory for "Demos".
- Remove duplicate or stale samples 

### Dependencies

### Concerns and Unresolved Questions

## Alternatives

aasthagr, 2022-07-22:
@gregshue I think the manual test applications still go under the tests category but the manual tests can be marked build only in the YAML file since CI doesn't have actual hardware support yet.

Also, can you elaborate on what you mean by actual product configurations? Kconfigs?

gregshue, 2022-07-22:
@aasthagr 

> Also, can you elaborate on what you mean by actual product configurations? Kconfigs?

Sure. 

Many end users will be creating FW images for commercial products (e.g., a USB dongle providing a BLE mesh interface; multi-function printer; automotive anti-lock brake unit; defibrillator; home door lock; etc.). 

Similar to existing tests and samples, each "product" directory in the workspace will have:
- a top-level CMakeLists.txt
- a prj.conf
- one or more project-unique source files (which may be empty)
- a YAML file for building under `twister` as part of a CI pipeline
- top-level documentation

Downstream users need a place to put these that will not conflict with current or future Zephyr directory tree structure and namespaces, yet can have all the current and future build/test-runner/documentation functionality available here just like under `tests/` and `samples/` - but the scope and purpose is different than either. It is not a test and it is not a sample (API/functionality demonstration). The same justification for having both `tests/` and `samples/` trees applies for these other types of executables.

> I think the manual test applications still go under the tests category but the manual tests can be marked build only in the YAML file since CI doesn't have actual hardware support yet.

Manual test images are different enough from fully automated tests they require special consideration. It does not matter whether CI has actual hardware support or not. The tests themselves will require external ("human") input or observation, so CI cannot verify that the tests run correctly.   Marking them as build-only may get through current `twister` execution, but how does an agent consistently identify a "manual test" vs a fully automated test that only requires building code (e.g., under `tests/cmake/`)?

aasthagr, 2022-07-22:
@gregshue If I am following you correctly, do you mean to have a framework in place which allows for easier integration of products? 

And for the second part, do you mean to have support for a separate subcategory of tests that can determine the test as a manual test which is not build-only or automated test?

gregshue, 2022-07-22:
> do you mean to have a framework in place which allows for easier integration of products?

Yes, but it can be as simple as adding a `products/` tree parallel to `samples/`, `tests/`, and `demos/`, modeling a naming patter with a prefix to avoid collisions with other trees and other modules, and putting a base-level "product" in there for verifying CI builds this tree.

> And for the second part, do you mean to have support for a separate subcategory of tests that can determine the test as a manual test which is not build-only or automated test?

I don't know if "subcategory" is the right term.  Manual tests and automated tests are as different as samples and demos and products.  Don't think about only the needs of the build system.  Also think about what is needed to verify the upcoming requirements have been met.  Some of that verification will require manual interactions.  The results of those tests will need to be aggregated with the automated test status to determine which requirements have not been met for a particular build.

nashif, 2022-07-25:
> @gregshue I think the manual test applications still go under the tests category but the manual tests can be marked build only in the YAML file since CI doesn't have actual hardware support yet.

tests that can run manual or automated, should never be marked build-only. build_only is ONLY for code that is not designed to run. For this, we use harness which indicates how something be tests.

nashif, 2022-07-25:
"products" is out of scope, those should be maintained downstream and I do not think they should be part of the scope of this RFC. We probably can have a page in the documentation pointing to products and product repos and leave everything to the maintainers. The closest we can get to "products" is what is listed as "demos" above which are a bit more involved and are part of the project and run on multiple hardware, contrary to products which are not maintained by the project.

Reading through, I guess the first priority of this RFC is to cleanup samples/ and tests/ and prevent further misuse, i.e. use samples as tests and copy samples for one piece of hardware to demonstrate another hardware of the same class, something we could unify into one sample. An issue we often see with sensors.

Defining criteria for samples will help improve the quality of sample and also put us on the right path to establish the 3rd category of demos.

nashif, 2022-07-25:
> > those should be maintained downstream ...
> 
> Like in `example-application`?

this is not downstream. 

> Why not? This RFC is explicitly talking about Zephyr directory tree structure related to subtrees containing "bill-of-features" (e.g., prj.conf). This tree structure pattern is also applied to modules both inside and outside of zephyrproject-rtos, so must accommodate projects (e.g., example-application's `app/` folder), as well as avoid namespace conflicts with what other downstream modules may do.

I am not following why we as an upstream project that is not tied to any product directly and in the upstream tree have to carry and maintain product code that is not part of the project. 

> "can" is an interesting word. What prevents us from getting closer? What prevents us from having a simple, commodity-level product (e.g., USB-to-Ethernet dongle? USB-to-802.15.4 sniffer? 802.15.4 secure repeater?) built out of Zephyr and targeted at some commercially available board (e.g., Raspberry Pi Pico?)

Again, why would we maintain products as part of the vanilla upstream zephyr. If there are products out there that are seeking esposure, they should do that in their own environment and IMO should not be in zephyr itself.

> hat may be impetus, but the first priority needs to be recognizing the related set of **needs** (requirements), and and identifying at a top-level **how those needs will be met** (architecture/organization/definitions/strategic planning).

the requirement here is: cleanup samples and tests and set criteria for both and diffrentiating them.

> . Ignoring the more broad set of needs just sets us up for rework sooner than necessary.

what broad set of needs and who needs them? Create an issue about those needs and we can take from there.

gregshue, 2022-07-26:
> > Like in example-application?
> 
> this is not downstream.

Dependency wise, `example-application` is dependent on Zephyr repository, but the Zephyr repository is entirely independent and unaware of `example-application`.  This relationship was chosen to specifically represent and verify that downstream repositories provided outside of zephyrproject-rtos would also be supported. For all practical purposes, this is "downstream".

 > I am not following why we as an upstream project that is not tied to any product directly and in the upstream tree have to carry and maintain product code that is not part of the project.
 
That's because we are using two different definitions of "product".  You seem to be thinking of sellable goods with a particular branding and warranty.  I am thinking of a field-deployable, security updatable feature configuration implemented completely from reusable secure (and safe?) component implementations.  The feature set does not need to be large and it does not have to be deployed on hardware, but it is needed to verify that the components in the extensible product line platform of Zephyr do provide the core functionality needed for a product: e.g., factory reset, factory programmable serial number, manufacturer, and model numbers, configurable MAC addresses, etc. This capability is intrinsically part of Zephyr Project verification.

> what broad set of needs and who needs them? Create an issue about those needs and we can take from there.

I did create an issue - two years ago.  That resulted in the creation of `example-application`, which made some of them explicit and visible within the Zephyr Project.  The existing needs of it and other similar downstream users are already visible, and whatever solution is decided here must meet those needs too.

gregshue, 2022-07-26:
@aasthagr 

> [Demos] could be a set of applications demonstrating multiple subsystems that show how zephyr can be used in real-world applications involving multiple subsystems.

🤔 I'd like to see more clarification on the purpose of Demos.  Consider:
- Don't a large portion of Samples already use multiple subsystems (e.g., console + something else; net + something else)?
- Aren't usage of multiple subsystems demonstrated by integration tests?  Will the integration tests needed for verifying system (or system element) requirements be sufficient?
- How is "can be used in real-world applications" different than making and supporting one or more production-level "products"?

SebastianBoe, 2022-09-26:
Some Kconfig options have different reasonable default values in a sample meant for testing / development purposes and a demo meant to demonstrate a final product.

E.g. dummy passwords are reasonable during development, but must be enforced to not go into production images.
How to handle errors (reset in production, halt in development/debug).

If we were to introduce a "Demo" category, could we define a Kconfig option to distinguish between a sample and a demo and define a demo to be built optimized for production, and not for development/debug?

nashif, 2023-02-07:
completed.

closed 2023-02-07 by nashif