zephyrproject-rtos/zephyr #47877: ECSPI support for NXP i.MX devices

Abrahamh08, 2022-07-15:
**Is your feature request related to a problem? Please describe.**
There is no spi driver for freertos i.MX supported devices such as the i.MX7D M4.

**Describe the solution you'd like**
A spi driver using the freertos ECSPI module for i.MX devices.

**Describe alternatives you've considered**
I've tried adding MCUXpresso SDK support for the i.MX7D. It becomes too complicated with their custom non-CMSIS board files.

pdgendt, 2022-07-20:
I've worked on this [here](https://github.com/pdgendt/zephyr/tree/drivers-spi-mcux-ecspi). But this was ported from zephyr 2.7 where pinctrl wasn't implemented yet.
So it still has some `TODO` entries and an [update to the HAL](https://github.com/zephyrproject-rtos/hal_nxp/commit/ec24aed9945377cbbb02c1f8d634deaa331b7ed8) is needed. Feel free to test and fill in the blanks :)

Abrahamh08, 2022-07-21:
> I've worked on this [here](https://github.com/pdgendt/zephyr/tree/drivers-spi-mcux-ecspi). But this was ported from zephyr 2.7 where pinctrl wasn't implemented yet. So it still has some `TODO` [...]

Hi, thanks for making that repo! I have looked at it quite a bit while trying to implement this. I am also the person who asked you about it in discord recently. However, this requires MCUXpresso SDK support, and as I mentioned I tried adding support for the i.MX7D to MCUXpresso with no success. They have i.MX7ULP support, so you would think it is trivial to port that, but I also tried that and it turned into a crazy complex problem. Not to mention, the ecspi driver does not even work with the i.MX7ULP in MCUXpresso, as it contains a lot of references to missing feature macros.

closed 2023-02-03 by github-actions[bot]