zephyrproject-rtos/zephyr #47895: samples: smp_svr missing CONFIG_MULTITHREADING=y dependency

gregshue, 2022-07-16:
**Describe the bug**
`zephyr/samples/subsys/mgmt/mcumgr/smp_svr/src/main.c::main()` call to `k_sleep()` depends on multithreading being defined, but does not force this in the `prj.conf`.  This implicit dependency bug blocks reorganizing the `mcuboot` module zephyr port for more consistency with the Zephyr config and build structure (which reveals this bug).

Please also mention any information which could help others to understand
the problem you're facing:
 - Building on Ubuntu 20.04 with Zephyr repository at commit `dddb5dd6`, using one of zephyr-sdk-0.13.1 or zephyr-sdk-0.13.2
 - Running the command `./zephyr/scripts/twister --inline-logs -v- N -M --integration --overflow-as-errors --retry-failed 2 -T zephyr/samples/subsys/mgmt/mcumgr/smp_svr`
 - Examined the `.config` files of the resulting  twister-out and verified all builds of an unmodified system have `CONFIG_MULTITHREADING=y`
 - Ran the same test with a reorganized `mcuboot` and generated build failures on NRF targets. When I added `CONFIG_MULTITHREADING=y` to the shared `prj.conf` then the twister run passed all tests again.
 -  Comparing the original vs new `.config` files for a particular test showed the originals lacked many relevant CONFIGs, so trying to keep the `.config` contents the same across the reorganization is undesirable.

**To Reproduce**
Steps to reproduce the behavior:
1. `./zephyr/scripts/twister --inline-logs -v- N -M --integration --overflow-as-errors --retry-failed 2 -T zephyr/samples/subsys/mgmt/mcumgr/smp_svr`
2. find twister-out -name '\.config' -exec grep MULTITHREADING {} \;
3. See that all report CONFIG_MULTITHREADING=y (even for NRF targets)

**Expected behavior**
I expect the CONFIG_MULTITHREADING=y dependency to be explicitly set in the sample's `prj.conf` so that MCUboot PRs are not blocked by zephyr sample configuration errors.

**Impact**
This is blocking a zephyr port reorganization (and bug fix) PR in github.com/mcu-utils/mcuboot repository. The PR cannot pass the build checks because the `mcu-utils/mcuboot` commit checks require successfully builds against the tip of `zephyrproject-rtos/zephyr/main` branch.

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Ubuntu 20.04
 - Zephyr SDK
 - Commit SHA zephyr.git: `dddb5dd6`,

**Additional context**
N/A

gregshue, 2022-07-19:
@nvlsianpu I understand why this ends up being priority:low, but this is blocking a PR in upstream mcuboot related to an informal TSC initiative. Can we approve this quickly to allow that to get unblocked?

nvlsianpu, 2022-07-27:
"@gregshue Can't build of mcuboot itself and building an application which uses only its components (and is not mcuboot bootloader) be distinguished?
What is signaled as issue here was not an issue before.
"an informal TSC initiative" - which TSC?

gregshue, 2022-08-05:
> Which TSC?

Zephyr Project's TSC.  After discussion at a dev-review meeting this proposal is being closed.  We recognized this specific issue could (and should) be addressed within Kconfig files within the MCUBoot refactoring.  We also recognized this revealed a much larger issue around **when** should a component depend on a default=y. That needs to be discussed with a different RFC.

closed 2022-08-05 by gregshue