zephyrproject-rtos/zephyr #47908: tests/kernel/mem_protect/stack_random works unreliably and sporadically fails

evgeniy-paltsev, 2022-07-17:
The `tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random` works unreliably and sporadically fails after changes not related to it (ot the part of code which is tested by it).

The fail symptoms:

```
Running TESTSUITE stack_pointer_randomness
===================================================================
START - test_stack_pt_randomization
Test Stack pointer randomization
stack pointer changed 0 times out of 64 tests

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_protect/stack_random/src/main.c:81: stack_pointer_randomness_test_stack_pt_randomization: (sp_changed equal to 0)
Stack pointer is not randomized
 FAIL - test_stack_pt_randomization in 0.15 seconds
===================================================================
TESTSUITE stack_pointer_randomness failed.
===================================================================
PROJECT EXECUTION FAILED
```

The test is checking the `CONFIG_STACK_POINTER_RANDOM` feature which should enable initial thread stack pointer randomization.

I've debugged the test and it appeared to be that thread stack pointer is actually not randomized, but any change to the code make test passing.

The test uses TEST_RANDOM_GENERATOR (which is known for low quality of random values provided) and when I've checked the values provided by it I've found that last byte of 'random' value is always the same:

```
427784e7
7e1251e7
b9ad1ee7
f547ebe7
30e2b8e7
```

And for stack randomization we shrink the random value (see random_offset function):
```
size_t fuzz = random_val % CONFIG_STACK_POINTER_RANDOM;
```

As `CONFIG_STACK_POINTER_RANDOM=64` the value we use for randomization is always the same (0x27).

The TEST_RANDOM_GENERATOR uses following algo to provide random values:

```
_rand32_counter += 1000000003U;

random_ret_val = k_cycle_get_32() + _rand32_counter;
```

As in the test we just continuously executing the same code in a loop, the differences between k_cycle_get_32() calls in TEST_RANDOM_GENERATOR are the same (0x2FD in this case) which cause the random returning values with identical last byte.

There were identical issues with `kernel.memory_protection.stack_random` with some hacks to TEST_RANDOM_GENERATOR:

https://github.com/zephyrproject-rtos/zephyr/pull/36996
https://github.com/zephyrproject-rtos/zephyr/pull/37499

But looks like we either need to significantly improve TEST_RANDOM_GENERATOR or make the test less vulnerable to such random issue or disable the test to avoid sporadic failures.

evgeniy-paltsev, 2022-07-17:
Pinging the developers, who discussed previous similar issue @fkokosinski @cfriedt @d3zd3z @dleach02

cfriedt, 2022-07-17:
Not entirely surprising after seeing that it uses the cycle counter, which can exhibit flakiness / time compression on certain qemu platforms.

Do we effectively need to find a better way to generate random numbers on such platforms?

evgeniy-paltsev, 2022-07-17:
> Do we effectively need to find a better way to generate random numbers on such platforms?

Yep, I guess we need to change TEST_RANDOM_GENERATOR implementation from this custom (and broken) algo to some well-know pseudo random number generator algorithm.

One possible candidate is the example from C standard:

```
static unsigned long int next = 1;

int rand(void) // RAND_MAX assumed to be 32767 
{ 
    next = next * 1103515245 + 12345; 
    return (unsigned int)(next / 65536) % 32768; 
}
```

dleach02, 2022-07-19:
I'm not exactly sure why TEST_RANDOM_GENERATOR doesn't just ultimately use the XORSHIRO algorithm but with a bad seed. By definition, it is just a test generator not to be used for anything real and from what you highlight, it is probably important to ensure there are no repeating parts of the rand values.

evgeniy-paltsev, 2023-01-19:
As the #48895 was reverted - reopen again.

PR #48895 (2023-01-09): [WIP] random: changed implementation of rand32_timer.c
Changes the implementation of rand32_timer.c because it was not generating sufficiently "random"-looking values on some platforms, the implementation is now a "psuedo-xoshiro"

Requesting comments

Fixes #47908
closed 2023-03-21 by evgeniy-paltsev