zephyrproject-rtos/zephyr #47916: FRAM device support

sslupsky, 2022-07-17:
## Introduction

This RFC is to add an API for serial FRAM devices.  The API was based on the work Vestas did for serial EEPROM devices.

## Detailed RFC

```
/*
 * Copyright (c) 2022 Scanimetrics Inc.
 * Copyright (c) 2019 Vestas Wind Systems A/S
 *
 * Heavily based on drivers/flash.h which is:
 * Copyright (c) 2017 Nordic Semiconductor ASA
 * Copyright (c) 2016 Intel Corporation
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/**
 * @file
 * @brief Public API for FRAM drivers
 */

#ifndef ZEPHYR_INCLUDE_DRIVERS_FRAM_H_
#define ZEPHYR_INCLUDE_DRIVERS_FRAM_H_

/**
 * @brief FRAM Interface
 * @defgroup fram_interface FRAM Interface
 * @ingroup io_interfaces
 * @{
 */

#include <zephyr/types.h>
#include <stddef.h>
#include <sys/types.h>
#include <device.h>

#ifdef __cplusplus
extern "C" {
#endif

typedef int (*fram_api_read)(struct device *dev, off_t offset, void *data,
			       size_t len);
typedef int (*fram_api_write)(struct device *dev, off_t offset,
				const void *data, size_t len);
typedef size_t (*fram_api_size)(struct device *dev);

__subsystem struct fram_driver_api {
	fram_api_read read;
	fram_api_write write;
	fram_api_size size;
};

/**
 *  @brief Read data from FRAM
 *
 *  @param dev FRAM device
 *  @param offset Address offset to read from.
 *  @param data Buffer to store read data.
 *  @param len Number of bytes to read.
 *
 *  @return 0 on success, negative errno code on failure.
 */
__syscall int fram_read(struct device *dev, off_t offset, void *data,
			  size_t len);

static inline int z_impl_fram_read(struct device *dev, off_t offset,
				     void *data, size_t len)
{
	const struct fram_driver_api *api =
		(const struct fram_driver_api *)dev->driver_api;

	return api->read(dev, offset, data, len);
}

/**
 *  @brief Write data to FRAM
 *
 *  @param dev FRAM device
 *  @param offset Address offset to write data to.
 *  @param data Buffer with data to write.
 *  @param len Number of bytes to write.
 *
 *  @return 0 on success, negative errno code on failure.
 */
__syscall int fram_write(struct device *dev, off_t offset, const void *data,
			   size_t len);

static inline int z_impl_fram_write(struct device *dev, off_t offset,
				      const void *data, size_t len)
{
	const struct fram_driver_api *api =
		(const struct fram_driver_api *)dev->driver_api;

	return api->write(dev, offset, data, len);
}

/**
 *  @brief Get the size of the FRAM in bytes
 *
 *  @param dev FRAM device.
 *
 *  @return FRAM size in bytes.
 */
__syscall size_t fram_get_size(struct device *dev);

static inline size_t z_impl_fram_get_size(struct device *dev)
{
	const struct fram_driver_api *api =
		(const struct fram_driver_api *)dev->driver_api;

	return api->size(dev);
}

#ifdef __cplusplus
}
#endif

/**
 * @}
 */

#include <syscalls/fram.h>

#endif /* ZEPHYR_INCLUDE_DRIVERS_FRAM_H_ */
```

### Concerns and Unresolved Questions

FRAM devices are similar to serial RAM devices.  Should a similar API be proposed for those devices as well?

henrikbrixandersen, 2022-07-19:
I must be missing something. The proposed FRAM API is an exact copy of the EEPROM API with `eeprom` replaced by `fram`?

sslupsky, 2022-08-05:
@henrikbrixandersen I largely based this proposal on the existing FLASH API.  But, there are unique differences between the parts that I do not think belong in a FLASH API.  An FRAM more closely resembles an SRAM than a FLASH device.  Incidentally, there is no SRAM API for adding an external serial SRAM at the moment and there probably should be.  So I suggest this RFC could be made into a more generic serial RAM API.  Alternatively, the FLASH API could be changed into a more generic memory device API but then we would need to consider other types of support such as NAND flash, DRAM and others.

henrikbrixandersen, 2022-08-05:
I still don't get it. You're proposing adding an exact copy of the EEPROM API with the word "eeprom" replaced by "fram"?

Why not just implement your FRAM device driver using the EEPROM API, if it behaves the same?

sslupsky, 2022-08-06:
@henrikbrixandersen It is not an exact copy of the flash api.  There is no block erase for a RAM device.  I had a look at the EEPROM api and I see your point now, the EEPROM api does not have the block erase and so yes, it is the same as my proposal.

So, should the EEPROM API be renamed to reflect that is can be used for other devices as well?

henrikbrixandersen, 2022-08-06:
> So, should the EEPROM API be renamed to reflect that it can be used for other devices as well?

I don't think it should, no.

sslupsky, 2022-08-06:
Hmmm, why?  It does not seem obvious to use an API named "EEPROM" for reading and writing to a SRAM or FRAM?

nordicjm, 2023-04-20:
The retained memory driver seems like the right base driver for this: https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/retained_mem.h

nordicjm, 2023-04-21:
The description could be updated and a "capabilities" or "flags" function added which returns if it is volatile or not. I guess the big question is what would FRAM be used for? For the retained memory, the ideal use case is for application state (between boots or different applications being chain loaded).

sslupsky, 2023-04-21:
I would think the storage use would be application dependent.  For instance, logging data, code storage, application state, updates, XIP.

nordicjm, 2023-04-24:
> I would think the storage use would be application dependent. For instance, logging data, code storage, application state, updates, XIP.

For code storage, updates and XIP it would need to be a flash device

sslupsky, 2023-04-24:
> For code storage, updates and XIP it would need to be a flash device

I do not understand?  There is no reason FRAM cannot be used for these purposes as well?  There are existing MCU's that use FRAM for code storage and code execution.  I have used it for data storage, updates, logging, etc.

nordicjm, 2023-04-24:
> > For code storage, updates and XIP it would need to be a flash device
> 
> I do not understand? There is no reason FRAM cannot be used for these purposes as well? There are existing MCU's that use FRAM for code storage and code execution. I have used it for data storage, updates, logging, etc.

All the functions for this thing in-tree use the flash_* functions, which expects a flash driver with the correct function pointers, so using dfu as an example: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/dfu/img_util/flash_img.c this means that the fram device struct would need to be compatible with the flash device struct, at which point the question is why have 2 structs that are the same?

henrikbrixandersen, 2023-04-24:
> All the functions for this thing in-tree use the flash_* functions, which expects a flash driver with the correct function pointers, so using dfu as an example: [...]

This mirrors my concern in https://github.com/zephyrproject-rtos/zephyr/issues/47916#issuecomment-1189093609 about the proposed FRAM API being identical to the EEPROM API but with structs and functions renamed.

hubertmis, 2023-06-06:
> Unfortunately the API did not consider other non volatile memories such as NAND, FRAM, etc. At some point, this should be fixed.

I'm thinking if subsystems like dfu or settings should be ported to generic non-volatile memories (called xRAM, NVRAM, EEPROM, or whatever) to allow applications to use the same "settings" API that would work efficiently regardless of the storage is Flash, EEPROM, FRAM, MRAM, or anything else with random read-write access capability.

sslupsky, 2023-06-06:
> I'm thinking if subsystems like dfu or settings should be ported to generic non-volatile memories (called xRAM, NVRAM, EEPROM, or whatever) to allow applications to use the same "settings" API that [...]

Agreed, to properly address this issue requires some consideration to make subsystems that use storage independent of the physical medium.

henrikbrixandersen, 2023-08-24:
> This mirrors my concern in [#47916 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/47916#issuecomment-1189093609) about the proposed FRAM API being identical to the EEPROM API but with structs and functions renamed.

Example of adding FRAM support using the existing EEPROM API: #61653

closed 2023-08-30 by henrikbrixandersen

benner, 2023-12-01:
Just FIY. I'm using FRAM too (on SPI bus) but as `jedec,spi-nor` type flash. Almost everything works as expected (e.g. littlefs). What's different - default byte is 0x00 and not 0xff, erase is nop. One problem I found is - mcuboot which expects 0xff after erase.