zephyrproject-rtos/zephyr #47921: test: pin_get_config failed on it8xxx2_evb

hunterhu3000, 2022-07-18:
**Describe the bug**
Test case tests/drivers/gpio/gpio_basic_api failed in daily test. When setting a GPIO to high, and then get the GPIO result, the result is not as expected.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? it8xxx2_evb
 - What have you tried to diagnose or workaround this issue? Will provide a PR to fix it.

**To Reproduce**
Steps to reproduce the behavior:
1.  twister  -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests/drivers/gpio/gpio_basic_api/ -vv -X gpio_loopback

**Expected behavior**
The test can pass

**Impact**
1 test case failed in daily test

**Logs and console output**
```
DEBUG   - DEVICE: START - test_gpio_port
DEBUG   - DEVICE: Validate device GPIO_A
DEBUG   - DEVICE: Check GPIO_A output 0 connected to input 1
DEBUG   - DEVICE: OUT 0 to IN 1 linkage works
DEBUG   - DEVICE: - bits_physical
DEBUG   - DEVICE: - pin_physical
DEBUG   - DEVICE: - check_raw_output_levels
DEBUG   - DEVICE: - check_logic_output_levels
DEBUG   - DEVICE: - check_input_levels
DEBUG   - DEVICE: - bits_logical
DEBUG   - DEVICE: - check_pulls
DEBUG   - DEVICE: @@@ flags_set = a0000
DEBUG   - DEVICE: @@@ flags_get = a1000
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_gpio_port.c:653: pin_get_config: (flags_set not equal to flags_get)
DEBUG   - DEVICE: flags are different
DEBUG   - DEVICE: FAIL - test_gpio_port in 0.54 seconds
```
flags_set means (GPIO_OUTPUT | GPIO_OUTPUT_INIT_HIGH) 
flags_get means (GPIO_OUTPUT | GPIO_OUTPUT_INIT_HIGH) and GPIO_VOLTAGE_3P3

When setting the flags, the voltage infomation is not given, but when get the flags, it8xxx2_evb filled in this information automatically.
So we only need to compare the bit of GPIO_OUTPUT and GPIO_OUTPUT_INIT_HIGH in the test, instead of 
```
zassert_equal(flags_set, flags_get, "flags are different");
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

mmahadevan108, 2022-07-20:
cc @semihalf-barnas-michal

hunterhu3000, 2022-08-30:
The patch is still under review.

PR #47923 (2022-09-07): tests: Change the result compare method in pin_get_config()
Fixes #47921
When setting a GPIO to certain value, and then get the GPIO value, on some
platforms, such as ITE, these two values may not be the same. For example
setting GPIO_OUTPUT | GPIO_OUTPUT_INT_HIGH, it will read back
GPIO_OUTPUT | GPIO_OUTPUT_INT_HIGH | GPIO_VOLTAGE_3P3. This is as expected,
as GPIO_VOLTAGE is a read only property. So when comparing the gpio set
and get values, it is necessary to filter the read only bits first and
then compare to the set value.
commit 99ab3fba (2022-09-07): tests: Change the result compare method in pin_get_config()

Fixes #47921
When setting a GPIO to certain value, and then get the GPIO value, on some
platforms, such as ITE, these two values may not be the same. For example
setting GPIO_OUTPUT | GPIO_OUTPUT_INT_HIGH, it will read back
GPIO_OUTPUT | GPIO_OUTPUT_INT_HIGH | GPIO_VOLTAGE_3P3. This is as expected,
as GPIO_VOLTAGE is a read only property. So when comparing the gpio set
and get values, it is necessary to filter the read only bits first and
then compare to the set value.
closed 2022-09-07 by carlescufi