zephyrproject-rtos/zephyr #47930: tests: arch: arm: arm_interrupt: arch.interrupt.no_optimizations - Data Access Violation - MPU Fault

nordic-piks, 2022-07-18:
**Describe the bug**
The   tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations fails on-target tests - Data Access Violation - MPU Fault

Observed for
- nrf52dk_nrf52832
- nrf52840dk_nrf52840

There is MPU Fault detected and system halts, no other output.

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/arch/arm/arm_interrupt -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See stack trace and system halting

**Expected behavior**
Valid test output

**Impact**
Not clear

**Logs and console output**
ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-2265-g62f19cc6b3d4  ***
Running TESTSUITE arm_interrupt
===================================================================
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x20002c38
E: r0/a1:  0x6aeac82c  r1/a2:  0x134fe27e  r2/a3:  0xb0b2edf1
E: r3/a4:  0xb6c46227 r12/ip:  0xae826fff r14/lr:  0x3d334f60
E:  xpsr:  0x62c51000
E: Faulting instruction address (r15/pc): 0xff3add1b
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200001b0 (main)
Caught system error -- reason 2
Was not expecting a crash

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: zephyr-v3.1.0-2265-g62f19cc6b3d4

hakehuang, 2022-07-20:
without config_no_optimzation=y, nxp platform test pass, but with CONFIG_NO_OPTMIZATION=y, there is no console output, as the exception overried the serial output function

lawrencek52, 2022-07-20:
Try adding "add_compile_options(-mno-unaligned-access)" to the top of the
CMakeLists.txt in the project.

microbuilder, 2022-07-20:
Please confirm this isn't simply the stack size being too small for this platform, as per https://github.com/zephyrproject-rtos/zephyr/issues/47855#issuecomment-1190806681=

closed 2022-08-08 by nashif

nordic-piks, 2022-08-09:
#48776 Fixes the issue, thanks for help.

PR #48776 (2022-08-08): tests: arm: Increase main stack size with no optimizations
If the test is run with the config NO_OPTIMIZATIONS enabled then the
stack size usage increases by around 80% for ARM platforms.
Increase the stack size used in test cases that enables building with no
optimizations for ARM.

Update description on TEST_ARM_CORTEX_M since it was outdated and said
it was only used for a single purpose.

Fixes: #47930
Fixes: #47929
Fixes: #47855
commit fcb98131 (2022-08-08): tests: arm: Increase main stack size with no optimizations

If the test is run with the config NO_OPTIMIZATIONS enabled then the
stack size usage increases by around 80% for ARM platforms.
Increase the stack size used in test cases that enables building with no
optimizations for ARM.

Update description on TEST_ARM_CORTEX_M since it was outdated and said
it was only used for a single purpose.

Fixes: #47930
Fixes: #47929
Fixes: #47855