zephyrproject-rtos/zephyr #47942: Mutex priority inheritance when thread holds multiple mutexes

alwa-nordic, 2022-07-18:
The section on [priority inheritance in mutexes] contains the following warning.

> The kernel does not fully support priority inheritance when a thread holds two or more mutexes simultaneously. This situation can result in the thread’s priority not reverting to its original non-elevated priority when all mutexes have been released. It is recommended that a thread lock only a single mutex at a time when multiple mutexes are shared between threads of different priorities.

This is clearly a bug. There is a tangential discussion in https://github.com/zephyrproject-rtos/zephyr/issues/15138#issuecomment-540892530.

The specification would be simpler if it did not talk about the "original non-elevated priority" and when it's reverted. I think we should just say that each thread has a base priority and an effective priority. The effective priority is at all times the most urgent of the thread's own base priority and all of the effective priorities of threads waiting on a mutex taken by this thread.

[priority inheritance in mutexes]: https://github.com/zephyrproject-rtos/zephyr/blob/38635f889979030328ff3b35c8d25d2990e9819e/doc/kernel/services/synchronization/mutexes.rst#priority-inheritance

mmahadevan108, 2022-07-19:
cc @peter-mitsis

alwa-nordic, 2022-07-19:
> "What is it that you want as the outcome?" Is it just clearer documentation? Is it a more robust priority inheritance algorithm? Is it both?

Both, of course! Thank you for asking :D

But seriously. The main problem is that the warning is too vague, so it comes of sounding quite scary. If we document the limitation/requirement that each thread must be release mutexes in the reverse order of acquiring them, we have solved the issue. We can keep a explanation of the bad stuff that may happen if the reverse order is not honored. But it must be clear that if the programmer follows the rules of the API, nothing bad will happen.

A robust algorithm would be nice, of course.

Bonus suggestion: I think it's beneficial to explicitly say in the API doc that a mutex shall be acquired and released in the same thread. It's maybe also educational to point to semaphores that don't have the that limitation, but don't have thread-aware features like priority inheritance and reentrancy.

PR #48132 (2022-07-22): doc: Update mutex priority inheritance description
Updates the priority inheritance description to better explain what
happens during priority inheritance and warn of the consequences of
not following best-practices.

Fixes #47942
closed 2022-07-22 by nashif