zephyrproject-rtos/zephyr #47970: SFDP parameter address is not correct

marekmatej, 2022-07-18:
https://github.com/zephyrproject-rtos/zephyr/blob/03912c23e380c9e36a7ce40cc061f58720f67e73/drivers/flash/spi_nor.c#L916

The jesd216_param_addr() returns parameter table address in DW units. This number is used in the read_sfdp() function without converting it to byte address. For example, Spansion dataflash has parameter pointer 0x448, which multiplied by 4 gives correct SFDP parameter pointer 0x1120.

henrikbrixandersen, 2022-07-19:
Please use the bug report template when reporting a bug (you can copy it from here since you missed it in the first place: https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md).

nvlsianpu, 2022-09-15:
> Parameter Table Pointer from SFDP param. header represents the address of parameter table in dual-words, therefore it need to be converted to byte address by multiplying by 4.

<b>NACK</b> According to the JESD216E SFDP specification: PTP is specified in byte unit and must be dword aligned.

I did test how SFDP readout works on nRF52840dk_nrf52840 (it has MX25R6435F serial flash) with qspi nor driver (using samples/drivers/jesd216). Procedure in the sample is the same as in spi_nor driver and it does work well (addressing is correct, ).

nvlsianpu, 2022-09-15:
I tested readout using spi-nor driver on nrf52840dk_nrf52840. It works well as well.
I open PR with configuration used by my: https://github.com/zephyrproject-rtos/zephyr/pull/50277

nvlsianpu, 2022-09-15:
According to S25FL127S specification it should behave the same.

I'm closing as can not to reproducer a bag here. JESD216 are addressed properly.

closed 2022-09-15 by nvlsianpu