zephyrproject-rtos/zephyr #47986: Rework of STM32 bxCAN driver filter handling required

martinjaeger, 2022-07-19:
## Background

The STM32 bxCAN assigns a number to the filters in the order they appear in the filter banks. This number is used in the RX FIFOs to associate a stored message to the used filter (the filter number is called filter match index in this context).

One filter bank can store 1 (extended ID + mask) to 4 (std ID, non-masked) filters depending on their mode and scale. If a bank configuration is changed e.g. from extended ID filter with a mask to std ID with mask, all filter numbers in the following banks get an offset by 2 as the same bank has 2 more filters.

Each of the two FIFOs has its own numbering scheme. However, the Zephyr driver currently only uses FIFO 0.

See below figure from RM0390 for an example for filter numbering for different FIFOs (assigned per bank).
[image]

This behaviour makes the filter assignment in the driver very complicated, especially with respect to the maximum number of filters required by the application, configured using `CONFIG_CAN_MAX_FILTER`.

## Current implementation and issues

### Direct mapping between filter number and callback

The driver uses the filter number as the index in the callback array, which allows to deliver an incoming message to the application very fast without searching for the right callback depending on the CAN ID.

If there is an unused part of a filter bank, also the callback array at that position must be left empty.

Example:
1. Add std ID filter without mask (list mode). This filter will get filter number 0 and the remaining 3 slots in this bank are unused.
2. Add extended ID filter with mask: This filter will get number 4 and occupy one entire bank.

If `CONFIG_CAN_MAX_FILTER` was set to 5, no further extended ID filter can be added even though only 2 filters are actually configured.

This leads to unpredictable behavior, as a user would expect to be able to add up to `CONFIG_CAN_MAX_FILTER` filters.

Adding `can_get_max_filters` API call to the driver also makes the `tests/drivers/can/api` test fail for `CONFIG_CAN_MAX_FILTER=5` because the `can_stm32_shift_arr` function requires some more empty elements at the end to work correctly.

As stated in #44725, the driver may even overflow the callback arrays under some circumstances.

## Possible solutions

### 1. Configure std and ext filters individually and always use mask

The STM32 CAN-FD driver has individual settings for CONFIG_CAN_MAX_STD_ID_FILTER and CONFIG_CAN_MAX_EXT_ID_FILTER. This could also be used for the bxCAN driver.

In order to make the number of filters predictable, filters would always be stored with a mask and the two FIFOs can be used to differentiate between std and ext filters because they have their individual numbering scheme (FIFO 1 is currently not used).

The lower numbers of `filter_id` returned by `can_add_rx_filter` (less than `CONFIG_CAN_MAX_STD_ID_FILTER`) would be used for std ID, the higher numbers for ext ID (same as in the CAN-FD driver).

**Pro**
- Simplified driver implementation: no shifting shifting operations required

**Contra**
- Filter bank usage not fully optimized, maximum 28 std ID filters or 14 ext ID filters possible.

### 2. Change CAN API `can_get_max_filters`

This function only works if the available number of filters is constant, independent of the order or type of filters already present in the banks.

The function could be replaced by a more dynamic function `can_get_free_filters` which returns the number of filters still available for the given type (std or ext). This would also be more useful for an application, as there is currently no way to determine if there are free filters left except for trying to add one and checking if it fails.

**Pro**
- Amount of useable filters is maximized for bxCAN

**Contra**
- API change for all CAN drivers

henrikbrixandersen, 2022-07-19:
> Filter bank usage not fully optimized, maximum 28 std ID filters or 14 ext ID filters possible.

Thank you for this analysis. The above sounds like the simplest solution. I'd say 28 std + 14 ext filters should suffice most needs?

martinjaeger, 2022-07-19:
> The above sounds like the simplest solution. I'd say 28 std + 14 ext filters should suffice most needs?

It would not be 28 + 14, but 14 if all filters are ext filters, 28 if all are std and something in between if you use mixed IDs. But I still think it should be sufficient. If someone really needs more, he/she can come up with a more sophisticated driver implementation.

I did actually start with implementation if this solution already, but wanted to get some more opinions to see if there are any concerns before finishing it completely.

closed 2022-08-03 by fabiobaltieri