zephyrproject-rtos/zephyr #47987: test: samples/boards/mec15xxevb_assy6853/power_management failed after commit 5f60164a0fc

hunterhu3000, 2022-07-19:
**Describe the bug**
test case samples/boards/mec15xxevb_assy6853/power_management failed after commit 5f60164a0fc

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? mec15xxevb_assy6853
 - What have you tried to diagnose or workaround this issue? git bisect and found first bad commit is 5f60164a0fc

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b mec15xxevb_assy6853  samples/boards/mec15xxevb_assy6853/power_management/
2. west flash

**Expected behavior**
The test should pass

**Impact**
1 test case fail in daily test

**Logs and console output**
```
[00:00:00.006,927] <wrn> pwrmgmt_test: PM single-thread test started for cycles: 5, logging: 0
[00:00:00.006,958] <inf> pwrmgmt_test: About to enter light sleep
[00:00:01.508,941] <inf> pwrmgmt_test: About to enter deep Sleep
[00:00:02.129,028] <inf> pwrmgmt_test: PM sleep entry latency 0.621 seconds
[00:00:02.129,058] <wrn> pwrmgmt_test: Sleep entry latency is higher than expected
[00:00:04.609,527] <err> os: ***** USAGE FAULT *****
[00:00:04.609,527] <err> os:   Illegal use of the EPSR
[00:00:04.609,558] <err> os: r0/a1:  0x000e9119  r1/a2:  0x00000000  r2/a3:  0x00118e20
[00:00:04.609,588] <err> os: r3/a4:  0xffffffff r12/ip:  0x001186a0 r14/lr:  0x000e5d83
[00:00:04.609,588] <err> os:  xpsr:  0x60000000
[00:00:04.609,619] <err> os: Faulting instruction address (r15/pc): 0x00001080
[00:00:04.609,619] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:04.609,649] <err> os: Current thread: 0x118d90 (main)
[00:00:04.665,527] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...)  0.14.1
 - Commit SHA or Version used 5f60164a0fc

mmahadevan108, 2022-07-19:
cc @semihalf-barnas-michal

mmahadevan108, 2022-07-19:
cc @nordic-krch

hunterhu3000, 2022-07-28:
@nordic-krch It seems commit 5f60164a0fc36484b6 causes this problem. Could you please help to check?

hunterhu3000, 2022-08-04:
The commit https://github.com/zephyrproject-rtos/zephyr/commit/5f60164a0fc36484b60004310320dfd6f3ef0525 think if there is no activated log backend, the timeout will be set to K_FOREVER. If there is at least 1 log backend activated, the value is 50ms. 
Before this commit, the value is hard code to 50ms and works well. After this commit, it cannot find any activated backend, and set the value to K_FOREVER and cause the problem.
Actually, the UART log backend is just enabled and activated, but the log_backend_uart_api has not enabled the function of "is_ready()". So the system always think the UART backend is not activated. And then set the timeout to the wrong value K_FOREVER

nordic-krch, 2022-08-10:
> if there is no activated log backend, the timeout will be set to K_FOREVER. If there is at least 1 log backend activated, the value is 50ms.

I am not sure what is the reason for this bug but this statement above is wrong. K_FOREVER is set when all backends which had autostart flag are ready and there is no reason for periodic polling for backend readiness. Also if `is_ready` is not implemented then `log_backend_is_ready` returns 0 which means that backend is ready. Thus there should be no need for implementation of the `is_ready` api if backend initialization is synchronous.

hunterhu3000, 2022-08-10:
So, when the uart backend is enabled and activated, the return value of z_log_init() should be 0, right?
It looks like the mask will always be 0, when will it be 1 or non-zero?

hunterhu3000, 2022-08-14:
I wonder https://github.com/zephyrproject-rtos/zephyr/commit/5f60164a0fc36484b60004310320dfd6f3ef0525 is just a trigger of the problem.
The test case works like this:
k_msleep(3100) -> swap from main to idle -> swap from idle to logging -> swap from logging to idle -> k_cpu_idle -> z_thread_timeout -> arch_swap: ret = -11 -> z_arm_fault
The logs with some printk
```

DEBUG   - DEVICE: @@@ ready_thread: thread->name = main
DEBUG   - DEVICE: @@@ runq_add: thread->name = main
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ ready_thread: thread->name =
DEBUG   - DEVICE: @@@ runq_add: thread->name =
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ z_impl_k_sleep 1: ticks = 33
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ z_tick_sleep: 1
DEBUG   - DEVICE: @@@ before arch_swap
DEBUG   - DEVICE: -----> arch_swap 1: _current->name = main
DEBUG   - DEVICE:
DEBUG   - DEVICE: @@@ z_thread_timeout: _current->name = main, thread->name = main
DEBUG   - DEVICE: @@@ killed = 0
DEBUG   - DEVICE: @@@ ready_thread: thread->name = main
DEBUG   - DEVICE: @@@ runq_add: thread->name = main
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ z_thread_timeout: done
DEBUG   - DEVICE: -----> arch_swap after unlock 1: _current->name = main
DEBUG   - DEVICE: <----- arch_swap 1 done: _current->name = main
DEBUG   - DEVICE: @@@ after arch_swap: ret = -11
DEBUG   - DEVICE: @@@ z_tick_sleep: 2
DEBUG   - DEVICE: @@@ z_impl_k_sleep 2: ticks = 0
DEBUG   - DEVICE: @@@ ready_thread: thread->name =
DEBUG   - DEVICE: @@@ runq_add: thread->name =
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ before arch_swap
DEBUG   - DEVICE: -----> arch_swap 2: _current->name = main
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ before arch_swap
DEBUG   - DEVICE: -----> arch_swap 3: _current->name =
DEBUG   - DEVICE: -----> arch_swap after unlock 2: _current->name = main
DEBUG   - DEVICE: <----- arch_swap 2 done: _current->name = main
DEBUG   - DEVICE: @@@ after arch_swap: ret = -11
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-3385-g6de4cc5293b4  ***
DEBUG   - DEVICE: @@@ before arch_swap
DEBUG   - DEVICE: -----> arch_swap 4: _current->name = main
DEBUG   - DEVICE: -----> arch_swap after unlock 4: _current->name = main
DEBUG   - DEVICE: <----- arch_swap 4 done: _current->name = main
DEBUG   - DEVICE: @@@ after arch_swap: ret = -11
DEBUG   - DEVICE: @@@ test_pwr_mgmt_multithread
DEBUG   - DEVICE:

DEBUG   - DEVICE: @@@@@@ before light Sleep 1500ms
DEBUG   - DEVICE: @@@ z_impl_k_sleep 1: ticks = 49152
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ z_tick_sleep: 1
DEBUG   - DEVICE: @@@ before arch_swap
DEBUG   - DEVICE: -----> arch_swap 5: _current->name = main

DEBUG   - DEVICE: [00:00:00.111,694] <wrn> pwrmgmt_test: PM single-thread test started for cycles: 5, logging: 0
DEBUG   - DEVICE: [00:00:00.111,724] <inf> pwrmgmt_test: About to enter light sleep
DEBUG   - DEVICE: @@@ k_sem_take log_process_thread_sem
DEBUG   - DEVICE: @@@ k_sem_take log_process_thread_sem done
DEBUG   - DEVICE: @@@ k_sem_take log_process_thread_sem
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ before arch_swap
DEBUG   - DEVICE: -----> arch_swap 6: _current->name = logging

DEBUG   - DEVICE: idle: cc = 1
DEBUG   - DEVICE: idle: before arch_irq_lock
DEBUG   - DEVICE: idle: after arch_irq_lock
DEBUG   - DEVICE: @@ pm_system_suspend ticks = 47609
DEBUG   - DEVICE: @@@ pm_system_suspend: pm_stats_start
[... 64 lines cut ...]
DEBUG   - DEVICE: @@ pm_system_suspend ticks = 30810
DEBUG   - DEVICE: @@@ pm_system_suspend return false 1
DEBUG   - DEVICE: idle: k_cpu_idle !!!!!!!!!!!!!!!
DEBUG   - DEVICE:
DEBUG   - DEVICE: @@@ z_thread_timeout: _current->name = idle, thread->name = main
DEBUG   - DEVICE: @@@ killed = 0
DEBUG   - DEVICE: @@@ ready_thread: thread->name = main
DEBUG   - DEVICE: @@@ runq_add: thread->name = main
DEBUG   - DEVICE: @@@ thread_runq
DEBUG   - DEVICE: @@@ z_thread_timeout: done
DEBUG   - DEVICE: -----> arch_swap after unlock 8: _current->name = main
DEBUG   - DEVICE: <----- arch_swap 8 done: _current->name = main
DEBUG   - DEVICE: @@@ after arch_swap: ret = -11
DEBUG   - DEVICE: @@@ z_arm_fault
DEBUG   - DEVICE: [00:00:04.821,350] <err> os: ***** USAGE FAULT *****
DEBUG   - DEVICE: [00:00:04.821,380] <err> os:   Illegal use of the EPSR
DEBUG   - DEVICE: [00:00:04.821,380] <err> os: r0/a1:  0x00000022  r1/a2:  0x0000000a  r2/a3:  0x400f2c00
DEBUG   - DEVICE: [00:00:04.821,411] <err> os: r3/a4:  0x00000030 r12/ip:  0x00000000 r14/lr:  0x000e479b
DEBUG   - DEVICE: [00:00:04.821,411] <err> os:  xpsr:  0x40000000
DEBUG   - DEVICE: [00:00:04.821,441] <err> os: Faulting instruction address (r15/pc): 0x00119e6c
DEBUG   - DEVICE: [00:00:04.821,472] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
DEBUG   - DEVICE: [00:00:04.821,472] <err> os: Current thread: 0x118cb8 (main)
DEBUG   - DEVICE: [00:00:04.877,349] <err> os: Halting system

```

PR #48720 (2022-08-18): tests: mec15xxevb_assy6853: Enlarge the IDLE_STACK_SIZE from 256 to 512
The test case "samples/boards/mec15xxevb_assy6853/power_management" could
always pass until commit 5f60164 was merged in. This patch changes the sem
take timeout value from 50ms to forever of the log thread, it will make the
idle thread run longer and takes more stack memory. Analyzed the coredump
of the failed test case, it can be found that the test case stucks at
z_idle_stacks. The test case used default CONFIG_IDLE_STACK_SIZE = 256 in
kernel/Kconfig, now we increase the value to 512 in prj.conf for this test
only. With [... cut]
commit 424668f5 (2022-08-18): tests: mec15xxevb_assy6853: Enlarge the IDLE_STACK_SIZE from 256 to 512

The test case "samples/boards/mec15xxevb_assy6853/power_management" could
always pass until commit 5f60164 was merged in. This patch changes the sem
take timeout value from 50ms to forever of the log thread, it will make the
idle thread run longer and takes more stack memory. Analyzed the coredump
of the failed test case, it can be found that the test case stucks at
z_idle_stacks. The test case used default CONFIG_IDLE_STACK_SIZE = 256 in
kernel/Kconfig, now we increase the value to 512 in prj.conf for this test
only. Wit [... cut]
closed 2022-08-18 by fabiobaltieri