zephyrproject-rtos/zephyr #47988: JSON parser not consistent on extra data

mrfuchs, 2022-07-19:
For json strings containing additional objects that are not specified in the descriptor, the behavior of `json_obj_parse()` is inconsistent. It sometimes even is incorrect.

1. Extra objects are supported (`ret == 1`)

```c
struct test_struct_t {
  const char * str;
} ts;

const struct json_obj_descr test_descr[] = {
  JSON_OBJ_DESCR_PRIM(test_struct_t, str, JSON_TOK_STRING)
};

char encoded[] = "{"
  "\"str\":\"abc\","
  "\"dummy\":1"
  "}\n";

int ret = json_obj_parse(encoded, sizeof(encoded) - 1, test_descr, ARRAY_SIZE(test_descr), &ts);
```

2. Extra arrays are not supported (`ret == -EINVAL`)

```c
struct test_struct_t {
  const char * str;
} ts;

const struct json_obj_descr test_descr[] = {
  JSON_OBJ_DESCR_PRIM(test_struct_t, str, JSON_TOK_STRING)
};

char encoded[] = "{"
  "\"str\":\"abc\","
  "\"arr\": [1,2]"
  "}\n";

int ret = json_obj_parse(encoded, sizeof(encoded) - 1, test_descr, ARRAY_SIZE(test_descr), &ts);
```

3. Extra nested objects may manipulate desired objects specified in descriptor (`ret == 1`, but `ts.str == "def"` after parsing)

```c
struct test_struct_t {
  const char * str;
} ts;

const struct json_obj_descr test_descr[] = {
  JSON_OBJ_DESCR_PRIM(test_struct_t, str, JSON_TOK_STRING)
};

char encoded[] = "{"
  "\"extra\": {"
    "\"str\":\"def\""
  "},"
  "\"str\":\"abc\""
  "}\n";

int ret = json_obj_parse(encoded, sizeof(encoded) - 1, test_descr, ARRAY_SIZE(test_descr), &ts);
```

4. Nested extra objects will cause parsing to end prematurely with success return value (`ret == 6`and `ts.str1` not set after parsing although field is given in input)

```c
struct nested_test_struct_t {
  uint32_t num;
};

struct test_struct_t {
  const char * str1;
  nested_test_struct_t obj;
  const char * str2;
} ts;

const struct json_obj_descr nested_test_descr[] = {
  JSON_OBJ_DESCR_PRIM(nested_test_struct_t, num, JSON_TOK_NUMBER)
};

const struct json_obj_descr test_descr[] = {
  JSON_OBJ_DESCR_PRIM(test_struct_t, str1, JSON_TOK_STRING),
  JSON_OBJ_DESCR_OBJECT(test_struct_t, obj, nested_test_descr),
  JSON_OBJ_DESCR_PRIM(test_struct_t, str2, JSON_TOK_STRING)
};

char encoded[] = "{"
  "\"str2\":\"xyz\","
  "\"obj\": {"
    "\"num\":-2,"
    "\"extra\": {"
      "\"str1\":\"def\""
    "}"
  "},"
  "\"str1\":\"abc\""
  "}\n";

int ret = json_obj_parse(encoded, sizeof(encoded) - 1, test_descr, ARRAY_SIZE(test_descr), &ts);
```

Reason for these behaviors is that unknown/additional objects are not skipped, but tried to be parsed instead. In case of arrays this leads to parsing errors due to the unexpected array syntax. In case of objects, unknown parent fields are ignored and child fields are parsed.

erickcinco, 2023-02-13:
Will this be revisited?

bandogora, 2023-02-15:
I also just want to point out that this could help with some of the NULL value handling requests (i.e. #27600).

For example, in my case I have a vendor generated JSON file with many `null` values all belonging to fields I don't care about. The vendor is unlikely to update JSON file for me in any timely fashion; so my best bet is to skip parsing that extra data.

closed 2023-07-18 by fabiobaltieri

fabiobaltieri, 2023-07-18:
> Would there be any interest in a less fragile json parser? I just cooked up a parser using https://github.com/keith-packard/lola which is a table-driven LL parser generator designed for (oddly) embedded systems. Mapping this to the existing API would be straightforward, and that would prevent user-provided JSON from potentially causing a stack overflow while still validating the JSON structure, even portions that aren't used by the implementation.

There's been few discussions recently in the dev review meeting about json libraries, maintaining the internal code vs proposal for an external one. @keith-packard are you proposing to do the integration?

@nashif @MaureenHelm @mrfuchs what do you think?

keith-packard, 2023-07-18:
If you're interested, cooking up an API compatible JSON parser based on the sample I just pushed to the lola repository would be a pretty small piece of work. To avoid depending on GPLv3 code, we could add the generated parse tables to the repository (it's not like JSON syntax is likely to change anytime soon). We'd re-use the existing lexer code.

On x86_64 (-Os), the parser is about 1kB and the parse tables compress down to 136 bytes. I'm not sure how that compares with the existing JSON parser code, but it's probably comparable. The key benefit is that the parse stack is explicitly managed, instead of being implicit in the C stack. So, it's both smaller *and* you can catch overflows and bail when the provided input is nested too deeply. That should make it reasonable to parse all of the input, rather than attempting to skip over unused elements.

PR #47989 (2023-07-18): json: Skip unknown fields on parsing
Skip child objects and arrays that are not specified in the given object
descriptor when parsing a JSON input string.
This patch adds support for extra child arrays which previously were not
supported by the parser as opposed to additional child objects.

Fixes #47988