zephyrproject-rtos/zephyr #47991: BLE functionality for STM32WB55 is limited with full version of BLE stack

r2r0, 2022-07-19:
**Describe the bug**
The stm32wb_ble sample on nucleo_wb55rg board hangs on when it is used in combination with stm32wb5x_BLE_Stack_full_fw.bin (version 0.13.2) as a BLE firmware.
After updating BLE stack with stm32wb5x_BLE_HCILayer_extended_fw.bin sample works as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Update BLE FW with stm32wb5x_BLE_Stack_full_fw.bin (version 0.13.2)
2. west build -p -b nucleo_wb55rg boards/stm32/power_mgmt/stm32wb_ble
3. west flash
4. see error on console and take mesurement of  power consumption after shutdown

**Expected behavior**
In case when full BLE stack version is used mentioned sample should not output any errors and power consumption after shutdown should be much lower (not above 500 uA).

**Impact**
showstopper

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.0.0-6702-g60271524087b  ***
Starting Beacon Demo
Bluetooth init failed (err -5)
BLE disable
BLE restart
Bluetooth init failed (err -5)
BLE disable
Shutdown
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.14.2
 - Commit SHA 60271524087bb4382c8b4db0f2e17a87e4120065

**Additional context**
It is marked as showstopper because our application code also uses bt_disable before shutdown.

erwango, 2022-07-26:
@r2r0, STM32WB Cube package 13.2.0, "_full_" binaries are no more compatible with use in Zephyr. Instead, you should now use one of the following binaries:
stm32wb5x_BLE_HCILayer_extended_fw.bin
stm32wb5x_BLE_HCILayer_fw.bin

I'll update board documentation to make this clear and take this into account in flash partition .

erwango, 2022-07-27:
Re-opening as this issue should be used to backport #48349 on V3.1.1

PR #48349 (2022-08-12): boards: nucleo_wb55rg: Update regarding supported M0 BLE f/w
Since STM32WBCube release V1.13.2, only "HCI Only" f/w are now compatible
for a use with Zephyr.
"Full stack" f/w which used to be supported are not compatible anymore.

Update board documentation to mention this restriction.

Additionally, update flash partition to reflect the gain implied
by the use of this smaller f/w.

Fixes #47991
PR #49030 (2022-09-22): [Backport v3.1-branch] boards: nucleo_wb55rg: Update regarding supported M0 BLE f/w
Backport d00f8505bf6c62540875db9cd40836ae13c2d374 from #48349

Fixes #47991
closed 2022-09-22 by carlescufi