zephyrproject-rtos/zephyr #48007: tests: gpio driver fails in pin_get_config

FRASTM, 2022-07-19:
**Describe the bug**
When running the testcase  tests/drivers/gpio/gpio_basic_api/ on various stm32 boards, the sequence pin_get_config fails because
flag set differs from flag get. 

At least  on the following target boards : stm32f3_disco,  nucleo_g474re, nucleo_g071rb,  nucleo l152r, nucleo_wl55jc  

During the gpio_stm32_config, the _flags_ is set with value  = 0xa0000                                                                                                                                    
During the gpio_stm32_get_config, the _flags_ is get at value = 0x20000b74 

Moreover it seems that :
- the `gpio_stm32_pincfg_to_flags` and `gpio_stm32_flags_to_conf`  functions have parameters that do not match exactly the 32bit values. (gpio_flags_t is of type uint32_t)
- LL_GPIO_GetPinPull, LL_GPIO_GetPinOutputType, LL_GPIO_GetPinMode  returned values are uint32_t assigned to unsigned int mode, otype, pupd
- When disabling the CONFIG_GPIO_GET_CONFIG (with CONFIG_GPIO_GET_CONFIG=n) an compilation error occurs:
```
in function `gpio_pin_get_config':
/local/home/frq09495/zephyrproject/zephyr/build/zephyr/include/generated/syscalls/gpio.h:111: undefined reference to `z_impl_gpio_pin_get_config'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto  -b stm32f3_disco tests/drivers/gpio/gpio_basic_api
3. west flash
4. See error

**Expected behavior**
test PASSED 

**Impact**
test campaign failure

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-2346-g5b5ccee821cf  ***
Running TESTSUITE gpio_basic_test
===================================================================
START - test_gpio_port
Validate device GPIOC
Check GPIOC output 6 connected to input 7
OUT 6 to IN 7 linkage works
- bits_physical
- pin_physical
- check_raw_output_levels
- check_logic_output_levels                                                                                                                     
- check_input_levels                                                                                                                            
- bits_logical                                                                                                                                  
- check_pulls                                                                                                                                   
                                                                                                                                                
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_gpio_port.c:652: pin_get_config: (flags_set not equal to )
flags are different                                                                                                                             
 FAIL - test_gpio_port in 0.37 seconds
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) ubuntu
 - Toolchain (e.g Zephyr SDK, ...) : zephyr-sdk-0.14.0
 - Commit SHA : 60271524087bb4382c8b4db0f2e17a87e4120065

**Additional context**

mmahadevan108, 2022-07-19:
cc @semihalf-barnas-michal

hunterhu3000, 2022-08-04:
@FRASTM As this bug is not related to https://github.com/zephyrproject-rtos/zephyr/issues/47921, can I unassign me in the Assignees list?

closed 2022-08-12 by fabiobaltieri