zephyrproject-rtos/zephyr #48018: ztest: static threads are not re-launched for repeated test suite execution.

smrtos, 2022-07-19:
**Is your enhancement proposal related to a problem? Please describe.**

New ztest fx doesn't have a logic to launch static threads. For test suites that use static threads in a "one-shot" way, when they are shuffled and executed repeatedly, later iterations may run in a different context. And tests can fail. Such as the tests/kernel/mutex/sys_mutex.

For now, static threads can be replaced with dynamic threads. Or shall we add static thread support to new ztest?

aaronemassey, 2022-07-21:
@yperess

aaronemassey, 2022-07-21:
My understanding from the test working group meeting is this is specifically reproduced by shuffling the tests and rerunning the test.

yperess, 2022-07-21:
@smrtos can you link an example or paste a minimal code example? From what I'm hearing you want to have a static thread which is stateful, but order of execution or repeated execution of tests breaks that. Can't you reset state in the thread during the before function of your suite? You'll need some synchronization tools for that, but it should be possible in the current API.

Generally I would recommend to just use dynamic threading. State should never be retained between tests and dynamic threads are the easiest way to guarantee that.

If you have multiple suites, you can use test rules to reset state.

asemjonovs, 2022-07-21:
@smrtos - is it expected behavior of the test to make the static thread exit?

smrtos, 2022-07-22:
> @smrtos can you link an example or paste a minimal code example? From what I'm hearing you want to have a static thread which is stateful, but order of execution or repeated execution of tests [...]

I don't mean stateful static thread. Actually, I don't think a test function should be stateful. I just want to ensure each run of a test function has the exactly same context.

For example, below are some static threads: 
https://github.com/zephyrproject-rtos/zephyr/blob/45cecadb35a2fa8078759bd0c42583a15758c4fb/tests/kernel/mutex/sys_mutex/src/main.c#L394-L395

They interact with below test function:
https://github.com/zephyrproject-rtos/zephyr/blob/45cecadb35a2fa8078759bd0c42583a15758c4fb/tests/kernel/mutex/sys_mutex/src/main.c#L242

These static threads are launched **only once** when the zephyr application starts.  Either with new or old ztest fx. So only the first test run can see them. Then these threads will exit. With shuffle enabled in new ztest, later runs of the same test function have nothing to interact with. The test will fail.

For the specific purpose of this test case, dynamic threads can be used instead. But in case where static threads are the whole point of testing, new ztest fx is not able to test them more than once. 

I did try making the above static threads never exit and use semaphore to synchronize them with the multiple runs of the test function. But it affected the timing subtly and break the test. Finally I use dynamic threads instead. (See #48015)

smrtos, 2022-07-22:
> @smrtos - is it expected behavior of the test to make the static thread exit?

For the design of below test, I think the static threads are expected to exit after the test function ends. https://github.com/zephyrproject-rtos/zephyr/blob/45cecadb35a2fa8078759bd0c42583a15758c4fb/tests/kernel/mutex/sys_mutex/src/main.c#L394

asemjonovs, 2022-07-22:
> For the design of below test, I think the static threads are expected to exit after the test function ends. [...]

Ok, looks like this test wasn't designed to be ran multiple times. Since all the threads are defined within the test, we should be fine with making these dynamic like you mentioned in your other response.

nashif, 2022-08-09:
not sure why the reporter is also the assignee, @yperess  can you please triage and assign to someone to look at this or propose a solution?

closed 2022-10-23 by github-actions[bot]