zephyrproject-rtos/zephyr #48037: Grove LCD Sample Not Working

rod-watt, 2022-07-19:
**Describe the bug**
I have built the samples for the Grove LCD found in zephyr/samples/drivers/misc/grove_display/ but the characters on the screen jump around.  It looks as if the curser position is moving between individual writes. I know it needs to move, but it look like it's moving incorrectly. 

I am using the nrf52840 Dongle board and I have used the Grove LCD with older versions of the code so I know my hardware is OK.

**To Reproduce**

west build -b nrf52840dongle_nrf52840 grove_display --pristine
west flash

**Expected behavior**
I expect to see the LCD flash different colours and the display number that increase

**Impact**
 showstopper

**Environment (please complete the following information):**
 - OS: MacOS,
 - Toolchain Zephyr SDK
 - Commit SHA or Version used 2.0.0

**Additional context**
My overlay file looks like this

/ {
        chosen {
		zephyr,display = &glcd;
        };

};

&pinctrl {
	uart0_default_alt: uart0_default_alt {
		group0 {
			psels = <NRF_PSEL(UART_TX, 0, 24)>, 
				<NRF_PSEL(UART_RX, 0, 22)>;
		};
	};

	uart0_sleep_alt: uart0_sleep_alt {
		group1 {
			psels = <NRF_PSEL(UART_TX, 0, 24)>, 
				<NRF_PSEL(UART_RX, 0, 22)>;
				low-power-enable;
		};
	};

	i2c1_default_alt: i2c1_default_alt {
       		group2 {
          		psels = <NRF_PSEL(TWIM_SDA, 0, 17)>, 
				<NRF_PSEL(TWIM_SCL, 0, 20)>;
       			};
    		};

	i2c1_sleep_alt: i2c1_sleep_alt {
       		group3 {
          		psels = <NRF_PSEL(TWIM_SDA, 0, 17)>,
                  		<NRF_PSEL(TWIM_SCL, 0, 20)>;
				low-power-enable;
       			};
    		};
};

&uart0 {
	pinctrl-0 = <&uart0_default_alt>;
	pinctrl-1 = <&uart0_sleep_alt>;	
	pinctrl-names = "default", "sleep";

	compatible = "nordic,nrf-uart";
	current-speed = <115200>;
	status = "okay";
};

&i2c1 {
	status = "okay";
	compatible = "nordic,nrf-twim";	
	pinctrl-0 = <&i2c1_default_alt>;
	pinctrl-1 = <&i2c1_sleep_alt>;
	pinctrl-names = "default", "sleep";
//    	clock-frequency = <I2C_BITRATE_STANDARD>;  
	clock-frequency = <I2C_BITRATE_FAST>;  
	zephyr,concat-buf-size = <520>;
	glcd: grove-lcd-rgb@3e {
		compatible = "seeed,grove-lcd-rgb";
		reg = <0x3e>;
		label = "GLCD";
	};
};
&i2c1 {
/delete-property/sda-gpios;
/delete-property/scl-gpios;
};
&uart1 {
    status = "disabled";
};
&spi0 {
    status = "disabled";
};
&spi1 {
    status = "disabled";
};
&spi2 {
    status = "disabled";
};
&spi3 {
    status = "disabled";
};

gmarull, 2022-07-25:
Unfortunately, I don't have the hardware to test this. The grove lcd driver has been unmaintained for a while, I just ported it to DT in https://github.com/zephyrproject-rtos/zephyr/pull/42237. Can you check if that PR introduced the issue?

rod-watt, 2022-07-25:
I think so, I tried to run a previous version of the code but its complaint that the parenthesis on my overlay file is incorrect "  parse error: expected number or parenthesized expression". But yes the overlay file works fine on version 2.0.0.

closed 2022-12-09 by github-actions[bot]

Ashu12Aj, 2024-03-29:
is any one  share the grove LCD  working overlay code