zephyrproject-rtos/zephyr #48049: Obtaining the DNS domain name from the DHCP lease

cperera-aud, 2022-07-20:
We would like to obtain the DNS domain name from the DHCP lease so that this DNS domain name can be used later on. In our solution we need to use the this DNS domain name as a search path for looking up DNS records. Following is the workflow expected in our solution:

1. Obtain a DHCP lease with IP address, DNS server, DNS domain name
2. Use the DNS domain name for DNS SRV, TXT record lookups e.g. _service._udp.<DNS domain name>
3. Use information in the DNS record for additional workflows

What we would like is for the Zephyr DHCP client to be enhanced so that it is able to parse the Domain Name DHCP option as per RFC 2132:

[image]

https://www.rfc-editor.org/rfc/rfc2132.html#section-3.17

Subsequently, we would then need a API to access the domain name.

cperera-aud, 2022-07-31:
I'll contribute to this issue. Feel free to assign it to me, I don't seem to be able to assign the issue to myself.

m-byte, 2023-08-07:
The first step can be accomplished using DHCP option callbacks (https://github.com/zephyrproject-rtos/zephyr/blob/01ba923bb203c749f96bfa9d757f515d1fd91838/samples/net/dhcpv4_client/src/main.c#L98)

cperera-aud, 2023-08-07:
Thanks Matthias, for the tip, this work wasn't high priority until recently. I'm hoping to address this in the coming weeks.

cperera-aud, 2024-02-01:
I've tried out the changes on my end and with my application I'm able to get the DNS domain name from the DHCP lease. I didn't have much time to try setting a static DNS domain name in the case of setting a static IP address in my application.

cperera-aud, 2024-06-28:
We're happy with the changes to get the DNS domain, so we're closing this case.

closed 2024-06-28 by cperera-aud