zephyrproject-rtos/zephyr #48051: samples: logger: samples/subsys/logging/logger/sample.logger.basic failed on acrn_ehl_crb board

enjiamai, 2022-07-20:
**Describe the bug**

The samples/subsys/logging/logger/sample.logger.basic is running failed on acrn_ehl_crb board. 

**To Reproduce**
Steps to reproduce the behavior:
1.  west build samples/subsys/logging/logger -p always -b acrn_ehl_crb
2.  west flash /opt/remotehw/remotehw-x86-acrn.sh ehlsku11
3.  After seeing the ACRN boot, press "enter", when you see ACRN:\> enter "vm_console 0" 
4.  You can see the full log but lacking of the last line: complete (passed) after xxxx seconds
or
1.  source zephyr-env.sh
2.  twister -p acrn_ehl_crb --device-testing --device-serial-pty="/opt/remotehw/acrn-test-pty.exp,ehlsku11" --west-flash="/opt/remotehw/remotehw-x86-acrn.sh,ehlsku11" -T samples/subsys/logging/logger -vv
3.  You can see the full log but lacking of the last line: complete (passed) after xxxx seconds

**Expected behavior**
The tests should pass.

**Impact**
Minor

**Logs and console output**
```
ACRN:\>vm_console 0
----- Entering VM 0 Shell -----
Warning message example.
demo_tag [00:00:00.498,980] <inf> main: Info message example.
Logging performance showcase.
demo_tag [00:00:00.247,858] <inf> main: performance test - log message 0
demo_tag [00:00:01.252,558] <inf> main: performance test - log message 0
demo_tag [00:00:02.257,259] <inf> main: performance test - log message 0
demo_tag [00:00:01.001,449] <inf> main: performance test - log message 0
demo_tag [00:00:02.006,150] <inf> main: performance test - log message 0
demo_tag [00:00:00.750,340] <inf> main: performance test - log message 0
demo_tag [00:00:01.755,040] <inf> main: performance test - log message 0
demo_tag [00:00:00.499,230] <inf> main: performance test - log message 0
demo_tag [00:00:01.503,931] <inf> main: performance test - log message 0
demo_tag [00:00:00.248,121] <inf> main: performance test - log message 0
demo_tag [00:00:01.252,822] <inf> main: performance test - log message 0
demo_tag [00:00:02.257,523] <inf> main: performance test - log message 0
demo_tag [00:00:01.001,712] <inf> main: performance test - log message 0
demo_tag [00:00:02.006,413] <inf> main: performance test - log message 0
demo_tag [00:00:02.006,418] <inf> main: performance test - log message 1
demo_tag [00:00:02.006,419] <inf> main: performance test - log message 2
demo_tag [00:00:02.006,419] <inf> main: performance test - log message 3
demo_tag [00:00:02.006,421] <inf> main: performance test - log message 4
demo_tag [00:00:02.006,421] <inf> main: performance test - log message 5
demo_tag [00:00:02.006,422] <inf> main: performance test - log message 6
demo_tag [00:00:00.750,612] <inf> main: performance test - log message 0
demo_tag [00:00:00.750,619] <inf> main: performance test - log message 1
demo_tag [00:00:00.750,620] <inf> main: performance test - log message 2
demo_tag [00:00:00.750,620] <inf> main: performance test - log message 3
demo_tag [00:00:00.750,621] <inf> main: performance test - log message 4
demo_tag [00:00:00.750,621] <inf> main: performance test - log message 5
demo_tag [00:00:00.750,622] <inf> main: performance test - log message 6
demo_tag [00:00:00.750,622] <inf> main: performance test - log message 7
demo_tag [00:00:00.750,623] <inf> main: performance test - log message 8
demo_tag [00:00:00.750,623] <inf> main: performance test - log message 9
demo_tag [00:00:00.750,624] <inf> main: performance test - log message 10
demo_tag [00:00:00.750,624] <inf> main: performance test - log message 11
demo_tag [00:00:00.750,625] <inf> main: performance test - log message 12
demo_tag [00:00:00.750,625] <inf> main: performance test - log message 13
demo_tag [00:00:00.750,626] <inf> main: performance test - log message 14
demo_tag [00:00:00.750,626] <inf> main: performance test - log message 15
demo_tag [00:00:00.750,627] <inf> main: performance test - log message 16
demo_tag [00:00:00.750,627] <inf> main: performance test - log message 17
demo_tag [00:00:00.750,628] <inf> main: performance test - log message 18
demo_tag [00:00:00.750,628] <inf> main: performance test - log message 19
demo_tag [00:00:00.765,911] <inf> main: performance test - log message 3
--- 3 messages dropped ---
demo_tag [00:00:00.765,912] <inf> main: performance test - log message 4
demo_tag [00:00:00.765,912] <inf> main: performance test - log message 5
demo_tag [00:00:00.765,913] <inf> main: performance test - log message 6
demo_tag [00:00:00.765,913] <inf> main: performance test - log message 7
demo_tag [00:00:00.765,914] <inf> main: performance test - log message 8
demo_tag [00:00:00.765,914] <inf> main: performance test - log message 9
demo_tag [00:00:00.765,915] <inf> main: performance test - log message 10
demo_tag [00:00:00.765,915] <inf> main: performance test - log message 11
demo_tag [00:00:00.765,915] <inf> main: performance test - log message 12
demo_tag [00:00:00.765,916] <inf> main: performance test - log message 13
demo_tag [00:00:00.765,916] <inf> main: performance test - log message 14
demo_tag [00:00:00.765,917] <inf> main: performance test - log message 15
demo_tag [00:00:00.765,917] <inf> main: performance test - log message 16
demo_tag [00:00:00.765,918] <inf> main: performance test - log message 17
demo_tag [00:00:00.765,918] <inf> main: performance test - log message 18
demo_tag [00:00:00.765,919] <inf> main: performance test - log message 19
demo_tag [00:00:00.765,919] <inf> main: performance test - log message 20
demo_tag [00:00:00.765,921] <inf> main: performance test - log message 21
demo_tag [00:00:00.765,921] <inf> main: performance test - log message 22
demo_tag [00:00:00.765,922] <inf> main: performance test - log message 23
demo_tag [00:00:00.765,922] <inf> main: performance test - log message 24
demo_tag [00:00:00.765,923] <inf> main: performance test - log message 25
demo_tag [00:00:00.765,923] <inf> main: performance test - log message 26
demo_tag [00:00:00.765,924] <inf> main: performance test - log message 27
demo_tag [00:00:00.765,924] <inf> main: performance test - log message 28
demo_tag [00:00:00.765,925] <inf> main: performance test - log message 29
demo_tag [00:00:00.765,925] <inf> main: performance test - log message 30
demo_tag [00:00:00.765,926] <inf> main: performance test - log message 31
demo_tag [00:00:00.765,927] <inf> main: performance test - log message 32
demo_tag [00:00:00.765,927] <inf> main: performance test - log message 33
demo_tag [00:00:00.765,928] <inf> main: performance test - log message 34
demo_tag [00:00:00.765,928] <inf> main: performance test - log message 35
demo_tag [00:00:00.765,929] <inf> main: performance test - log message 36
demo_tag [00:00:00.765,929] <inf> main: performance test - log message 37
demo_tag [00:00:00.765,929] <inf> main: performance test - log message 38
demo_tag [00:00:00.765,930] <inf> main: performance test - log message 39
demo_tag [00:00:00.765,931] <inf> main: performance test - log message 40
demo_tag [00:00:00.765,931] <inf> main: performance test - log message 41
demo_tag [00:00:00.765,932] <inf> main: performance test - log message 42
demo_tag [00:00:00.765,932] <inf> main: performance test - log message 43
demo_tag [00:00:00.765,933] <inf> main: performance test - log message 44
demo_tag [00:00:00.765,933] <inf> main: performance test - log message 45
demo_tag [00:00:00.765,933] <inf> main: performance test - log message 46
demo_tag [00:00:00.765,935] <inf> main: performance test - log message 47
demo_tag [00:00:00.765,936] <inf> main: performance test - log message 48
demo_tag [00:00:00.765,936] <inf> main: performance test - log message 49
demo_tag [00:00:00.765,937] <inf> main: performance test - log message 50
demo_tag [00:00:00.765,938] <inf> main: performance test - log message 51
demo_tag [00:00:00.765,938] <inf> main: performance test - log message 52
Estimated logging capabilities: 29232 messages/second
Logs from external logging system showcase.
demo_tag [00:00:00.550,501] <err> critical level log
demo_tag [00:00:00.550,502] <err> error level log, 1 arguments: 1
demo_tag [00:00:00.550,503] <wrn> warning level log, 2 arguments: 1 2
demo_tag [00:00:00.550,505] <inf> notice level log, 3 arguments: 100, string, 0x00000255
demo_tag [00:00:00.550,506] <inf> info level log, 4 arguments : 1 2 3 4
demo_tag [00:00:00.550,509] <dbg> debug level log, 5 arguments: 1 2 3 4 5
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA: 

**Additional context**

Zhaoningx, 2022-08-22:
Hi, after shifting the ACRN board from JF lab to SH lab, the issue can't be reproduced. In recent testing, the test case is passed on the ACRN board. So next, I'll compare the board differences between JF lab and SH lab and check if a relatied bug-fixed commit got merged recently.

yerabolu, 2022-08-23:
This issue is not reproducible on different HW. @Zhaoningx will be comparing the differences between the two boards and also checking if there is a related bug-fixed commit that got merged recently. Based on the outcome, the issue will be updated.

Zhaoningx, 2022-08-26:
Hi, the issue can't be reproducible on different HW. I tested the case, the log:
```
  Booting `ACRN hypervisor Hybrid Scenario'
PXE boot test --- Loading hypervisor Hybrid scenario ...
WARNING: no console will be available to OS
error: no suitable video mode found.
UEFI environment detected.
Multiboot2 Bootloader: GRUB 2.04
ACRN Hypervisor
[26519914us][cpu=0][(null)][sev=2][seq=1]:HV version 2.7-2022-01-04 10:57:02-2753dc6c9-dirty DBG (daily tag:acrn-2021w43.2-180000p) hybrid@ehl-crb-b build by mm, start time 26517569us
[26538360us][cpu=0][(null)][sev=2][seq=2]:API version 1.0
[26544309us][cpu=0][(null)][sev=2][seq=3]:Detect processor:    Intel Atom(R) x6427FE Processor @ 1.90GHz
[26604424us][cpu=0][(null)][sev=2][seq=23]:Start VM id: 0 name: SAFETY_VM0
[26607629us][cpu=2][(null)][sev=3][seq=24]:failed to load VM 1 kernel module
[26623562us][cpu=0][vm0:vcpu0][sev=3][seq=28]:vlapic: Start Secondary VCPU1 for VM[0]...
[26632559us][cpu=1][vm0:vcpu1][sev=3][seq=31]:vlapic: WARNING:invalid vpic wire mode change
ACRN:\>
ACRN:\>vm_console 0
----- Entering VM 0 Shell -----
*** Booting Zephyr OS build zephyr-v3.1.0-4151-g180f50feda25  ***
---=< RUNNING LOGGER DEMO FROM KERNEL THREAD >=---
Module logging showcase.
demo_tag [00:00:01.618,322] <inf> sample_module: log in test_module 11
demo_tag [00:00:01.618,322] <inf> sample_module: Inline function.
Disabling logging in the sample_module module
Function called again but with logging disabled.
Instance level logging showcase.
```

So closed this.

closed 2022-08-26 by Zhaoningx