zephyrproject-rtos/zephyr #48055: samples: subsys: usb: audio: headphones_microphone and headset - Can not get USB Device

nordic-piks, 2022-07-20:
**Describe the bug**
The samples/subsys/usb/audio/headphones_microphone/sample.usb.audio.headphones_microphone
and
samples/subsys/usb/audio/headset/sample.usb.audio.headset
fails (due to USB error?)

Observed for
- nrf52840dk_nrf52840
- nrf5340dk_nrf5340_cpuapp
- nrf5340dk_nrf5340_cpuapp_ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/subsys/usb/audio/headphones_microphone -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-2343-g7d7988d1a995  ***
[00:00:00.465,087] <inf> main: Entered main
[00:00:00.465,118] <err> main: Can not get USB Headphones Device
```
```
ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-2343-g7d7988d1a995  ***
[00:00:00.453,277] <inf> main: Entered main
[00:00:00.453,277] <err> main: Can not get USB Headset Device
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: zephyr-v3.1.0-2343-g7d7988d1a995

FRASTM, 2022-07-20:
I found similar issue on the stm32 boards: see https://github.com/zephyrproject-rtos/zephyr/pull/48000

galak, 2022-07-21:
Merged #48000 see if this is still an issue

closed 2022-07-21 by galak

nordic-piks, 2022-07-26:
Issue no longer visible.