zephyrproject-rtos/zephyr #48058: Reading out a GPIO pin configured as output returns invalid value.

Klievan, 2022-07-20:
**Describe the bug**
On an nRF52-device, if an output-pin is set using gpio_pin_set() and is subsequently read by gpio_pin_get() or gpio_pin_get_raw() it does not return the updated value. I went through the documentation on gpio_pin_get() and it does not specify anything about the direction of the pin having to be input. On first glance it seems that this function can be used to check the output-state of a pin. Current workaround is to keep track of the pin-state in the application itself.

**To Reproduce**
Steps to reproduce the behavior:
1. call gpio_pin_set() on an arbitrary output pin on an nRF52 device. 
2. call gpio_pin_get() or gpio_pin_get_raw() and observe that the value does not update.3. 

**Expected behavior**
It to return the correct value, regardless of the direction, or a proper warning in the documentation that this only reads input pins. Not sure if this is a bug with the nRF GPIO driver or expected behavior which isn't documented.

**Impact**
Minimal, minor loss of development time

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr OS build v3.0.99-ncs1

**Additional context**
The article below documents such behavior on the nRF52 with their driver. Not sure if this is relevant.
https://devzone.nordicsemi.com/f/nordic-q-a/20706/nrf_gpio_pin_read-return-always-0-with-a-gpio-as-output

FRASTM, 2022-07-21:
is it a similar  to issue https://github.com/zephyrproject-rtos/zephyr/issues/48007 ?

Klievan, 2022-07-22:
Don't think so. Went through the PR you mentioned and that would not fix this issue. That issue seems related to the pin configuration, while this issue is about reading out the output state of a pin.

bpbradley, 2022-07-23:
I can confirm this behavior. I rolled back to 2.7 though and it is still present, so it may be intentional. On nrf52840, gpio_pin_get always returns the default value, and does not change when the state is changed.

anangl, 2022-08-08:
> I went through the documentation on gpio_pin_get() and it does not specify anything about the direction of the pin having to be input.

Well, its description says it is intended for input pins: https://github.com/zephyrproject-rtos/zephyr/blob/eafc4f875b7443bceffe7f8bee14aa97d7ca4bd6/include/zephyr/drivers/gpio.h#L1203
Similarly for `gpio_pin_get_raw()`:
https://github.com/zephyrproject-rtos/zephyr/blob/eafc4f875b7443bceffe7f8bee14aa97d7ca4bd6/include/zephyr/drivers/gpio.h#L1170

You can configure a pin as both output and input. Then, `gpio_pin_get()` will return the actual output state of the pin.

closed 2022-08-08 by anangl

stemschmidt, 2025-07-16:
Configuring the pin as both output and input is confusing, because I do not want to use it as an input. I just want to read the output state of the pin, so that I do not have to track it in a variable.