zephyrproject-rtos/zephyr #48081: tests/drivers/clock_control/stm32_clock_configuration/stm32u5_core not working with msis 48

tpennors, 2022-07-20:
**Describe the bug**
Unit test "tests/drivers/clock_control/stm32_clock_configuration/stm32u5_core" not working with clock msis 48MHz
Not working means that core doesn't even start. No zephyr banner.
Info : Example with overlay pll are working.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b b_u585i_iot02a tests/drivers/clock_control/stm32_clock_configuration/stm32u5_core/ -DDTC_OVERLAY_FILE="boards/clear_clocks.overlay;boards/msis_48.overlay" -p
2. west flash
3. See MCU not starting

**Impact**
stm32u5 does not start if we choose to use msis as clock

**Environment**
 - OS: ubuntu 20.04 running on WSL2
 - Toolchain : Zephyr SDK 0.13.1
 - Commit SHA : b7f1e9872495ee29d223cd7b5c2f8fb5eceab2cd

FRASTM, 2022-07-21:
@tpennors could you please check with the PR

tpennors, 2022-07-21:
> @tpennors could you please check with the PR

Wonderful, it works for me now !

closed 2022-08-10 by carlescufi