zephyrproject-rtos/zephyr #48089: AF_PACKET sockets not filling L2 header details in `sockaddr_ll`

krish2718, 2022-07-20:
**Describe the bug**

We have agreed [here](https://github.com/zephyrproject-rtos/zephyr/pull/45592#issuecomment-1127794846) that all unknown ether types will be passed and its up to application to filter. But with [this](https://github.com/zephyrproject-rtos/zephyr/pull/45592) L2 header is now removed but the `recvfrom` doesn't fill the `sockaddr_ll` details, so, we don't have access to the protocol (ether type) for filtering. 

**To Reproduce**
Open and Receive packets from AF_PACKET and try to check `sockaddr_ll` information, `sll_addr`, `sll_protocol` etc are zeros.

**Expected behavior**
All removed L2 header details should be filled in `sll_*` fields.

**Impact**
Extra processing and filtering should be one at app.

**Logs and console output**
Can provide on demand.

**Environment (please complete the following information):**
nrf7002DK

krish2718, 2022-07-20:
@rlubos unable to assign, so, tagging you.

rlubos, 2022-07-28:
@krish2718 I've submitted https://github.com/zephyrproject-rtos/zephyr/pull/48422, please give it a try.

PR #48422 (2022-08-01): net: sockets: Fill the address structure provided in recvfrom() 
The packet socket implementation did not fill the address structure
provided by the application. This commit fixes this.

Note, that the implementation needs to cover two cases: SOCK_RAW and
SOCK_DGRAM. In the first case, the information is extracted directly
from the L2 header (curently only Ethernet supported). In latter case,
the header is already removed from the packet as the L2 has already
processed the packet, so the information is obtained from the net_pkt
structure.

Fixes #48089
closed 2022-08-01 by carlescufi